sle118/squeezelite-esp32 #232: Device keeps rebooting after exit initial recovery mode.

jairbj, 2023-03-15:
Hi friends, I've received a ESP32 Wrover board and flashed it using web installer, successfully connected it to my wifi but when I click in exit recovery, the device keeps rebooting and the only way I found to come back to recovery is flash it again.

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6480
load:0x40078000,len:15868
load:0x40080400,len:4384
entry 0x40080684
I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (27) boot: compile time 21:33:16
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 1
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (88) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 recovery         factory app      00 00 00010000 00140000
I (103) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (110) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (118) boot: End of partition table
I (122) boot_comm: chip revision: 3, min. application chip revision: 0
I (129) esp_image: segment 0: paddr=00150020 vaddr=3f400020 size=af1ach (717228) map
I (350) esp_image: segment 1: paddr=001ff1d4 vaddr=3ffbdb60 size=00e44h (  3652) load
I (352) esp_image: segment 2: paddr=00200020 vaddr=400d0020 size=1c3b30h (1850160) map
I (904) esp_image: segment 3: paddr=003c3b58 vaddr=3ffbe9a4 size=0618ch ( 24972) load
I (913) esp_image: segment 4: paddr=003c9cec vaddr=40080000 size=1de48h (122440) load
I (955) esp_image: segment 5: paddr=003e7b3c vaddr=400c0000 size=00068h (   104) load
I (955) esp_image: segment 6: paddr=003e7bac vaddr=50000000 size=00010h (    16) load
I (976) boot: Loaded app from partition at offset 0x150000
I (977) boot: Disabling RNG early entropy source...
I (988) psram: This chip is ESP32-D0WD
I (988) spiram: Found 64MBit SPI RAM device
I (988) spiram: SPI RAM mode: flash 80m sram 80m
I (994) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1001) cpu_start: Pro cpu up.
I (1005) cpu_start: Starting app cpu, entry point is 0x4008146c
I (998) cpu_start: App cpu up.
I (1510) spiram: SPI SRAM memory test OK
I (1523) cpu_start: Pro cpu start user code
I (1523) cpu_start: cpu freq: 240000000
I (1523) cpu_start: Application information:
I (1526) cpu_start: Project name:     Squeezelite-ESP32
I (1532) cpu_start: App version:      I2S-4MFlash.16.2.1254.master-v4
I (1539) cpu_start: Compile time:     Dec  2 2022 21:35:38
I (1545) cpu_start: ELF file SHA256:  569947c6e9a94421...
I (1551) cpu_start: ESP-IDF:          v4.3.2-dirty
I (1557) heap_init: Initializing. RAM available for dynamic allocation:
I (1564) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
W (1570) REMOVE: NEW POOL of 7268 bytes, ctrl_size: 416 sli_c:8 fli_c:10 small_b 32 max_b:16384
I (1579) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
W (1586) REMOVE: NEW POOL of 16628 bytes, ctrl_size: 736 sli_c:16 fli_c:10 small_b 64 max_b:32768
I (1595) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
W (1601) REMOVE: NEW POOL of 98172 bytes, ctrl_size: 872 sli_c:16 fli_c:12 small_b 64 max_b:131072
I (1611) heap_init: At 3FFC8070 len 00017F90 (95 KiB): DRAM
I (1617) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1624) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
W (1630) REMOVE: NEW POOL of 8612 bytes, ctrl_size: 416 sli_c:8 fli_c:10 small_b 32 max_b:16384
I (1640) heap_init: At 4009DE48 len 000021B8 (8 KiB): IRAM
I (1646) spiram: Adding pool of 4019K of external SPI memory to heap allocator
W (1654) REMOVE: NEW POOL of 4116187 bytes, ctrl_size: 2168 sli_c:32 fli_c:16 small_b 128 max_b:4194304
I (1664) spi_flash: detected chip: generic
I (1669) spi_flash: flash io: qio
I (1676) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
W (1683) REMOVE: NEW POOL of 15052 bytes, ctrl_size: 416 sli_c:8 fli_c:10 small_b 32 max_b:16384
W (1693) REMOVE: NEW POOL of 113820 bytes, ctrl_size: 872 sli_c:16 fli_c:12 small_b 64 max_b:131072
I (1703) spiram: Reserving pool of 64K of internal memory for DMA/internal allocations
W (1713) REMOVE: NEW POOL of 65515 bytes, ctrl_size: 804 sli_c:16 fli_c:11 small_b 64 max_b:65536
I (1723) esp_app_main: Reset reason is: 4
I (1723) esp_app_main: Reboot counter=1

I (1733) esp_app_main: Starting app_main
I (1733) nvs_utilities: Initializing flash nvs 
I (1753) nvs_utilities: Initializing nvs partition settings
I (1793) esp_app_main: Setting up telnet.
I (1793) telnet: ***Redirecting log output to telnet

assert failed: 0x400d560e

Backtrace:0x40081a9d:0x3f8075b0 0x40097c45:0x3f8075d0 0x4009c2e5:0x3f8075f0 0x400d560e:0x3f807630 0x400dde97:0x3f807650 0x400eeca0:0x3f807980

ELF file SHA256: 569947c6e9a94421

Board:
[image]

sle118, 2023-03-15:
Please deactivate telnet from the options so we can have a complete trace. There's likely something wrong with your configuration that makes the system reboot.

jairbj, 2023-03-16:
Here you are

```
ELF file SHA256: 569947c6e9a94421

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6480
load:0x40078000,len:15868
load:0x40080400,len:4384
entry 0x40080684
I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (27) boot: compile time 21:33:16
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 1
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (88) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 recovery         factory app      00 00 00010000 00140000
I (103) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (110) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (118) boot: End of partition table
I (122) boot_comm: chip revision: 3, min. application chip revision: 0
I (129) esp_image: segment 0: paddr=00150020 vaddr=3f400020 size=af1ach (717228) map
I (350) esp_image: segment 1: paddr=001ff1d4 vaddr=3ffbdb60 size=00e44h (  3652) load
I (352) esp_image: segment 2: paddr=00200020 vaddr=400d0020 size=1c3b30h (1850160) map
I (904) esp_image: segment 3: paddr=003c3b58 vaddr=3ffbe9a4 size=0618ch ( 24972) load
I (913) esp_image: segment 4: paddr=003c9cec vaddr=40080000 size=1de48h (122440) load
I (955) esp_image: segment 5: paddr=003e7b3c vaddr=400c0000 size=00068h (   104) load
I (955) esp_image: segment 6: paddr=003e7bac vaddr=50000000 size=00010h (    16) load
I (976) boot: Loaded app from partition at offset 0x150000
I (977) boot: Disabling RNG early entropy source...
I (988) psram: This chip is ESP32-D0WD
I (988) spiram: Found 64MBit SPI RAM device
I (988) spiram: SPI RAM mode: flash 80m sram 80m
I (994) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1001) cpu_start: Pro cpu up.
I (1005) cpu_start: Starting app cpu, entry point is 0x4008146c
I (998) cpu_start: App cpu up.
I (1510) spiram: SPI SRAM memory test OK
I (1523) cpu_start: Pro cpu start user code
I (1523) cpu_start: cpu freq: 240000000
I (1523) cpu_start: Application information:
I (1526) cpu_start: Project name:     Squeezelite-ESP32
I (1532) cpu_start: App version:      I2S-4MFlash.16.2.1254.master-v4
I (1539) cpu_start: Compile time:     Dec  2 2022 21:35:38
I (1545) cpu_start: ELF file SHA256:  569947c6e9a94421...
I (1551) cpu_start: ESP-IDF:          v4.3.2-dirty
I (1557) heap_init: Initializing. RAM available for dynamic allocation:
I (1564) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
W (1570) REMOVE: NEW POOL of 7268 bytes, ctrl_size: 416 sli_c:8 fli_c:10 small_b 32 max_b:16384
I (1579) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
W (1586) REMOVE: NEW POOL of 16628 bytes, ctrl_size: 736 sli_c:16 fli_c:10 small_b 64 max_b:32768
I (1595) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
W (1601) REMOVE: NEW POOL of 98172 bytes, ctrl_size: 872 sli_c:16 fli_c:12 small_b 64 max_b:131072
I (1611) heap_init: At 3FFC8070 len 00017F90 (95 KiB): DRAM
I (1617) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1624) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[... 125 lines cut ...]
  -b <stream>:<output>	Specify internal Stream and Output buffer sizes in Kbytes
  -c <codec1>,<codec2>	Restrict codecs to those specified, otherwise load all available codecs; known codecs: flac,pcm,mp3,ogg (mad,mpg for specific mp3 codec)
  			Codecs reported to LMS in order listed, allowing codec priority refinement.
  -C <timeout>		Close output device when idle after timeout seconds, default is to keep it open while player is 'on'
  -d <log>=<level>	Set logging level, logs: all|slimproto|stream|decode|output, level: info|debug|sdebug
  -e <codec1>,<codec2>	Explicitly exclude native support of one or more codecs; known codecs: flac,pcm,mp3,ogg (mad,mpg for specific mp3 codec)
  -f <logfile>		Write debug to logfile
  -m <mac addr>		Set mac address, format: ab:cd:ef:12:34:56
  -M <modelname>	Set the squeezelite player model name sent to the server (default: SqueezeESP32)
  -n <name>		Set the player name
  -N <filename>		Store player name in filename to allow server defined name changes to be shared between servers (not supported with -n)
  -W			Read wave and aiff format from header, ignore server parameters
  -r <rates>[:<delay>]	Sample rates supported, allows output to be off when squeezelite is started; rates = <maxrate>|<minrate>-<maxrate>|<rate1>,<rate2>,<rate3>; delay = optional delay switching rates in ms
  -R -u [params]	Resample, params = (b|l|m)[:i],
   			 b = basic linear interpolation, l = 13 taps, m = 21 taps, i = interpolate filter coefficients
  -Z <rate>		Report rate to server in helo as the maximum sample rate we can support
  -t 			License terms
  -? 			Display this help text

Build options: EMBEDDED LOOPBACK RESAMPLE16 NO_FAAD LINKALL

assert failed: 0x400d560e

Backtrace:0x40081a9d:0x3f8075b0 0x40097c45:0x3f8075d0 0x4009c2e5:0x3f8075f0 0x400d560e:0x3f807630 0x400dde97:0x3f807650 0x400eeca0:0x3f807980
```

jairbj, 2023-03-16:
My fault. I forgot to add the '-s' before server IP. Added it and now it's working.

closed 2023-03-16 by jairbj