sle118/squeezelite-esp32 #233: console_push could not be resolved in telnet.c

janvdsar, 2023-03-24:
Hi, I’m trying to get the telnet part working. 
But before the code would compile I need to find the console_push function. 
It’s not anywhere in the squeeze-lite library and not in the libtelnet library.  Can you help me?

static void handle_telnet_events(telnet_t *thisTelnet, telnet_event_t *event, void *userData) {
	struct telnetUserData *telnetUserData = (struct telnetUserData *)userData;

	switch(event->type) {
	case TELNET_EV_SEND:
		send(telnetUserData->sockfd, event->data.buffer, event->data.size, 0);
		break;
	case TELNET_EV_DATA:
		console_push(event->data.buffer, event->data.size);
		break;
	case TELNET_EV_TTYPE:
		telnet_ttype_send(telnetUserData->tnHandle);
		break;
	default:
		break;
	}
}

sle118, 2023-03-25:
Actually, a simple search in the repository will show you where it is.

https://github.com/sle118/squeezelite-esp32/blob/982ac3dc8a0908e0cfa98dfe088293edad7f8593/components/platform_console/platform_console.c#L337

There are several branches in the project. It is important that you provide some details about your setup and what you are trying to do as the are to many possibilities for anyone to guess you trying to achieve and what your environment looks like.

sle118, 2023-03-28:
I'll go ahead and close this for now. You can reopen when you have some time.

closed 2023-03-28 by sle118