sle118/squeezelite-esp32 #234: Not working on some devices

AndreasBVB, 2023-03-29:
I have a WROVER knockoff board whose flash is wired slightly differently from common boards. I built ESPHome with QIO and it works fully. Squeezelite-ESP32 works after first flash, but won't boot anymore after power unplug/replug.

This board:
 - 3v3 flash chips, VCC wired to VDD SDIO, with MTDI/GPIO12 low at boot
Common boards:
 - Either: 3v3 flash chips, VCC wired directly to 3v3
 - OR: 1v8 flash chips, VCC wired to VDD SDIO, with MTDI/GPIO12 high at boot

As soon as I rewired the flash chip VCC to 3v3, Squeezelite-ESP32 works fully. Sadly I forgot to keep the log before mod... I am wondering if there is any coding improvements on the Squeezelite-ESP32 side to accommodate this kind of setup. Though less common, it is still using a valid wiring. I also found some might-be related discussions:
https://github.com/espressif/esp-idf/issues/1944

BTW this is related to a previously brought up issue:
https://github.com/sle118/squeezelite-esp32/issues/224
[Help wanted: building with a different flashmode (e.g. QOUT)]
The initial flash chip was very bad, not supporting ESPHome QIO either. I swapped in a Winbond W25Q32 chip then ESPHome QIO is working but still not Squeezelite-ESP32. Another knockoff board like this is being shipped to me. Hopefully I can do more tests with it.

AndreasBVB, 2023-03-29:
[image]

sle118, 2023-04-01:
Not sure what to say. Hardware issues are what they are and we cannot possibly purchase every device out there and then spend hours on trying to make it work; this is an open source project after all.

Anyone willing to dedicate time and do a proper pull request is welcome.

closed 2023-04-01 by sle118