sle118/squeezelite-esp32 #242: Feature request: use 10 instead of 6 presets for buttons

jofland, 2023-04-02:
At the moment it is only possible to use 6 presets  in button config:

`BCTRLS_PS1, BCTRLS_PS2, BCTRLS_PS3, BCTRLS_PS4, BCTRLS_PS5, BCTRLS_PS6​`

I would like to use the whole 10 possible presets from lms´s preset editor.

philippe44, 2023-04-02:
Done and pushed. Please do your share of work and kindly test the feature for *all* buttons 7 to 10 as I don't have time myself to run the test.

jofland, 2023-04-03:
Thanks for implementing so quickly. I would like to test it, but I cannot build it on my own. I need a binary for I2S-4MFlash or an release in "software updates". Please give me a hint when I can start testing.

How is preset 10 labeled? `BCTRLS_PS10` or `BCTRLS_PS0`?

philippe44, 2023-04-03:
It's already build, go into the action/artifacts and take the latest package. Then do a local download (from a file). 10 is P10

sle118, 2023-04-03:
Bottom if this page has binaires for all platforms:
https://github.com/sle118/squeezelite-esp32/actions/runs/4586803592

jofland, 2023-04-04:
Just tried to test my feature request. There are two obstacles:

1. I tried to upload version `squeezelite-esp32-master-v4.3-I2S-4MFlash-16-21279 ` (zip file) via "local firmware upload" to my device and got Error "Error: OTA Partition write failure. ()". Do I have to unzip the file first?
2. Then I noticed that this version was build 3 days ago. You implemented the feature 2 days ago. So this version might not cover the feature at all?

sle118, 2023-04-04:
Have to unzip so you can upload the bin file

jofland, 2023-04-05:
How is preset 10 labeled? BCTRLS_PS10 or BCTRLS_PS0? Both not working for me.

philippe44, 2023-04-05:
It is 10 but if it does not work I'll check for an error. The others work ?

jofland, 2023-04-05:
Yes. 1 to 9 work.

jofland, 2023-04-05:
Build: squeezelite-esp32-master-v4.3-I2S-4MFlash-16-21282.zip

Buttons:
`[{
	"gpio": 107,
	"type": "BUTTON_LOW",
	"pull": true,
	"long_press": 1000,
	"normal": {
		"pressed": "BCTRLS_PS1"
	},
	"longpress": {
		"pressed": "BCTRLS_PS5"
	}
}, {
	"gpio": 106,
	"type": "BUTTON_LOW",
	"pull": true,
	"long_press": 1000,
	"normal": {
		"pressed": "BCTRLS_PS2"
	},
	"longpress": {
		"pressed": "BCTRLS_PS6"
	}
}, {
	"gpio": 105,
	"type": "BUTTON_LOW",
	"pull": true,
	"long_press": 1000,
	"normal": {
		"pressed": "BCTRLS_PS3"
	},
	"longpress": {
		"pressed": "BCTRLS_PS7"
	}
}, {
	"gpio": 104,
	"type": "BUTTON_LOW",
	"pull": true,
	"long_press": 1000,
	"normal": {
		"pressed": "BCTRLS_PS4"
	},
	"longpress": {
		"pressed": "BCTRLS_PS8"
	}
}, {
	"gpio": 103,
	"type": "BUTTON_LOW",
	"pull": true,
	"long_press": 1000,
	"normal": {
		"pressed": "BCTRLS_PS9"
	},
	"longpress": {
		"pressed": "BCTRLS_PS10"
	}
}]`
gpio 103 with normal press PS9 is working; gpio 103 with long press PS10 does not do anything. All other PS in this config is working.

philippe44, 2023-04-05:
You're probably right, 10 is 0. I've changed that, let me know how it works

jofland, 2023-04-06:
Test successful.

Build: squeezelite-esp32-master-v4.3-I2S-4MFlash-16-21284 

Buttons:
`[{ 	"gpio": 107, 	"type": "BUTTON_LOW", 	"pull": true, 	"long_press": 1000, 	"normal": { 		"pressed": "BCTRLS_PS1" 	}, 	"longpress": { 		"pressed": "BCTRLS_PS5" 	} }, { 	"gpio": 106, 	"type": "BUTTON_LOW", 	"pull": true, 	"long_press": 1000, 	"normal": { 		"pressed": "BCTRLS_PS2" 	}, 	"longpress": { 		"pressed": "BCTRLS_PS6" 	} }, { 	"gpio": 105, 	"type": "BUTTON_LOW", 	"pull": true, 	"long_press": 1000, 	"normal": { 		"pressed": "BCTRLS_PS3" 	}, 	"longpress": { 		"pressed": "BCTRLS_PS7" 	} }, { 	"gpio": 104, 	"type": "BUTTON_LOW", 	"pull": true, 	"long_press": 1000, 	"normal": { 		"pressed": "BCTRLS_PS4" 	}, 	"longpress": { 		"pressed": "BCTRLS_PS8" 	} }, { 	"gpio": 103, 	"type": "BUTTON_LOW", 	"pull": true, 	"long_press": 1000, 	"normal": { 		"pressed": "BCTRLS_PS9" 	}, 	"longpress": { 		"pressed": "BCTRLS_PS0" 	} }]`

closed 2023-04-06 by jofland