sle118/squeezelite-esp32 #243: Opus file doesn't play (created from 96 kHz flac)

robho, 2023-04-02:
I have some opus files which don't play with squeezelite-esp32 (but they do play with "normal" squeezelite). One thing that these unplayable opus files have in common is that they were created from 24 bit, 96 kHz flac files.

The output I get is:
```
[12:32:56.726710] output_thread_i2s:463 Output state is 1
[12:32:58.048191] opus_decompress:138 open_callbacks error: -133
[12:32:58.048898] decode_thread:100 decode complete

```

-133 seems to be OP_EBADHEADER.

Opus encoded files use 48 kHz sample rate exclusively (input is resampled) so it feels like there should be no difference playing these files compared to other files, created from 44.1 kHz, for example. One difference I see is that the unplayable files have "Input sample rate" set to 96 kHz in the [opus header](https://datatracker.ietf.org/doc/html/rfc7845#section-5.1), but this value shouldn't affect playback, as I understand it.

(I was trying to debug this further but I couldn't find how the opus decoder was built and from which source. If you can tell me I can dig a little further.)

Here's an unplayable opus file:
[01 - Matt Gray - Bangkok Knights Loader.zip](https://github.com/sle118/squeezelite-esp32/files/11132732/01.-.Matt.Gray.-.Bangkok.Knights.Loader.zip)

philippe44, 2023-04-05:
Congratulations! you've find 2 issues

1- one that impact all squeezelite instances, including the main version (was my own mistake ...)
2- but, that was not enough... on esp32, even the fix does not work as the default opus reader tries to load the whole image in memory, which does not work very well of course

So to solve 2/, I had to remove the entire opusfile wrapper, which is really convenient and provided by opus itself, and use primitive to access to raw stream and skip the "VorbisComment".

Now, there is always something good as getting rid of the opusfile wrapper gained another 7kB of memory and seems to have gained 10% (to be confirmed) of CPU. 

Version is being built as I'm writing this, so please give it a try (not released) but more verifications to be made

robho, 2023-04-05:
Let me know if you want some help testing (a binary would be good then since I don't have a build environment at the moment).

(I have some changes for the AudioScan library that enables seek in opus files, in case you want to test that)

philippe44, 2023-04-10:
Done for both

closed 2023-04-10 by philippe44