sle118/squeezelite-esp32 #245: Muse Luxe doesn't reconnect when LMS is shut down overnight

andarotajo, 2023-04-14:
Hey, I've been experiencing some connection issues with my 2 Muse Luxe speakers using this firmware. For my smart home, I'm shutting down my server that's running the Logitech Media Server for the night. The speakers are both permanently powered so they are kept on over the night.

At morning when I start my server (with LMS) the speakers don't reconnect though. I have to physically restart the speakers for them to reconnect to the LMS. And its not like they just don't find the LMS but they lose their wifi connection completely for some reason. Like they go into hibernate or sth. I would've remote restarted them otherwise but this way I have to do it manually.

Can this be a problem with the firmware or maybe the speaker itself? Has anyone else experienced this?

sle118, 2023-04-14:
As always, if you don't provide more details on the firmware you're running, there's nothing I can do.

sle118, 2023-04-14:
Not all changes are logged, but you're running some recent version. 

Since it's not always obvious to reproduce user issues, it would be really helpful if you left your Muse Luxe connected to a USB port in order to capture some logs. If you don't have the espressif built tools installed, you can navigate to our installer site, connect to the Muse's usb port and then choose to view logs.  Having a trace will jump start my analysis.

https://sle118.github.io/squeezelite-esp32-installer/

andarotajo, 2023-04-15:
Okay so I tried it with your site but it only logged some start up stuff when I clicked "download logs" but nothing else. 

That sounded wrong to me so I used telnet instead and logged it over night but I couldn't reproduce my problem this time. The logs look normal to me for this time: https://pastebin.com/qr2Bjb7d

I can try again with the installer site today, it just seemed weird to me that it didn't actively log something as it did with telnet

sle118, 2023-04-15:
Alternatively, you can try putty or another terminal tool to get these logs. That might work

andarotajo, 2023-04-16:
That worked it seems :) Here's the log: https://pastebin.com/YmBmjNwX hopefully this helps

I've removed all the duplicate stuff because it would be over 9k lines otherwise, it was just the battery and `strm` stuff. The first 270 lines were some disconnect where I manually restarted the speaker. I think the relevant part starts at line 390, which seems to be around 3:44 AM.

sle118, 2023-04-16:
Thank you for providing the logs. This might be related to a big in the esp-idf's wifi stack. There's an open issue that discusses this and they made a suggestion on the power save mode. Seems like at some point the esp32 losses connection with the wifi and that's not good.

But I think there's another thing that happens. The Squeezelite likely gives up trying to connect and eventually just exits. Previously, we were rebooting the platform automatically when that thread excited, but for the sake of troubleshooting command lines this bit was refactored.

I'll see if I can do something later today about this, unless @philippe44 is around, as he typically owns this part of the code.

philippe44, 2023-04-16:
I see 2 things in that log. One is that I've left a bug in the squeezelite exit mechanism and with the exit cause, it should have rebooted but the other one of a WiFi loss of connection which might be an AP issue but per our other discussion, I'm wondering if we don't also have an issue with our own network state machine. But I'll fix the squeezelite exit first

sle118, 2023-04-16:
This is what I thought, with regards to the exit code.  I'll take care of the state machine then!

philippe44, 2023-04-16:
Fixed

closed 2023-04-16 by philippe44

andarotajo, 2023-04-16:
Thanks for the quick responses, I'll try it out as soon as I can

kajmaj, 2023-12-10:
I would like to jump in this issue as well. I run several Squeezelite ESP devices with FW 2.1652-16, I2S-4MFLASH. After powercut (cca 10 min) when router and LMS server were down "Squeezelite"devices did not reconnect when wifi was up and running again. I had to disconnect/reconnect them all to the power. Is there any solution available?

philippe44, 2023-12-10:
If you have forced a LMS server on the command line, remove it.

kajmaj, 2023-12-10:
As I never heard about such an option so I can assume that I haven´t forced the server on the command line.

philippe44, 2023-12-10:
yes, do not set server

kajmaj, 2023-12-11:
Thanks. Deleted server from the setting, everything seems to be working. BTW what is the purpose of the server setting?

philippe44, 2023-12-11:
It locks the player to that server and to that server only. I usually don't recommend setting it as discovery is automatic and locking has negative side effects

kajmaj, 2023-12-11:
Copy that. Thank you