sle118/squeezelite-esp32 #246: building cspot/nanopb

wizmo2, 2023-04-22:
Is there a trick to getting cspot to build?

I've been relying on pre-builds until recently, using an existing 'cmake' condition .  The current version requires a rebuild, but I cant figure out what's wrong.

Here are the logs (built on W10 and the v43 docker image)
```
root@8a4f6eddc373:/project# idf.py clean
Executing action: clean
Running ninja in directory /project/build
Executing "ninja clean"...
[1/1] Cleaning all built files...
Cleaning... 634 files.
root@8a4f6eddc373:/project# idf.py all
Executing action: all (aliases: build)
Running ninja in directory /project/build
Executing "ninja all"...
[4/1518] cd /project/build && /opt/esp/tools/cmake/3.16.4/bin/cmake -P /project/generate_debug_scripts.cmake
Generating debug script files
[5/1518] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
recovery,app,factory,0x10000,1280K,
ota_0,app,ota_0,0x150000,2688K,
settings,data,nvs,0x3f0000,64K,
*******************************************************************************
[248/1518] Performing configure step for 'bootloader'
-- Component directory /opt/esp/idf/components/esp_phy does not contain a CMakeLists.txt file. No component will be added
-- Component directory /opt/esp/idf/components/ieee802154 does not contain a CMakeLists.txt file. No component will be added
-- Component directory /opt/esp/idf/components/openthread does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /project/sdkconfig
-- Adding linker script /opt/esp/idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /opt/esp/idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script /opt/esp/idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32 esp_common esp_hw_support esp_rom esp_system esptool_py hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /opt/esp/idf/components/bootloader /opt/esp/idf/components/bootloader_support /opt/esp/idf/components/efuse /opt/esp/idf/components/esp32 /opt/esp/idf/components/esp_common /opt/esp/idf/components/esp_hw_support /opt/esp/idf/components/esp_rom /opt/esp/idf/components/esp_system /opt/esp/idf/components/esptool_py /opt/esp/idf/components/hal /opt/esp/idf/components/log /opt/esp/idf/components/bootloader/subproject/main /opt/esp/idf/components/bootloader/subproject/components/micro-ecc /opt/esp/idf/components/newlib /opt/esp/idf/components/partition_table /opt/esp/idf/components/soc /opt/esp/idf/components/spi_flash /opt/esp/idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /project/build/bootloader
[267/1518] Performing build step for 'bootloader'
[1/2] Linking C executable bootloader.elf
[2/2] Generating binary image from built executable
esptool.py v3.2-dev
Merged 1 ELF section
Generated /project/build/bootloader/bootloader.bin
[550/1518] Building C object esp-idf/freertos/CMakeFiles/__idf_freertos.dir/event_groups.c.obj
/opt/esp/idf/components/freertos/event_groups.c: In function 'xEventGroupCreateStatic':
/opt/esp/idf/components/freertos/event_groups.c:107:20: warning: unused variable 'xSize' [-Wunused-variable]
    volatile size_t xSize = sizeof( StaticEventGroup_t );
                    ^~~~~
[554/1518] Building C object esp-idf/freertos/CMakeFiles/__idf_freertos.dir/stream_buffer.c.obj
/opt/esp/idf/components/freertos/stream_buffer.c: In function 'xStreamBufferGenericCreateStatic':
/opt/esp/idf/components/freertos/stream_buffer.c:335:20: warning: unused variable 'xSize' [-Wunused-variable]
    volatile size_t xSize = sizeof( StaticStreamBuffer_t );
                    ^~~~~
/opt/esp/idf/components/freertos/stream_buffer.c: In function 'prvInitialiseNewStreamBuffer':
/opt/esp/idf/components/freertos/stream_buffer.c:1231:20: warning: unused variable 'xWriteValue' [-Wunused-variable]
   const BaseType_t xWriteValue = 0x55;
                    ^~~~~~~~~~~
[558/1518] Building C object esp-idf/freertos/CMakeFiles/__idf_freertos.dir/tasks.c.obj
/opt/esp/idf/components/freertos/tasks.c: In function 'vTaskSuspend':
/opt/esp/idf/components/freertos/tasks.c:1962:16: warning: unused variable 'suspended' [-Wunused-variable]
     BaseType_t suspended = uxSchedulerSuspended[xPortGetCoreID()];
                ^~~~~~~~~
[625/1518] Generating nanopb/generator/proto/nanopb_pb2.py
FAILED: esp-idf/spotify/cspot/nanopb/generator/proto/nanopb_pb2.py
cd /project/build/esp-idf/spotify/cspot && /project/components/spotify/cspot/bell/external/nanopb/generator/protoc -I/project/build/esp-idf/spotify/cspot/nanopb/generator/proto --python_out=/project/build/esp-idf/spotify/cspot/nanopb/generator/proto /project/build/esp-idf/spotify/cspot/nanopb/generator/proto/nanopb.proto
/usr/bin/env: ‘python3\r’: No such file or directory
[630/1518] Building C object esp-idf/esp_serial_slave_link/CMakeFiles/__idf_esp_serial_slave_link.dir/essl_spi.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
root@8a4f6eddc373:/project#
```

Tried all-sorts (fresh clone; update docker v43 and fresh v4-master load; messing with bell options, read and re-read instructions!)  but to no avail.

sle118, 2023-04-22:
There's really no reason for this not to work with the 4.3 docker container, as this is what we use in the build system. Perhaps there's something wrong with some of the files in your local repo. For example Windows line termination instead of Linux style.

wizmo2, 2023-04-22:
🚀 Thanks for the 💡 

Deleting the build folder, then in the docker
```
apt install dos2unix
find components -type f -print0 | xargs -0 dos2unix --
idf.py build
````

Bit brute force, but it did the trick.

closed 2023-04-22 by wizmo2