sle118/squeezelite-esp32 #247: Need advice: communicating with another ESP32 board

AndreasBVB, 2023-04-24:
Hi all, I managed to squeeze a Squeezelite WROVER and another ESP32 board into one speaker box, with the Squeezelite streaming music and ESP32 controlling the speaker buttons. Wondering if it is possible to setup a two-way direct communication between two boards (WIFI independent):
 - Squeezelite sharing streaming status: playing/paused, volume, etc...
 - ESP32 sending commands: play, previous, next, pause, volume up/down, etc...
Kindly point to a good direction so that I can search for more. Thanks!

sle118, 2023-04-25:
Why separate button controls to a different ESP32? This project includes everything to play music as well as to handle hardware buttons. If you run out of gpio's you can even use an io expander.

AndreasBVB, 2023-04-25:
> Why separate button controls to a different ESP32? This project includes everything to play music as well as to handle hardware buttons. If you run out of gpio's you can even use an io expander.

Hi sorry for the confusion. I meant controlling the speaker buttons (as opposed to Squeezelite playback "buttons"). The extra ESP32 simulates factory defined button presses, hijacks the physical button presses, decodes LED status lights through ADC and also included a PIR sensor. Some involving logic is added to "smartify" the speaker. AFAIK the Squeezelite ESP32 project is not the right place to deal with all these complexities.

However, since the speaker itself has become smarter, I'm wondering if some communication between the ESP32 and the Squeezelite can happen to improve the experience.

closed 2023-05-06 by sle118

AndreasBVB, 2023-05-06:
Hi thank you for your information and sorry I forgot to follow up.

I'll look into the Squeezelite32 buttons first and simulate button clicks from the ESP32. I'll keep the code merge (would be ESPHome+Squeezelite32) as an ambitious long term goal.