sle118/squeezelite-esp32 #251: Compatibility with the muVox board (Merus MA12070P I2C amplifier)

agdamsbo, 2023-05-07:
Hi! I have Squeezelite-esp32 installed in on a board with a cheap dac and love it. With Music Assistant in Home Assistant I have whole house synchronized playback (bye bye Apple and AirPlay2!). I love that!

I have been following the [muVox project](https://github.com/muvox-io/muVox/tree/Rev0.92). Do you expect the Squeezelite-esp32-firmware would support this board? I realise they do not have a final board yet to fully test, but I was hoping to be able to order the board when completed, and I would love to have to option to choose between Squeezelite-esp32 and the announced Euphonium.

I really appreciate your work with this project!

sle118, 2023-05-08:
It has all the hardware needed, so yes it will be possible to run our firmware on that board. @philippe44 has actually implemented @feelfreelinux 's cspot, so we can definitely say that the open source music streaming community is pretty small.

I don't have one of these to test, and AFAIK @philippe44 neither so we can't pre configure it as a new target unless someone sends a prototype or something. Short of a build target, you can configure the board yourself in software, no compiling needed.

agdamsbo, 2023-05-08:
Thank you for the prompt answer. I did notice @philippe44 in the discussion, so my guess was that you were engaged in some way.

I think it it very interesting to see these different high quality boards, even though it is a small community. I saw you mentioned a new board, @sle118 on slimdevices? Anything new? :)

I will reach out, if I get around to ordering a board, as I expect to be ordering a couple of boards.

sle118, 2023-05-08:
@agdamsbo I will try to make the second test version this month; real life™ actually happened at a faster pace than anticipated and then COVID stuck last March or so and that stalled the project until now.

The target use case for my board is different: a wired music steamer with simultaneous analog/digital output and analog input. It supports all the basic features of the project, including a built in I/O expander and a dedicated screen connector for those wanting to have a display with all the supported navigation buttons. As for the analog input, I'm thinking of using it for local playback support (e.g. aux in) via the audio chip's internal mix, but also as a way to inject an audio stream back into LMS. so if you have ever wanted to stream your vinyl collection across your home, then this will be it.

From there, I'll see what the cost of each board looks like and check if there's interest to sell it to a wider audience; my main objective here is to replace a few old Logitech duets which are most likely going to fail at some point. Logitech's engineers did an excellent Job at hardware design, though, as they are still alive after so many years of being plugged 24h/7

agdamsbo, 2023-05-08:
@sle118 the use case with streaming vinyls is really nice and just what I was considering. Though it goes a little against the analogue feeling. I would love it. I am very much looking forward to updates on the project, but perfectly recognize the difficulties with time!
I don't have first hand experience with the Logitech products, but my impression is that they also did right in "handing over" source code to the community. 
I find many older speakers of good quality in secondhand stores and wanted to convert such to "modern" speakers.

sle118, 2023-05-09:
@philippe44 has, very early on when we started porting squeezelite to the esp32, created a nifty amplifier called SqueezeAmp, which several users have bought to do exactly that: modernize old speakers. I actually own several boards, some of which are in a special reserve, and some of which I offered to relative and friends.
https://forums.slimdevices.com/forum/user-forums/3rd-party-hardware/108002-announce-the-squeezeamp-next-generation-squeezeboxen-amplifier-receiver

He also made it open source:
https://github.com/philippe44/SqueezeAMP

I screwed two Bose 141 series together and slapped a battery pack and SqueezeAmp in the case with a display+knob on the front grill. These follow me everywhere I go in the summer and make a fantastic and capable Bluetooth speaker.

Other companies like PolyVection and raspi audio have made their own hardware based on our firmware, targeting various use case and price points.

agdamsbo, 2023-05-09:
I actually tried to order a couple of SqueezeAMPs from JLCPCB (I like their interface for populating boards over PCBWay), but ran into some formatting problems on the BOM-list. I will try to figure it out today. That is also an amazing project, and impressive work to also share as open source!
The thing about muVox, that caught my interest, was the seemingly cheaper but more capable amplifier, and the fact that only one side is populated, making it much cheaper to get manufactured. I dont have the skills or equipment to do that myself. Or the time to learn at the moment.
Kudos on your work!

sle118, 2023-05-09:
I also like JLCPCB and how well they are integrated with the parts selection. Dual side will definitely cost you more, but the beauty of this is size. It is so small that it fits in the palm of your hand. The amplifier is also very lean in power usage, which is good for running on batteries. 

Perhaps @philippe44 still have some units he could share... not sure.

agdamsbo, 2023-05-13:
True. The size really is important. I would be very interested in a board or two if available @philippe44

philippe44, 2023-08-27:
Just before I close that, did you get the boards?

sle118, 2023-09-13:
@agdamsbo, @philippe44 base received a new batch (probably the last to ever be made) of SqueezeAmp. If you are interested, now would be a good time while supplies last.

sixtyfive, 2024-01-25:
Hi, not sure of off-/on-topic this is ... there's other boards using ther MA120x0x chips, too. I currently have one running off of a Raspberry Pi, since Infineon provide open source Linux drivers. There's also an ESP32-based project (https://github.com/jorgenkraghjakobsen/merus_bt) which I've been trying to get to work - without success so far, but mostly because I've been stubborn enough to use the current IDF.

Anyways, the only thing that sets these chips apart from, say, the PCM5102, is their I2S factory defaults, which are well-documented in [the datasheet](https://www.infineon.com/dgdl/Infineon-MA12070P-DataSheet-v01_01-EN.pdf?fileId=5546d46264a8de7e0164b761f2f261e4) (cf. pp. 18 & 19). People are also using them hooked up to DSPs configured with SigmaStudio just fine. In addition to the correct I²S settings, it's important to pull `/ENABLE` high and `/MUTE` low while setting things up, then let `/ENABLE` drop and pull `/MUTE` high as needed. The `/CLIP` and `/ERROR` pins are used by Infineons reference driver, but are not really needed (my board doesn't even expose them).

Looking at your code, I'm wondering if [dac_external.c](https://github.com/sle118/squeezelite-esp32/blob/master-v4.3/components/squeezelite/external/dac_external.c) would be the right file to copy and make adjustments to. Unfortunately I can't make heads and tails from it, so hopefully, by leaving this comment here with all the links in it, that can get the Issue one little step further? :)

sle118, 2024-05-08:
@philippe44 did a good job with the DAC control set implementation which likely has everything you need in configuration, meaning you don't have to create a driver for it.

sixtyfive, 2024-05-08:
Thanks for the response, sle118, will give it a go then!

agdamsbo, 2024-05-08:
Hi. Just as a heads up on the muVox project. The developer just announced on their matrix channel, that the DAC on the board will be changed to the TAS5827 as Infineon dropped production of the MA12070P. Curious to see when the project will actually launch a final (or first) board. Thanks for your responses and work for now.

sixtyfive, 2024-05-08:
Sad, but good to know.