sle118/squeezelite-esp32 #252: Make use of microphones

ei23fxg, 2023-05-08:
Are there plans for supporting microphones?
Since you can use ESPHome Media Player with Home Assistant as smart speaker (but without LMS) it would be nice to have the option to upstream microphone audio for that.

sle118, 2023-05-09:
There are currently no plans for this, as our main focus is on building a rock solid streaming firmware that enables multi room playback. 

For the sake of documenting, and in case Philippe wants to comment, could you provide more background on your use case?

ei23fxg, 2023-05-09:
Boards like the esp32 audio kit / ai-thinker have mics on board so they could be used to send voice commands to home assistant.

Home Assistant is currently building a local voice assistant plattform with whisper and piper.

Squeezelite speakers already can be integrated to play voice / audio from Home Assistant.
The mics sit on the board with no function, so i thought they could upstream audio, when a button is pressed for example.

sle118, 2023-05-09:
it's an interesting idea, but we have limited resources on top of which we've crammed a lot of features and there's little room left on flash, not counting memory. I will let this sit here for a couple of days and then move it to the discussions area for future reference. you never know...

ei23fxg, 2023-05-10:
> it's an interesting idea, but we have limited resources on top of which we've crammed a lot of features and there's little room left on flash, not counting memory. I will let this sit here for a couple of days and then move it to the discussions area for future reference. you never know... 

Yeah, just an idea.
I made a video doorbell with an ESP32.
OTA for example is a pain since space is so little.
Anyway. Even without this Feature, this project is great! Thanks a lot!

sle118, 2023-08-28:
For now, this is too much effort and there are still some bugs to address, so I'll go ahead and close this. Perhaps in the future we'll come back and implement some ADC logic but for now there's no bandwidth.

Thank you!

closed 2023-08-28 by sle118