sle118/squeezelite-esp32 #253: Amplifier Control Issue?

rjgrandy, 2023-05-10:
I am using an ESP32-WROVER and the setup works well. However there is some noise when idle through the speaker. My amplifier has a mute pin option which is mute=high logic level. I believe there is an option for amplifer control in this project where a GPIO can be set to high/low when the device is playing music and then set to the opposite when idle. I added the following line to "set_GPIO" in the NVS editor:

19=amp:0 

This makes the gpio19 on the ESP32 go high when the device is idle, but I can never get it to go low, even if I try to play music. 

Conversely if I use the option 19=amp:1, the gpio19 is always low no matter the state of the device.

Is there some option that I am not setting correctly?

sle118, 2023-05-11:
You should try to pull the gpio low or high depending on what you want to achieve, and depending on the configuration you set. The code doesn't use the ESP32 's internal pull up/down so you have to pull yourself.

rjgrandy, 2023-05-14:
I can't explain why, but I erased everything, reflashed with the newest firmware and now it seems to be working. I think the noise is from my cheap speaker power supply. I have a nice bench supply which doesn't seem to cause noise. However with this amp feature I can disable the amp and it goes silent when not used even with the cheap power supply. Works great!

closed 2023-05-14 by rjgrandy