sle118/squeezelite-esp32 #254: "squeezelite_main:389 => Option error: -d" while trying to connect to bt speaker

Papadoma, 2023-05-12:
Running a couple of bare esp32s.
One I had no issue installing squeezelite and connecting to a generic bluetooth speaker on startup.
The other won't connect. The only indication I get on the status page is 
`[00:05:39.440] squeezelite_main:389 => Option error: -d`

sle118, 2023-05-12:
In the credits tab on the UI, you will find a Google to activate the NVS tab. From there, going the autoexec command and share here. 

Note that playback to a Bluetooth device isn't the main intention of this firmware and depending on the issue might not get a fix

Papadoma, 2023-05-12:
It was
`squeezelite  -o bt -b 500:2000 -C 30 -d`

whereas the one working was
`squeezelite  -o bt  -b  500:2000 -C  30 -d  all=info `

so that extra all=info added allowed the esp32 to connect successfully. Thanks!

closed 2023-05-12 by Papadoma