sle118/squeezelite-esp32 #256: Failing to build docker image in Windows

mph070770, 2023-05-23:
I've used these instructions:

https://github.com/sle118/squeezelite-esp32/issues/196#issuecomment-1296846638

and tried to build the docker image, without luck.  First issue was the same as experienced by wizmo2 here:

https://github.com/sle118/squeezelite-esp32/issues/246#issuecomment-1518891457

`FAILED: esp-idf/spotify/cspot/nanopb/generator/proto/nanopb_pb2.py
cd /project/squeezelite-esp32/build/esp-idf/spotify/cspot && /project/squeezelite-esp32/components/spotify/cspot/bell/external/nanopb/generator/protoc -I/project/squeezelite-esp32/build/esp-idf/spotify/cspot/nanopb/generator/proto --python_out=/project/squeezelite-esp32/build/esp-idf/spotify/cspot/nanopb/generator/proto /project/squeezelite-esp32/build/esp-idf/spotify/cspot/nanopb/generator/proto/nanopb.proto
/usr/bin/env: ‘python3\r’: No such file or directory`

so I did the following, as advised by wizmo2:

`apt install dos2unix`

`rm build -r`

`find components/spotify/cspot -type f -print0 | xargs -0 dos2unix --
`

`idf.py build
`

now it fails with the following:

`FAILED: esp-idf/_override/CMakeFiles/__idf__override.dir/esp32/i2s.c.obj
ccache /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DHIERARCHICAL_STATES=1 -DMODEL_NAME=SqueezeESP32 -DSTATE_MACHINE_LOGGER=1 -Iconfig -I/opt/esp/idf/components/driver -I/opt/esp/idf/components/driver/include/driver -I/opt/esp/idf/components/newlib/platform_include -I/opt/esp/idf/components/freertos/include -I/opt/esp/idf/components/freertos/port/xtensa/include -I/opt/esp/idf/components/esp_hw_support/include -I/opt/esp/idf/components/esp_hw_support/port/esp32/. -I/opt/esp/idf/components/heap/include -I/opt/esp/idf/components/log/include -I/opt/esp/idf/components/lwip/include/apps -I/opt/esp/idf/components/lwip/include/apps/sntp -I/opt/esp/idf/components/lwip/lwip/src/include -I/opt/esp/idf/components/lwip/port/esp32/include -I/opt/esp/idf/components/lwip/port/esp32/include/arch -I/opt/esp/idf/components/soc/include -I/opt/esp/idf/components/soc/esp32/. -I/opt/esp/idf/components/soc/esp32/include -I/opt/esp/idf/components/hal/esp32/include -I/opt/esp/idf/components/hal/include -I/opt/esp/idf/components/esp_rom/include -I/opt/esp/idf/components/esp_rom/esp32 -I/opt/esp/idf/components/esp_rom/include/esp32 -I/opt/esp/idf/components/esp_common/include -I/opt/esp/idf/components/esp_system/include -I/opt/esp/idf/components/esp32/include -I/opt/esp/idf/components/driver/include -I/opt/esp/idf/components/driver/esp32/include -I/opt/esp/idf/components/esp_ringbuf/include -I/opt/esp/idf/components/efuse/include -I/opt/esp/idf/components/efuse/esp32/include -I/opt/esp/idf/components/xtensa/include -I/opt/esp/idf/components/xtensa/esp32/include -I/opt/esp/idf/components/espcoredump/include -I/opt/esp/idf/components/esp_timer/include -I/opt/esp/idf/components/esp_ipc/include -I/opt/esp/idf/components/esp_pm/include -I/opt/esp/idf/components/vfs/include -I/opt/esp/idf/components/esp_wifi/include -I/opt/esp/idf/components/esp_wifi/esp32/include -I/opt/esp/idf/components/esp_event/include -I/opt/esp/idf/components/esp_netif/include -I/opt/esp/idf/components/esp_eth/include -I/opt/esp/idf/components/tcpip_adapter/include -I/opt/esp/idf/components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Os -freorder-blocks -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3.2-dirty\" -DESP_PLATFORM -MD -MT esp-idf/_override/CMakeFiles/__idf__override.dir/esp32/i2s.c.obj -MF esp-idf/_override/CMakeFiles/__idf__override.dir/esp32/i2s.c.obj.d -o esp-idf/_override/CMakeFiles/__idf__override.dir/esp32/i2s.c.obj   -c ../components/_override/esp32/i2s.c
../components/_override/esp32/i2s.c:41:10: fatal error: soc/chip_revision.h: No such file or directory
 #include "soc/chip_revision.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[1366/1515] Building CXX object esp-idf/spotify/cspot/bell/CMakeFiles/bell.dir/main/platform/esp/MDNSService.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

can anyone please help?  I've now tried docker build and from scratch with IDF in Windows, and failed for both :-(

sle118, 2023-05-24:
To get some help, you will have to provide more details. Which branch are you trying to build, and which docker image are you using?

mph070770, 2023-05-24:
Hi @sle118 .  I used the same method as Gerhardt, detailed in the post I linked to, copied below:

Installed docker on windows and pulled with:
`sle118/squeezelite-esp32-idfv43`
Clone the repository in git bash with:
`git clone --recursive -b master-v4.3 https://github.com/sle118/squeezelite-esp32.git`
Copied the I2S-4MFlash-sdkconfig.defaults file from build-scrips folder to the root folder and renamed it to sdkconfig thus overwriting the existing file.
Open a CMD consolse in the respective code folder and run`
docker run --rm -v %cd%:/project -w /project -it sle118/squeezelite-esp32-idfv43
`Then start the config with:`
idf.py menuconfig`
Then i can build with:
`idf.py build`

The idf version in docker is:
```
root@8bc84257bc63:/project# idf.py --version
ESP-IDF v4.3.2-dirty
```

sle118, 2023-05-25:
As with everything else, the code base evolves and we sometimes move forward with the IDF release. I'll try to update the readme, when I manage to get a bit of spare time. In the meantime, the source of truth is here:
https://github.com/sle118/squeezelite-esp32/blob/79164da5f6f3f507182d82500abe7e54bd06cacd/.github/workflows/Platform_build.yml#L21

mph070770, 2023-05-29:
Thanks for your help @sle118.  I've built the image - the key change was the version of the docker file - from v43 to v435.  For anyone interested, this is what I did:

`docker pull sle118/squeezelite-esp32-idfv435
`
`git clone --recursive -b master-v4.3 https://github.com/sle118/squeezelite-esp32.git
`
`cd squeezelite-esp32
`
### I have the same issue as https://github.com/sle118/squeezelite-esp32/issues/246#issuecomment-1518891457 so needed to fix the line termination:

```
apt install dos2unix
rm build -r
find components/spotify/cspot -type f -print0 | xargs -0 dos2unix -- 
```

It's then ready to build:

`docker run --rm -v %cd%:/project -w /project -it sle118/squeezelite-esp32-idfv435
`
`idf.py menuconfig
`
`idf.py build
`

This should give you a built image.

then, to allow flashing from docker, in a CLI window, use esptool:

`C:\Espressif\docker\esptool>python esp_rfc2217_server.py -v -p 4800 COM5`

Then from another CLI:

`docker run --rm -v %cd%:/project -w /project  -it sle118/squeezelite-esp32-idfv435 espressif/idf idf.py --port rfc2217://host.docker.internal:4800?ign_set_control flash monitor`

closed 2023-05-29 by mph070770