sle118/squeezelite-esp32 #257: Reconnect to WiFi

BaukeFahner, 2023-05-24:
When my ap restarts, or when board loses range temporarily, the local ap mode is kickestarted and it does not reconnect to the WiFi has it becomes available. I can’t seem to disable or change this behavior. 

It is most probably a human error on my side but is there a way to influence this behavior? Or could it be dom characteristic of the board I’m using that I missed?

I’m using an aithinker audio kit with squeezelite 4.3.

sle118, 2023-05-25:
What is the actual version string you are getting at the bottom of the web UI?

sle118, 2023-05-25:
There's a fix in progress, implemented here: 
https://github.com/sle118/squeezelite-esp32/blob/1390e258db5312ef29ef43811a25d2531d09e288/components/wifi-manager/network_manager_handlers.c

But that branch is not quite ready to be merged into our main branch. I'll see if I can partially merge the fix.

jds11111, 2023-05-25:
Is this related to my problem? I have a new install on a WROVER board, and it cannot connect to my network.  Here is a snippet of the logs
```
I (115465) network_wifi: Wifi Connecting to <redacted>
I (118305) wifi:ap channel adjust o:1,0 n:5,0
I (118305) wifi:new:<5,0>, old:<1,0>, ap:<5,0>, sta:<0,0>, prof:1
I (118315) wifi:new:<5,0>, old:<5,0>, ap:<5,0>, sta:<5,0>, prof:1
I (118865) wifi:state: init -> auth (b0)
I (118895) wifi:state: auth -> assoc (0)
I (119895) wifi:state: assoc -> init (400)
I (119895) wifi:new:<5,0>, old:<5,0>, ap:<5,0>, sta:<5,0>, prof:1
W (119895) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: <redacted>, reason code: 4 (WIFI_REASON_ASSOC_EXPIRE)
```

sle118, 2023-05-26:
Can you at least connect once and then eventually it disconnected? Or you can't connect at all?

jds11111, 2023-05-27:
On the latest firmware I can connect fine, though it times out very frequently. On 1180, I never get any chance to connect. It never broadcasts an ssid for an AP.

sle118, 2023-05-28:
how do you power the board? Often times, wifi problems are caused by a drop in power when the wifi radio starts transmissing. There are numerous cases where this has been reported and differences between firmware releases can sometimes be explained by the sequence of event during the boot sequence (e.g. wifi being accessed while BT starts, etc).

jds11111, 2023-05-28:
Yeah, I noticed that the power from being plugged into my desktop was not sufficient, so I tried a couple of different ways, each with at least 1amp current. That didn''t work either though.  I just tried 3amps, and the wifi connectivity seems more stable, but am getting different errors now (`squeezelite_main:440 => Debug settings error: -d true` or `squeezelite exited with error code 2` and the device never shows up in my list of players, it keeps resetting my amp BLE PIN to 0000, though it does not use one.)  I need to read some documentation on how to set this up.

sle118, 2023-05-28:
This is unrelated. This error seems to be caused by an invalid command line. You can enable the NVS editor from the credits tab in the UI. From there, check your autoexec1 content and report here

jds11111, 2023-05-28:
`squeezelite  -o bt  -b  500:2000 -C  30 -d  true -R`

sle118, 2023-05-28:
Keep posting logs. Without them there's nothing much I can do

jds11111, 2023-05-28:
Sure. I am not sure where to get logs when it is timing out. But before it does, I also get this
```
squeezelite exited with error code 2
Correct command line and reboot
```
Thanks for the help!

sle118, 2023-05-29:
you should connect to a USB cable to get the logs from the serial port. You can the use the installer link which has a console. https://sle118.github.io/squeezelite-esp32-installer/

jds11111, 2023-05-29:
Here is the log: [https://pastebin.com/6uRNEpn3](https://pastebin.com/6uRNEpn3)

jds11111, 2023-05-29:
Several minutes after this, it did show up as a player on the LMS server. It streams a bit, but stutters frequently. I don't see anything new showing up on the telnet logs.

sle118, 2023-05-29:
In order to troubleshoot the issue, *complete* logs are needed and because the problem we are trying to figure out is related to network, you can't get all lines right now, as your system is configured to redirect its logs to telnet only. So ideally you should change the following NVS setting `telnet_enable = N`

jds11111, 2023-05-30:
Thanks for your patience. That was extremely helpful, since it gives more details, and could see what was happening. It seems like part of the problem was my amp. If the squeezelite connected to it once, and had problems, it didn't unpair correctly. After changing the resampling to linear interpolation, there is a procedure that seems pretty reliable. If I unpower the amp to reset any BLE pairing, and then restart the ESP3, it seems to pair and stream pretty reliably. If the ESP32 does not pair with the amp, it sometimes struggles with wifi and connecting to the LMS server. I think that bluetooth and wifi share the same antenna, so don't know if this has something to do with it.  Thanks for your help!

sle118, 2023-05-31:
Also worth noting that output to Bluetooth is very hacky and likely to fail. We don't put lots of efforts in there and may decide to drop some day.

closed 2023-05-31 by sle118

BaukF, 2023-05-31:
I’m terribly sorry, I posted the issue and never returned. I’ll read through the comments and see if given suggestions provide a solution.

jds11111, 2023-05-31:
> Also worth noting that output to Bluetooth is very hacky and likely to fail. We don't put lots of efforts in there and may decide to drop some day.

That makes me sad, but I understand if none of the coders is using it. If anyone does pick it up and wants a tester, I am happy to help.