sle118/squeezelite-esp32 #259: Problems with WEB GUI

dkotte, 2023-05-29:
Hello everyone,
After installing the latest I2S 16-bit version via the Squeezelite ESP32 installer at the AI Thinker Audio Kit A1S , I only got a few bits and pieces of the GUI. I can connect to WiFi and only see snippets of the GUI.
I was able to change the WiFi from the CLI. But even in the "new" network, the GUI is still incomplete.
Tested with different browsers and different devices.
What can I do?
[image]

sle118, 2023-05-30:
What does the status line say about the version you are running? If you are running Windows, try to do a full refresh or the page, pressing ctrl-f5 (I am assuming you run Chrome).

If this fails, then open the developer menu and look at the console for errors

dkotte, 2023-05-30:
I've updated the screenshot. Please check.
No Message within status-line. I am unable to switch to another menu-point.
I've tried ctrl-f5, normal/private mode, other browsers, other devices. Same picture.
At the serial console, i have some errors & warnings... sometimes.
[image]

dkotte, 2023-05-31:
I am able to flash an "old" version from one of this web-installers:
https://paulusschoutsen.nl/squeezelite-esp32-install/
https://raspiaudio.github.io/
Afterwards i am able to use the gui and configure the system to play music from LMS (without sound output)
I update the firmware to the current version from GUI ... and the system was back ... but not within the recovery-partition.
After I flash all parts from new version with esp-flash tool, the gui is defect within both partitions.

Dev-tools only show 200 - successful loaded.
No errors.
... but the page-load-spinner spins indefinitely, like the webpage is not loaded completely

sle118, 2023-06-04:
The ES8388 has been supported since the beginning via the generic I2S build. There isn't a dedicated target as such like the SqueezeAmp. It is possible to send initialization commands through the dac_controlset NVS parameter if needed, but that DAC should work without init.

Since I am unable to replicate the issue, I will close this after a couple of days

dkotte, 2023-06-04:
Found the problem,
There must be parts on the webpage that are blocked by the virus/endpoint protection and cannot be rendered by the old IPad/Safari.
If I switch off Sophos Endpoint Protection on the PC, the website is displayed correctly. 
The old IPad still displays the page incorrectly.

That's enough for configuration.

Thanks a lot. Great project.

closed 2023-06-04 by dkotte