sle118/squeezelite-esp32 #260: Problem

VaBanck, 2023-05-30:
When the track is switched, the device reboots.[image]

sle118, 2023-05-31:
There's really nothing I can do to help unless you provide more information or at least a full log. To do so, start by disabling telnet so all the logs are visible in the log window. And please... No screen shot; they cannot be indexed for searching and make it impossible to search. Get the logs, copy paste them and describe exactly what your situation is. There's an infinite way to use the system and we just can't guess.

VaBanck, 2023-05-31:
> There's really nothing I can do to help unless you provide more information or at least a full log. To do so, start by disabling telnet so all the logs are visible in the log window. And please... [...]

VaBanck, 2023-05-31:
> There's really nothing I can do to help unless you provide more information or at least a full log. To do so, start by disabling telnet so all the logs are visible in the log window. And please... [...]

[00:00:58.630] discover_server:828 sending discovery
[00:01:01.549] rtsp_thread:393 got RTSP connection 54
[00:01:03.630] discover_server:828 sending discovery
[00:01:08.630] discover_server:828 sending discovery
[00:01:08.938] handle_rtsp:447 [0x3fabead8]: challenge perBDI9b5SplknyE1ZqQug==
[00:01:09.222] handle_rtsp:440 [0x3fabead8]: received ANNOUNCE
[00:01:09.496] search_remote:739 starting remote search
[00:01:09.497] handle_rtsp:659 [0x3fabead8]: responding:
RTSP/1.0 200 OK

Audio-Jack-Status: connected; type=analog

CSeq: 4

[00:01:09.530] handle_rtsp:440 [0x3fabead8]: received SETUP
[00:01:09.531] raop_sink_cmd_handler:280 resizing buffer 423360
[00:01:09.534] buffer_alloc:386 allocated 1153 buffers (min=375) from buffer of 1624640 bytes
[00:01:09.545] output_thread_i2s:463 Output state is 0
[00:01:09.548] output_thread_i2s:529 Restarting I2S.
[00:01:09.552] handle_rtsp:659 [0x3fabead8]: responding:
RTSP/1.0 200 OK

Transport: RTP/AVP/UDP;unicast;mode=record;control_port=56280;timing_port=56281;server_port=56279

Session: DEADBEEF

Audio-Jack-Status: connected; type=analog

CSeq: 5

[00:01:09.553] equalizer_open:71 equalizer initialized 0
[00:01:09.591] handle_rtsp:440 [0x3fabead8]: received RECORD
[00:01:09.592] rtp_record:373 [0x3fac2d54]: record 0 0
[00:01:09.593] raop_sink_cmd_handler:284 Stream
[00:01:09.605] handle_rtsp:659 [0x3fabead8]: responding:
RTSP/1.0 200 OK

Audio-Jack-Status: connected; type=analog

CSeq: 6

[00:01:09.637] handle_rtsp:440 [0x3fabead8]: received SET_PARAMETER
[00:01:09.639] handle_rtsp:609 [0x3fabead8]: SET PARAMETER volume -20.000000
[00:01:09.640] raop_sink_cmd_handler:317 Volume[0..1] 0.3333
[00:01:09.652] handle_rtsp:659 [0x3fabead8]: responding:
RTSP/1.0 200 OK

Audio-Jack-Status: connected; type=analog

CSeq: 7

[00:01:09.659] rtp_thread_func:708 [0x3fac2d54]: 1st sync packet received
[00:01:09.685] raop_sink_cmd_handler:305 Play
[00:01:09.686] raop_sink_cmd_handler:310 Starting at 39912 (in 2014 ms)
[00:01:09.711] handle_rtsp:440 [0x3fabead8]: received FLUSH
[00:01:09.713] rtp_flush:347 [0x3fac2d54]: FLUSH ignored as same as RECORD (20305 - 1216411328)
[00:01:09. 22] output_thread_i2s:463 Output state is 5
[00:01:09.724] rtp_flush:356 [0x3fac2d54]: flush 20305 1216411328
[00:01:09.727] handle_rtsp:659 [0x3fabead8]: responding:
RTSP/1.0 200 OK

Audio-Jack-Status: connected; type=analog

CSeq: 8

[00:01:10.819] handle_rtsp:440 [0x3fabead8]: received SET_PARAMETER
[00:01:10.821] handle_rtsp:638 [0x3fabead8]: received JPEG image of 27653 bytes
[00:01:10.833] handle_rtsp:659 [0x3fabead8]: responding:
[... 150 lines cut ...]
I (3036) wifi:Init static rx buffer size: 1600
I (3046) wifi:Init static rx buffer num: 12
I (3056) wifi:Init dynamic rx buffer num: 40
[0;32mI (3056) wifi_init: tcpip mbox: 32[0m
[0;32mI (3056) wifi_init: udp mbox: 32[0m
[0;32mI (3066) wifi_init: tcp mbox: 32[0m
[0;32mI (3066) wifi_init: tcp tx win: 8192[0m
[0;32mI (3066) wifi_init: tcp rx win: 32768[0m
[0;32mI (3076) wifi_init: tcp mss: 1440[0m
[0;32mI (3076) wifi_init: WiFi/LWIP prefer SPIRAM[0m
[0;32mI (3076) network_wifi: Starting wifi[0m
[0;32mI (3086) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07[0m
I (3176) wifi:mode : sta (c0:49:ef:93:38:40)
I (3176) wifi:enable tsf

[0;32mI (3186) network_handlers: Existing wifi config found. Attempting to connect.[0m
[0;32mI (3196) network: Starting DHCP client[0m
[0;32mI (3196) network_wifi: Wifi Connecting to Home-2...[0m
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

To automatically execute lines at startup:
	Set NVS variable autoexec (U8) = 1 to enable, 0 to disable automatic execution.
	Set NVS variable autoexec[1~9] (string)to a command that should be executed automatically

VaBanck, 2023-05-31:
iPhone 11, airplay

philippe44, 2023-06-01:
Can you identify with which version the problem started? Use a version prior to mid-March

VaBanck, 2023-06-01:
> Can you identify with which version the problem started? Use a version prior to mid-March 

2.1288-16#v4.3#12S-4MFlash# master-v4.3, works well before this version, installed version 1271

philippe44, 2023-06-01:
Sorry this is a bit confusing. 1288 works or not?

VaBanck, 2023-06-02:
1288 doesn't work

philippe44, 2023-06-02:
Now, I just had an idea and pushed a change. It will not be an official build, so please get the latest builds artifacts (under Actions) and do a local update

VaBanck, 2023-06-02:
> Now, I just had an idea and pushed a change. It will not be an official build, so please get the latest builds artifacts (under Actions) and do a local update

There is no display and encoder, only i2s dac. Installed the latest version (local files) did not help

VaBanck, 2023-06-02:
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "HomePod",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "HomePod",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "HomePod",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n squeezelite-933840",
  "bat_config": "",
  "bt_name": "HomePod-bl",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"deviceName\":\"HomePod\",\"bitrate\":320}",
  "dac_config": "model=I2S,bck=26,ws=25,do=32",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "Y",
  "enable_bt_sink": "N",
  "enable_cspot": "N",
  "equalizer": "0,0,0,0,0,0,0,0,0,0",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "HomePod",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "lms_ctrls_raw": "n",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "https://github.com/sle118/squeezelite-esp32/releases",
  "rotary_config": "",
  "set_GPIO": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "N",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "N"
}

philippe44, 2023-06-02:
Crap - so I've udone some mods of 1288 to see. That should be building while I'm writing this, under the "enable DMPA_FULL" tag. Let me know if this changes something because it really seems to be linked to receiving some metadata but I'm shooting in the dark as I can't reproduce and I can't figure out from re-reading the code, so it's trial and error

VaBanck, 2023-06-02:
> Crap - so I've undone some mods of 1288 to see. That should be building while I'm writing this, under the "enable DMPA_FULL" tag. Let me know if this changes something because it really seems to be linked to receiving some metadata but I'm shooting in the dark as I can't reproduce and I can't figure out from re-reading the code, so it's trial and error

Do I need to download the latest version from the repository And install it manually, am I doing the right thing?

philippe44, 2023-06-02:
You have to take this file https://github.com/sle118/squeezelite-esp32/suites/13330870751/artifacts/728549108, unzip-it and then in the WebUI of your SqueezeESP32, "Update" tab, select "Local File" and pick the "squeezelite.bin" file that was just extracted from the above mentioned archive.

NB: The link I've picked is the build I've made earlier today, the latest one was still under construct

VaBanck, 2023-06-02:
Now it doesn't work at all.

philippe44, 2023-06-02:
> Now it doesn't work at all.
>
What do you mean? I just downloaded the SqueezeAMP version on my board and it works as usual. Did you select the squeezelite.bin file (not the full zip)?

Also does it crash when you use iTunes?

VaBanck, 2023-06-02:
I'll leave version 1271, it works, maybe it will be better in the future, tomorrow I'll make a log, yes, I installed squeezelit.bin

I'm trying the radio via the Internet

sle118, 2023-06-06:
@VaBanck there was a recent change done around the vorbis component (https://github.com/sle118/squeezelite-esp32/commit/8b764c0c2d7502f4f76ab23293d1e6ce9ebcc898) you might want to give it another shot by installing the following binary from our automated build (as the changes were not yet made available through a release): https://github.com/sle118/squeezelite-esp32/suites/13388676178/artifacts/732857373

VaBanck, 2023-06-06:
> @VaBanck there was a recent change done around the vorbis component (https://github.com/sle118/squeezelite-esp32/commit/8b764c0c2d7502f4f76ab23293d1e6ce9ebcc898) you might want to give it another [...]

This version doesn't work for me, I'll try another esp, maybe it will work

VaBanck, 2023-06-07:
> zip)?

Excuse me please, I installed the wrong version, version 1319 works,[image]

VaBanck, 2023-06-07:
https://github.com/sle118/squeezelite-esp32/suites/13337862394/artifacts/729093462

philippe44, 2023-06-07:
I'm confused - can you summarize the situation?

philippe44, 2023-06-07:
This is more than weird because I've not changed anything compared to the previous version where you told me it would not start. Are you 100000% sure you took the artifacts from the previous version? If not, I'll undo the metadata change and to an official release so that we can compare

philippe44, 2023-06-07:
ok, I'll consider it closed then

closed 2023-06-07 by philippe44