sle118/squeezelite-esp32 #261: Opus playback ends before reaching the end

robho, 2023-06-02:
When playing opus files they are cut short during playback.

I've backtracked this to release 1299 so I think the problem was introduced in e0e7e718ba7ab242352c5eff3a4f1eee1cf67896.

I'm attaching a file which can be used to trigger the problem. It always ends at 4:50 (15 seconds before its length).

[1-03 Reyn Ouwehand - Last Ninja 3 (intro).zip](https://github.com/sle118/squeezelite-esp32/files/11638574/1-03.Reyn.Ouwehand.-.Last.Ninja.3.intro.zip)

philippe44, 2023-06-02:
Please have a look at the version that is being built as I write that. Should fix the issue while still ending on mis-encoded opus files that don't have EoS market set.

philippe44, 2023-06-03:
@sle118 brought be automatic build back, so you should be able to try now

robho, 2023-06-03:
I don't find the new build. Is it necessary to rebuild / upload the release artefacts?

philippe44, 2023-06-03:
It should be under action, the one created when @sle118 fixed the node issue

sle118, 2023-06-04:
@robho  have a look here: https://github.com/sle118/squeezelite-esp32/actions/runs/5165765805

note that the latest build has some UI dependency changes which I have not been able to fully test.

robho, 2023-06-04:
I tested 1324 and it seems to work properly now. Thanks!

closed 2023-06-04 by robho