sle118/squeezelite-esp32 #262: Failing to build docker image in Linux/Ubuntu -- cspot/nanopb_pb2.py

jochen-rath, 2023-06-05:
Hello

I can not build the docker image in Linux. I get the same errors as described in this issue:
https://github.com/sle118/squeezelite-esp32/issues/246
But I am not building in Windows, but Ubuntu 22.04.2 LTS. I tried the solution mentioned in the issue, but it didn't work. I also tried the solution mentioned in this issue:
https://github.com/sle118/squeezelite-esp32/issues/256

`git clone --recursive https://github.com/sle118/squeezelite-esp32.git`
`docker pull sle118/squeezelite-esp32-idfv435`
`cd squeezelite-esp32 `
`find components/spotify/cspot -type f -print0 | xargs -0 dos2unix -- c`
`docker run -it -v $(pwd):/workspace/squeezelite-esp32 sle118/squeezelite-esp32-idfv435`

I tried also 
`git clone --recursive -b master-v4.3 https://github.com/sle118/squeezelite-esp32.git`
and I tried to clone the git repository in docker.

I have tried with and without the solutions. Even without a docker image. By every build it fails with the error:
`-- Configuring done`
`-- Generating done`
`-- Build files have been written to: /workspace/inDocker/squeezelite-esp32/build`
`[3/1545] cd /workspace/inDocker/squeezelite-esp32/build && /opt/esp/tools/cmake/.../bin/cmake -P /workspace/inDocker/squeezelite-esp32/generate_debug_scripts.cmake`
`Generating debug script files`
`[5/1545] Generating nanopb/generator/proto/nanopb_pb2.py`
`FAILED: esp-idf/spotify/cspot/nanopb/generator/proto/nanopb_pb2.py `

I do not need spotify. Can I build the program without it?

wizmo2, 2023-06-05:
I had the same issue.  For me it was a windows lf/cr return issue.  change your git setting for linux format when cloning.

sle118, 2023-06-05:
> Hello
> 
> I can not build the docker image in Linux. I get the same errors as described in this issue: #246 But I am not building in Windows, but Ubuntu 22.04.2 LTS. I tried the solution mentioned in the issue, but it didn't work. I also tried the solution mentioned in this issue: #256
> 
> `git clone --recursive https://github.com/sle118/squeezelite-esp32.git` `docker pull sle118/squeezelite-esp32-idfv435` `cd squeezelite-esp32 ` `find components/spotify/cspot -type f -print0 | xargs -0 dos2unix -- c` `docker run -it -v $(pwd):/workspace/squeezelite-esp32 sle118/squeezelite-esp32-idfv435`
> 
> I tried also `git clone --recursive -b master-v4.3 https://github.com/sle118/squeezelite-esp32.git` and I tried to clone the git repository in docker.
> 
> I also tried both docker version: `root@7e2d6f8ffc3b:/workspace/squeezelite-esp32# idf.py --version` `ESP-IDF v4.3.2-dirty` and `root@7553d85b37ea:/# idf.py --version` `ESP-IDF v4.3.5-dirty`
> 
> I have also tried without a docker image. But every build fails with the same error: `-- Configuring done` `-- Generating done` `-- Build files have been written to: /workspace/inDocker/squeezelite-esp32/build` `[3/1545] cd /workspace/inDocker/squeezelite-esp32/build && /opt/esp/tools/cmake/.../bin/cmake -P /workspace/inDocker/squeezelite-esp32/generate_debug_scripts.cmake` `Generating debug script files` `[5/1545] Generating nanopb/generator/proto/nanopb_pb2.py` `FAILED: esp-idf/spotify/cspot/nanopb/generator/proto/nanopb_pb2.py `
> 
> I do not need spotify. Can I build the program without it?

Can you please share more context (lines before and after) in your build log?

jochen-rath, 2023-06-05:
Thanks for your replies.
@wizmo2:
I will try it, when I have understood, how to change the settings.

@sle118 
This is my full output:
`jochen@lenovo:~/esp/squeezelite-esp32$ docker run -it -v `pwd`:/workspace/squeezelite-esp32 sle118/squeezelite-esp32-idfv43`
`Detecting the Python interpreter`
`Checking "python" ...`
`Python 3.8.10`
`"python" has been detected`
`Adding ESP-IDF tools to PATH...`
`Using Python interpreter in /opt/esp/python_env/idf4.3_py3.8_env/bin/python`
`Checking if Python packages are up to date...`
`Python requirements from /opt/esp/idf/requirements.txt are satisfied.`
`Added the following directories to PATH:`
`  /opt/esp/idf/components/esptool_py/esptool`
`  /opt/esp/idf/components/espcoredump`
`  /opt/esp/idf/components/partition_table`
`  /opt/esp/idf/components/app_update`
`Done! You can now compile ESP-IDF projects.`
`Go to the project directory and run:`
``
`  idf.py build`
``
`npm version is 9.6.4`
`node version is v16.19.1`
`To build the web application, run:`
`pushd components/wifi-manager/webapp/ && npm rebuild node-sass && npm run-script build && popd`
``
`To run size statistics, run:`
`puncover/runner.py --gcc_tools_base /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf- --elf ./build/recovery.elf --build_dir build --src_root .`
``
`root@2587c3f5afd2:/# cd workspace/squeezelite-esp32/`
`root@2587c3f5afd2:/workspace/squeezelite-esp32# idf.py build`
`Executing action: all (aliases: build)`
`Running cmake in directory /workspace/squeezelite-esp32/build`
`Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 /workspace/squeezelite-esp32"...`
`-- Found Git: /usr/bin/git (found version "2.25.1") `
`-- Component directory /opt/esp/idf/components/esp_phy does not contain a CMakeLists.txt file. No component will be added`
`-- Component directory /opt/esp/idf/components/ieee802154 does not contain a CMakeLists.txt file. No component will be added`
`-- Component directory /opt/esp/idf/components/openthread does not contain a CMakeLists.txt file. No component will be added`
`-- ccache will be used for faster recompilation`
`-- The C compiler identification is GNU 8.4.0`
`-- The CXX compiler identification is GNU 8.4.0`
`-- The ASM compiler identification is GNU`
`-- Found assembler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc`
`-- Check for working C compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc`
`-- Check for working C compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works`
`-- Detecting C compiler ABI info`
`-- Detecting C compiler ABI info - done`
`-- Detecting C compile features`
`-- Detecting C compile features - done`
`-- Check for working CXX compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++`
`-- Check for working CXX compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works`
`-- Detecting CXX compiler ABI info`
`-- Detecting CXX compiler ABI info - done`
`-- Detecting CXX compile features`
`-- Detecting CXX compile features - done`
`-- Building ESP-IDF components for target esp32`
`-- Project sdkconfig file /workspace/squeezelite-esp32/sdkconfig`
`Loading defaults file /workspace/squeezelite-esp32/sdkconfig.defaults...`
`-- Found PythonInterp: /opt/esp/python_env/idf4.3_py3.8_env/bin/python (found version "3.8.10") `
`-- Found Perl: /usr/bin/perl (found version "5.30.0") `
`-- App "recovery" version: local.500.cmake-master`
`-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.ld`
`-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.api.ld`
`-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld`
`-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld`
`-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld`
[... 23 lines cut ...]
`-- Generating done`
`-- Build files have been written to: /workspace/squeezelite-esp32/build`
`Running ninja in directory /workspace/squeezelite-esp32/build`
`Executing "ninja all"...`
`[3/1515] Generating nanopb/generator/proto/nanopb_pb2.py`
`FAILED: esp-idf/spotify/cspot/nanopb/generator/proto/nanopb_pb2.py `
`cd /workspace/squeezelite-esp32/build/esp-idf/spotify/cspot && /workspace/squeezelite-esp32/components/spotify/cspot/bell/external/nanopb/generator/protoc -I/workspace/squeezelite-esp32/build/esp-idf/spotify/cspot/nanopb/generator/proto --python_out=/workspace/squeezelite-esp32/build/esp-idf/spotify/cspot/nanopb/generator/proto /workspace/squeezelite-esp32/build/esp-idf/spotify/cspot/nanopb/generator/proto/nanopb.proto`
`/bin/sh: 1: /workspace/squeezelite-esp32/components/spotify/cspot/bell/external/nanopb/generator/protoc: Permission denied`
`[4/1515] cd /workspace/squeezelite-esp32/build && /opt/esp/tools/cmake/3.16.4/bin/cmake -P /workspace/squeezelite-esp32/generate_debug_scripts.cmake`
`Generating debug script files`
`[7/1515] Generating ../../partition_table/partition-table.bin`
`Partition table binary generated. Contents:`
`*******************************************************************************`
`# ESP-IDF Partition Table`
`# Name, Type, SubType, Offset, Size, Flags`
`nvs,data,nvs,0x9000,16K,`
`otadata,data,ota,0xd000,8K,`
`phy_init,data,phy,0xf000,4K,`
`recovery,app,factory,0x10000,1280K,`
`ota_0,app,ota_0,0x150000,2688K,`
`settings,data,nvs,0x3f0000,64K,`
`*******************************************************************************`
`ninja: build stopped: subcommand failed.`
`ninja failed with exit code 1`
`root@2587c3f5afd2:/workspace/squeezelite-esp32# `

sle118, 2023-06-05:
Here's your problem: ``` /workspace/squeezelite-esp32/components/spotify/cspot/bell/external/nanopb/generator/protoc: Permission denied```

you **have** to chmod the protoc executables inside of the interactive docker container, before calling idf.py build: 

```
  chmod +x ./components/spotify/cspot/bell/external/nanopb/generator/protoc-gen-nanopb
  chmod +x ./components/spotify/cspot/bell/external/nanopb/generator/*.py
  chmod +x ./components/spotify/cspot/bell/external/nanopb/generator/*.py2
  chmod +x ./components/spotify/cspot/bell/external/nanopb/generator/proto/*.py
```

jochen-rath, 2023-06-05:
Thank you very much. Now it is working. Additionally I needed to do:
`chmod +x ./components/spotify/cspot/bell/external/nanopb/generator/protoc`

closed 2023-06-05 by jochen-rath

sle118, 2023-06-05:
The following change will take care of resolving the issue going forward: https://github.com/sle118/squeezelite-esp32/commit/ed53fa21a73b7c3d8c8685101cbfad3374b5a3e0