sle118/squeezelite-esp32 #265: building on linux fails with error

zelea2, 2023-06-22:
I have the latest esp-idf and tools configured just fine.
`$ idf.py --version`
`ESP-IDF v5.2-dev-1128-g03d4fa2869-dirty`
I can compile all the examples in the esp-idf directories with the $ idf.py build` command.

`$ git branch`
`* master-v4.3`
`$ git pull`
`Already up to date.`

When I try to build or configure squeezelite-esp32 I first get a complaint about missing the `nghttp` component
so then I do 
`$ idf.py add-dependency nghttp`
`Created "/home/xxx/squeezelite-esp32/main/idf_component.yml"`
`Successfully added dependency "espressif/nghttp*" to component "main"`

If then I try to build or configure the firmware I get this error and I don't know how to solve this:
`-- Building ESP-IDF components for target esp32`
`Processing 2 dependencies:`
`[1/2] espressif/nghttp (1.52.0)`
`[2/2] idf (5.2.0)`
`CMake Error at /opt/esp/esp-idf/tools/cmake/build.cmake:268 (message):`
`  Failed to resolve component 'esp32'.`

sle118, 2023-06-24:
Which docker image did you use? Having a look at our platform build config is the best way to build right now.:
https://github.com/sle118/squeezelite-esp32/blob/master-v4.3/.github/workflows/Platform_build.yml#L21

and we are currently using this docker image:
sle118/squeezelite-esp32-idfv435

hope this helps. Out of curiosity, why build locally when there are already binaries available for pretty much all platforms (hard coded or via nvs config)?

zelea2, 2023-06-25:
I'll try again on Monday following your instructions. The reason I'm trying to build this is that I'm about to design a PCB (which is supposed to be easy to solder) based on ESP32-TTGO-V1.8, ES9023 and a W5500-Module. Schematic is attached and I've already configured the `sdkconfig` accordingly. 
[Squeeze-ESP32.pdf](https://github.com/sle118/squeezelite-esp32/files/11859379/Squeeze-ESP32.pdf)

sle118, 2023-06-25:
Looking forward to see your progress with designing your board. Note that you do not have to build a custom binary, as @philippe44 and myself have been working hard to make sure everything was software configurable including any init sequence that has to be sent to the DAC.

zelea2, 2023-06-27:
I've managed to compile it via docker and idfv435 instead of idfv43 (maybe you should change the README as well).
I know that I can configure it via NVS but it's more convenient to have it already configured at build time.

sle118, 2023-06-28:
Done with this https://github.com/sle118/squeezelite-esp32/commit/b38354420c16902f0d08cde275a91d27082f5e27. Closing this for now.

closed 2023-06-28 by sle118