sle118/squeezelite-esp32 #270: invalid header: 0xffffffff

KerbalFoundries, 2023-07-17:
Bit confused about this one. Board is ESP32-WROVER-DEV. Supposedly it meets the requirements for flash and PSRAM with the WROVER chipset. I've tried flashing via the web GUI tool and the ESP32 Flash Download Tool, both with the same result:

`I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (27) boot: compile time 22:28:30
I (27) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 1
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (88) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 recovery         factory app      00 00 00010000 00140000
I (103) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (110) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (118) boot: End of partition table
I (122) boot: Defaulting to factory image
I (127) boot_comm: chip revision: 3, min. application chip revision: 1
I (134) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=5309ch (340124)       map
I (255) esp_image: segment 1: paddr=000630c4 vaddr=3ffbdb60 size=04988h ( 18824)       load
I (262) esp_image: segment 2: paddr=00067a54 vaddr=40080000 size=085c4h ( 34244)       load
I (275) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=ce1e8h (844264)       map
I (553) esp_image: segment 4: paddr=0013e210 vaddr=400885c4 size=0c480h ( 50304)       load
I (573) esp_image: segment 5: paddr=0014a698 vaddr=400c0000 size=00068h (   104)       load
I (573) esp_image: segment 6: paddr=0014a708 vaddr=50000000 size=00010h (    16)       load
I (589) boot: Loaded app from partition at offset 0x10000
I (589) boot: Disabling RNG early entropy source...
I (601) psram: This chip is ESP32-D0WD
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46
`
Using the I2S-4MFlash-16 and tried a couple of revisions. I have two of these board, and both flash fine (and work) with a generic ESPHome config or setting up for the cameras they came with.

Any pointers please! 

Many thanks in advance.

sle118, 2023-07-17:
Post a picture of your board. There are lots of low quality products out there. Also be more specific with the flasher you actually used, as there are a few out there.

KerbalFoundries, 2023-07-17:
Thanks for the quick response. 
[image]

First try was this using the generid/I2S: https://sle118.github.io/squeezelite-esp32-installer/

Second was the Espressif tool from here: https://www.espressif.com/en/support/download/other-tools 
following instructions from: https://forums.slimdevices.com/forum/user-forums/3rd-party-hardware/109412-announce-squeezelite-esp32-dedicated-thread?112697-ANNOUNCE-Squeezelite-ESP32-(dedicated-thread)=

Wading through the thread, on slimdevices, I've _just_ uncovered some posts about this very board. Apologies if I've wasted your time - it didn't come up as I was searching for similar issues. I'll continue reading and report back. Hopefully it might help someone in the future regardless.

KerbalFoundries, 2023-07-18:
@dr3amr 

To follow up, I found a post in the thread over on slimdevices. https://forums.slimdevices.com/forum/user-forums/3rd-party-hardware/109412-announce-squeezelite-esp32-dedicated-thread?p=1627715#post1627715

Following those instructions and reflashing the bootloader with the stock bootloader.bin from espressiff did the trick and all is well.

I've got a UDA1334A hooked up running generic I2S. Took some fiddling, but my config is model=i2s,bck=27ws=26,do=25 for anyone that's interstested.

For Future Ref: This board appears to be a clone of the Freenove board: https://github.com/Freenove/Freenove_Ultimate_Starter_Kit_for_ESP32/blob/master/About_Fakes.pdf

I wish I'd known when buying... you live and learn.

Anyway, I hope that's of some use! If I can help with troubleshooting, reproducing the fault or feeding back extra info, @sle118 , please do shout. I have a spare board to pay with.