sle118/squeezelite-esp32 #273: Can't find device via Airplay or Spotify Connect

davidakr, 2023-07-21:
Hi,

first of all, thank you for such a great project!

I built squeezelite with the Docker container, flashed it to my ESP32-A1S module, and connected it to my home network. It shows up as Bluetooth but it's not being detected in Spotify (premium) or via airplay. Both are checked in the UI. Anything I can do to troubleshoot?

The apple and speaker icon in the top right corner disappear after a few seconds.

sle118, 2023-07-22:
This needs more details. Which Docker image, for example, and which branch? Also, did you enable the services in the Web UI?

davidakr, 2023-07-25:
I took the latest sle118/squeezelite-esp32-idfv4-master image to build and master-v4.3 branch. I used the flash command from the Readme. Yes, all services are enabled in the UI. It might also be our current home network configuration. Do both of them use MDNS for discovery? The device is on the same network as my phone.
Here is the output from telnet.
[telnetresult.txt](https://github.com/sle118/squeezelite-esp32/files/12164937/telnetresult.txt)

Also, the Bluetooth keeps disconnecting. I checked the logs there is nothing unusual besides the disconnect message.
I attached my current config.
[nvs_config_kitchen-speaker.txt](https://github.com/sle118/squeezelite-esp32/files/12164811/nvs_config_kitchen-speaker.txt)

What can I do to further debug the issues?

davidakr, 2023-07-25:
It was an MDNS issue in my network. The configuration didn't allow for device discovery. I mark this issue as resolved for anyone who has the same issue.

closed 2023-07-25 by davidakr