sle118/squeezelite-esp32 #275: Muse Luxe - Toggle Restarts Song

LindyBalboa, 2023-08-03:
First off, thanks for this awesome contribution.

I just got my Muse Luxe and flashed the firmware and added the button configuration from the readme via the NVS editor. It mostly works. One quirk is that if I use a long press on the play button to power off the LMS client and then do a short press to start it back up, the song restarts. I have realized that if I again use a long press to power the LMS client back on, it works as expected. Being able to start it back up with a short press is just an intuitive nicety. Is there any way to achieve that?

philippe44, 2023-08-03:
Restart you mean starts from the beginning?

LindyBalboa, 2023-08-03:
Correct. And for context I just don't have the Muse Luxe playing, I have another client (Moode on an RPi). The other continues to play as expected after the first long press (power off). On the following short press the Pi cuts out and then both start at the start of the current track.

philippe44, 2023-08-03:
Isn't it a LMS setting how you want to restart after power off? It also depends on the source. Some won't allow re-positionning

closed 2023-08-12 by philippe44