sle118/squeezelite-esp32 #277: SPDIF Config in Hardware Parameters not displayed

wizmo2, 2023-08-08:
NVS Editor shows `bck=33,ws=25,do=32`, but Hardware Parameters do not display any values in the SPDIF Config. 

I came across this when testing my PR.  I'm officially entering this one,  as it seems to have been an issue on all recent builds (I2C-4M and SqueezeAmp 16bits) since at least 1295.   Possible duplicate with #175.

_On a possibly separate issue, I am unable to set the NVS value using the Hardware tab with my local builds on 1331,  It is possible this is associated with my development environment, but none of the webapp functions that use 'commands.json' work._

sle118, 2023-08-11:
There is an issue with the most recent release so my advice for now is to roll back to 1306 until the isue has been addressed.  The root cause is a change that was introduced on the most recent ESP-IDF 4.3 (which we bumped to a more recent version recently) and that broke the console redirection that is used by both Telnet and the web UI. This is an area where @philippe44 will need to dive in, and this will be a stinky dive to say the least.

philippe44, 2023-08-11:
Now if you want cspot fixes and YT working with vorbis files, you need to chose the latest version. Pick your poison :-)

philippe44, 2023-08-12:
Should be fixed

sle118, 2023-08-12:
@philippe44 you're much to comfortable with the esp-idf now :grin:

sle118, 2023-08-12:
@wizmo2 thanks for reporting this. Please close the issue when you have confirmed this works!

closed 2023-08-17 by wizmo2