sle118/squeezelite-esp32 #280: ILI9341 display only display initialization

lelithium, 2023-08-14:
Hi !

My ILI9341 display works, but only during the initialization code. When I reset the device, the display flashes to reset, then displays SqueezeESP32, and that's it.

Display: ILI9341 2.8" SPI https://www.amazon.fr/dp/B07YTWRZGR
Board: ESP32 "audio kit" 2.2 (with ES8388 DAC) https://docs.ai-thinker.com/en/esp32-audio-kit

Connected through (omitting VCC, GND, LED):
|Screen pin|ESP GPIO|
|-|-|
|CS|0|
|Reset|23|
|DC|5|
|MOSI|18|
|SCK|13|
|MISO|22|

Note that MISO seems to not be needed to display the `SqueezeESP32` text.

My (current, see explanations below) NVS config for display/GPIO related things:

|Key|Value|
|-|-|
|`dac_config`|`model=ES8388,bck=27,ws=25,do=26,sda=33,scl=32,di=35,i2c=16`|
|`display_config`|`SPI:width=320,height=240,cs=0,reset=23,speed=1000000,driver=ILI9341`|
|`set_GPIO`|`21=amp,39=jack:0`|
|`spi_config`|`mosi=18,clk=13,dc=5,host=1,miso=22`|

I've tried different things:
- *Tried moving to different pins.* I've tried a bunch of variants, some which failed to display the `SqueezeESP32` initial text, most succeed but none get me further than that
- *Tried to disable LMS.* I figured that maybe my LMS setup was faulty, so I disabled it, and tried with direct Bluetooth connection instead. In the UART logs, I do see messages like `I (60164) display: playing SONGNAME`, but still `SqueezeESP32` on the screen
- *Tried to reset the screen manually.* To test if the board was sending SPI data, I manually put RESET to ground, which displayed a white screen, then put RESET back to its assigned pin. No updates from the white screen.
- *Tried to lower the SPI speed.* I set it at 1 000 000 so that it would be clear if the display took time to update. That way, when reset, the `SqueezeESP32` text takes some time to appear (i can see the display lines appearing one after the other), but still it only displays that, no updates after.
- *Tried to unset all other config related to GPIO.* When I began, I had a `actrls_config` config with buttons for playback, volume, etc, which I removed, to no affect.
- *Tried to boot in RECOVERY*. When booting in recovery, I get the `SqueezeESP32` text, then it is replaced by `RECOVERY`. No updates after that.

I think I might be missing something obvious here in the config (this is my first SqueezeLite32). I followed the pinout from this blog post first, which is a different display, but still SPI, and that works `https://ale.cx/ALEX/2021/01/esp32-audio-kit-with-squeezelite-esp32/`.

The fact that I can get something to display seems to point out that the display connection is, in fact, working. I guess there's something tying one of the pins after that to GND or VCC, which would prevent the screen from receiving its SPI data, but I have no clue where to start.

Full logs, from reset to Bluetooth connection + play

<details>
<summary>Unroll</summary>
```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (27) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (27) boot: compile time 05:15:25
I (27) boot: chip revision: v3.0
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (104) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
[... 180 lines cut ...]
I (11854) httpd_handlers: serving /status.json to peer 192.168.1.126 port 24240
[00:00:14.866] discover_server:828 sending discovery
[00:00:18.718] bt_sink_cmd_handler:170 Setting BT sample rate 44100
I (20134) BT_AV: Configure audio player 21-15-2-35
I (20134) BT_AV: Audio player configured, sample rate=44100
[00:00:19.866] discover_server:828 sending discovery
[00:00:24.866] discover_server:828 sending discovery
[00:00:25.858] bt_sink_cmd_handler:143 BT sink started
[00:00:25.859] display_bus_handler:383 Display bus command 0
[00:00:25.860] bt_sink_cmd_handler:155 BT play
[00:00:25.874] display_bus_handler:383 Display bus command 0
I (27294) display: playing Krack
[00:00:25.883] output_thread_i2s:498 Output state is 2
[00:00:25.885] output_thread_i2s:564 Restarting I2S.
[00:00:25.888] equalizer_open:71 equalizer initialized 0
[00:00:27.301] bt_sink_cmd_handler:166 BT pause, just silence
[00:00:29.866] discover_server:828 sending discovery
[00:00:30.373] bt_sink_cmd_handler:150 BT sink stopped
[00:00:30.374] display_bus_handler:383 Display bus command 1
```
</details>

Let me know if there's anything else I can do to help pinpoint the issue here ! I

Thanks for your hard work on this project

sle118, 2023-08-15:
Did you install the most recent version of the SqueezeESP plugin by @philippe44 on your LMS instance? The screen is independent from it during boot, and seeing something during that time means config is ok and wiring too, but it's the server that sends display data when connected.

lelithium, 2023-08-16:
Thanks for the quick answer @sle118 !

Updated to latest (0.362) didn't change the issue
With the LMS available, I see in the logs messages such as:
```
[00:00:09.968] sendSETD:431 sending dimension 320x240
[00:00:09.975] notify:228 notified server 192.168.1.145 hport 9000 cport 9090
I (11394) network_status: LMS IP: 192.168.1.145, hport: 9000, cport: 9090
[00:00:09.989] decode_flush:251 decode flush
[00:00:09.990] output_flush:446 flush output buffer
[00:00:09.991] decode_flush:251 decode flush
[00:00:09.992] output_flush:446 flush output buffer
[00:00:10.004] grfb_handler:743 brightness 4
[00:00:10.204] visu_handler:1232 Stopping visualizer
[00:00:10.208] handler:54 got equalizer 14
[00:00:10.212] visu_handler:1232 Stopping visualizer
I (13124) httpd_handlers: serving /messages.json to peer 192.168.1.126 port 48845
[00:00:18.899] decode_flush:251 decode flush
[00:00:18.900] output_flush:446 flush output buffer
[00:00:18.902] visu_handler:1232 Stopping visualizer
[00:00:19.071] grfs_handler:795 scroller current size 1280 (w:360)
[00:00:19.072] grfs_handler:795 scroller current size 2560 (w:360)
[00:00:19.074] grfs_handler:795 scroller current size 2712 (w:360)
[00:00:19.276] codec_open:283 codec open: 'm'
[00:00:19.279] stream_sock:508 connecting to 192.168.1.145:9000

[00:00:19.656] stream_thread:264 headers: len: 486

[00:00:20.681] output_thread_i2s:498 Output state is 1
[00:00:20.683] output_thread_i2s:564 Restarting I2S.
[00:00:20.685] equalizer_open:71 equalizer initialized 0
[00:00:20.740] mad_decode:247 setting track_start
[00:00:20.800] _output_frames:65 start buffer frames: 7535
[00:00:20.802] _output_frames:150 track start sample rate: 44100 replay_gain: 0
[00:00:20.803] output_thread_i2s:498 Output state is 2
[00:00:29.979] decode_flush:251 decode flush
[00:00:29.980] output_flush:446 flush output buffer
[00:00:29.995] output_thread_i2s:498 Output state is 0
[00:00:30.575] codec_open:283 codec open: 'm'
[00:00:30.577] stream_sock:508 connecting to 192.168.1.145:9000
```

Which I think show that the SqueezeESP is trying to output to the screen (`scroller` and `visu` messages especially)

Also - I thought in headless mode (i.e. no LMS) the display should still work with Bluetooth, and display "Now playing - [trackname]" messages

philippe44, 2023-08-16:
ES8388 requires MCLK so GPIO0 is forced to that. This means that as soon as the codec is initialized, your display will stop to work as GPIO is preempted for it

lelithium, 2023-08-18:
Switched CS from 0 to 22 and it works - thank you so much @philippe44 and @sle118 !

closed 2023-08-18 by lelithium