sle118/squeezelite-esp32 #282: Undesirable reboots when no LMS is present

pmannk, 2023-08-20:
Hi, can you please look at adding a flag (even if it's something that has to be added manually via the nvs editor) that _prevents_ reboots when no LMS is detected?
(https://github.com/sle118/squeezelite-esp32/commit/ba156586e9fc55c9a8bc943312eee8bd8c65d239) seemed to introduce a behaviour where the device will reboot when LMS fails. 
For those of us who only use our devices for Bluetooth/Airplay/Spotify (and don't actually have an LMS server) it's undesirable to reboot when LMS fails to connect. It's an expected condition in our circumstances.

Thanks, I really appreciate this awesome project!

sle118, 2023-08-22:
@pmannk under the Audio tab, you can try to disable Squeezelite and see if this works for you
[image]

sle118, 2023-08-23:
I'll let @philippe44  take a stab at this request, as this is in his area of expertise.

philippe44, 2023-08-23:
This one is unfortunate because the only reason why I reboot is because WiFi would stop working properly. So squeezelite is stuck in a forever search loop for a server. We can adda flag, but IMHO, it would be better to figure out why WiFi stops working

pr10dr, 2023-08-25:
+1 For adding a flag to make the reboot optional. Additional the reboot could be indicated in the web ui logs in a more understandable way. Iam a new user focusing on the spotify streaming experience and the regular reboots and web ui showing "timeout" where not helping me in understanding that i might need an LMS (which i dont have any intentions to use) to get rid of the issue. Downgraded for now.

sidenote: Overall the documentation and user experience where GREAT (heart)

philippe44, 2023-08-25:
Maybe I can add one option to squeezelite command line that disables the whole slimproto in which case it's more consistent with a usage without LMS. Having said that, you still have other projects that are more suitable

philippe44, 2023-08-25:
When LMS server is "-disable", LMS search will not be done and no reboot will happen. Please test and tell me if this works for you, I don't have time to try for myself. Remember that as this is a squeezelite-centered project, for example the audio backend is inside squeezelite so airplay, BT and spotify are nasty hacked I've added on it. In a proper architecture, audio backend should be an independent component that any client request when needed. So again, better use projects like euphonium.

pmannk, 2023-08-26:
Thanks for taking a look at this @philippe44. 
I appreciate it's a squeezelite-centered project, but it's so good at everything else too it would be a shame to alienate potential users with a hard LMS requirement.

I tested out your fix with 2.1347, I saw you had merged https://github.com/sle118/squeezelite-esp32/commit/1d65bf910623c330ab7e2da5b55d67971ef884cb. Thanks heaps!
I tested on two devices today and both have been working flawlessly. 
No reboots and I saw the log event as expected:
`[00:00:01.469] squeezelite_main:816 LMS is disabled`
`E (2945) messaging: [00:00:01.469] squeezelite_main:816 LMS is disabled`

Entry from the NVS editor for anyone else who stumbles across this thread:
autoexec1: squeezelite -b 500:2000 -d all=info -C 30 -W -n "Workshop" -o I2S **-s -disable**

philippe44, 2023-08-26:
You should see that log every hour

closed 2023-08-27 by philippe44