sle118/squeezelite-esp32 #285: WIFI dont let me connect

LaneaLucy, 2023-08-21:
hi, have a t-watch here. Flashed with the [installer](https://sle118.github.io/squeezelite-esp32-installer/).
and it flashes successfully, but when i try to connect to wifi, it dont let me...
Log:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (26) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (26) boot: compile time 01:18:39
I (26) boot: chip revision: v1.0
I (30) qio_mode: Enabling QIO for flash chip WinBond
I (35) boot.esp32: SPI Speed      : 80MHz
I (40) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (80) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 recovery         factory app      00 00 00010000 00140000
I (95) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (110) boot: End of partition table
I (115) boot: Defaulting to factory image
I (119) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=53578h (341368) map
I (222) esp_image: segment 1: paddr=000635a0 vaddr=3ffbdb60 size=04818h ( 18456) load
I (228) esp_image: segment 2: paddr=00067dc0 vaddr=40080000 size=08258h ( 33368) load
I (239) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=d0d4ch (855372) map
I (475) esp_image: segment 4: paddr=00140d74 vaddr=40088258 size=0c834h ( 51252) load
I (492) esp_image: segment 5: paddr=0014d5b0 vaddr=400c0000 size=00068h (   104) load
I (502) boot: Loaded app from partition at offset 0x10000
I (502) boot: Disabling RNG early entropy source...
I (514) psram: This chip is ESP32-D0WD
I (514) spiram: Found 64MBit SPI RAM device
I (514) spiram: SPI RAM mode: flash 80m sram 80m
I (517) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (525) cpu_start: Pro cpu up.
I (528) cpu_start: Starting app cpu, entry point is 0x400812a4
I (0) cpu_start: App cpu up.
I (1031) spiram: SPI SRAM memory test OK
I (1040) cpu_start: Pro cpu start user code
I (1040) cpu_start: cpu freq: 240000000
I (1040) cpu_start: Application information:
I (1043) cpu_start: Project name:     Squeezelite-ESP32
I (1049) cpu_start: App version:      I2S-4MFlash-1336
I (1055) cpu_start: Compile time:     Aug 12 2023 01:20:18
I (1061) cpu_start: ELF file SHA256:  34241d6e99fd1d6b...
I (1067) cpu_start: ESP-IDF:          v4.3.5-dirty
I (1073) cpu_start: Min chip rev:     v1.0
I (1078) cpu_start: Max chip rev:     v3.99 
I (1083) cpu_start: Chip rev:         v1.0
I (1088) heap_init: Initializing. RAM available for dynamic allocation:
I (1095) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1101) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1107) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1113) heap_init: At 3FFC47F8 len 0001B808 (110 KiB): DRAM
I (1120) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1126) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1133) heap_init: At 40094A8C len 0000B574 (45 KiB): IRAM
I (1139) spiram: Adding pool of 4066K of external SPI memory to heap allocator
I (1147) spi_flash: detected chip: winbond
I (1152) spi_flash: flash io: qio
W (1156) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (1170) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1180) spiram: Reserving pool of 64K of internal memory for DMA/internal allocations
I (1190) esp_app_main: System is booting from power on.
I (1200) esp_app_main: Reset reason is: 1
I (1200) esp_app_main: Recovery Reboot counter=1

I (1200) esp_app_main: Starting app_main
I (1210) nvs_utilities: Initializing flash nvs 
I (1220) nvs_utilities: Initializing nvs partition settings
I (1270) esp_app_main: Setting up telnet.
I (1270) telnet: Telnet support disabled
I (1270) esp_app_main: Setting up config subsystem.
W (1340) nvs_utilities: Configuration memory usage.  Heap internal:263067 (min:262691) (used:14384) external:4159299 (min:4151951) (used:0)
I (1340) esp_app_main: Registering default values
I (1350) system_api: Base MAC address is not set
I (1350) system_api: read default base MAC address from EFUSE
W (1360) config: Waiting for config commit ...
I (2340) config: configuration has some uncommitted entries
I (2340) config: Committing configuration to nvs. Locking config object.
I (2340) config: Done Committing configuration to nvs.
I (2340) config: Config committed!
I (2350) esp_app_main: Configuring services
I (2350) services: Configuring I2C sda:-1 scl:-1 port:1 speed:400000
W (2360) services: no I2C configured
I (2360) services: Configuring SPI mosi:-1 miso:-1 clk:-1 host:1 dc:-1
W (2370) services: no SPI configured
W (2380) led: LED GPIO -1 ignored
W (2380) led: LED GPIO -1 ignored
I (2380) led: Configuring LEDs green:-1 (active:0 -1%), red:-1 (active:0 -1%)
I (2390) battery: No battery
I (2390) monitor: Initializing monitoring
I (2400) monitor: Heap internal:263547 (min:262663) external:4142963 (min:4142947) dma:217899 (min:217031)
I (2410) esp_app_main: Initializing display
I (2410) display: Trying to configure display with N/A
W (2420) display: No display driver
I (2420) esp_app_main: Checking for update url
W (2430) config: Value not found for key fwurl
0;33mW (2450) config: Value not found for key apdelay
I (2450) uart: queue free spaces: 3
I (2450) network_handlers: Network manager configuration: polling max 10176, polling min 15000, ap delay 20000, dhcp timeout 8000, eth timeout 8000
W (2470) network_wifi: No known access point found
E (2470) nvs_utilities: Could not open the nvs storage.
W (2480) network_wifi: network manager has no previous configuration. ESP_FAIL
I (2480) http_server: Initializing HTTP Server
I (2500) network_handlers: Running recovery. Skipping ethernet, starting WiFi
I (2510) wifi:wifi driver task: 3ffc9524, prio:23, stack:6656, core=0
I (2510) wifi:wifi firmware version: 9ec65ef
I (2510) wifi:wifi certification version: v7.0
I (2510) wifi:config NVS flash: enabled
I (2520) wifi:config nano formating: disabled
I (2520) wifi:Init data frame dynamic rx buffer num: 40
I (2520) wifi:Init management frame dynamic rx buffer num: 40
I (2530) wifi:Init management short buffer num: 32
I (2530) wifi:Init static tx buffer num: 12
I (2540) wifi:Init tx cache buffer num: 32
I (2550) wifi:Init static rx buffer size: 1600
I (2550) wifi:Init static rx buffer num: 12
I (2560) wifi:Init dynamic rx buffer num: 40
I (2560) wifi_init: tcpip mbox: 32
I (2560) wifi_init: udp mbox: 32
I (2560) wifi_init: tcp mbox: 32
I (2570) wifi_init: tcp tx win: 8192
I (2570) wifi_init: tcp rx win: 32768
I (2580) wifi_init: tcp mss: 1440
I (2580) wifi_init: WiFi/LWIP prefer SPIRAM
I (2580) network_wifi: Starting wifi
I (2590) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (2690) wifi:mode : sta (8c:aa:b5:83:84:ec)
I (2690) wifi:enable tsf

W (2690) network_handlers: No saved wifi. Starting AP configuration mode.
I (2690) network_wifi: Configuring Access Point.
I (2700) network: Stopping DHCP server
I (2700) network: Starting DHCP server
I (2710) network_wifi: AP SSID: squeezelite-8384ec
I (2710) network_wifi: AP Password: squeezelite
I (2730) wifi:mode : sta (8c:aa:b5:83:84:ec) + softAP (8c:aa:b5:83:84:ed)
I (2730) wifi:Total power save buffer number: 6
I (2730) wifi:Init max length of beacon: 752/752
I (2740) wifi:Init max length of beacon: 752/752
I (3270) wifi:Total power save buffer number: 6
I (3270) wifi:Total power save buffer number: 6
I (3270) wifi:Set ps type: 1

I (3280) network_wifi: Initiating wifi network scan
I (3280) dns_server: DNS Server listening on 53/udp
****************************************************************
RECOVERY APPLICATION
This mode is used to flash Squeezelite into the OTA partition
****

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

****************************
Your terminal application does not support escape sequences.
Line editing and history features are disabled.
On Windows, try using Putty instead.
****************************
recovery-squeezelite-esp32>I (9840) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
I (9850) wifi:station: f8:5b:6e:ef:f1:c4 join, AID=1, bgn, 20
I (13850) wifi:station: f8:5b:6e:ef:f1:c4 leave, AID = 1, bss_flags is 655458, bss:0x3f824970
I (13850) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
>`

LaneaLucy, 2023-08-21:
btw, even with the join command, i dont get it to work. then i get:

I (58700) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (58700) wifi:state: init -> auth (b0)
I (58710) wifi:state: auth -> assoc (0)
I (58730) wifi:state: assoc -> run (10)
I (61750) wifi:state: run -> init (fc0)
I (61750) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
W (61750) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: 96:83:c4:23:e4:60, reason code: 15 (WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT)
E (61760) network: network manager Unhandled Event NETWORK_WIFI_CONFIGURING_ACTIVE_STATE(WIFI_CONFIGURING_STATE) [EN_LOST_CONNECTION]
I (61780) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (61780) wifi:state: init -> auth (b0)
I (61790) wifi:state: auth -> assoc (0)
I (61810) wifi:state: assoc -> run (10)
I (64850) wifi:state: run -> init (fc0)
I (64850) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
W (64850) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: 96:83:c4:23:e0:94, reason code: 15 (WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT)
E (64860) network: network manager Unhandled Event NETWORK_WIFI_CONFIGURING_ACTIVE_STATE(WIFI_CONFIGURING_STATE) [EN_LOST_CONNECTION]
W (67270) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: 00:00:00:00:00:00, reason code: 205 (WIFI_REASON_CONNECTION_FAIL)
E (67270) network: network manager Unhandled Event NETWORK_WIFI_CONFIGURING_ACTIVE_STATE(WIFI_CONFIGURING_STATE) [EN_LOST_CONNECTION]
I (67280) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (67290) wifi:state: init -> auth (b0)
I (67300) wifi:state: auth -> assoc (0)
I (67310) wifi:state: assoc -> run (10)
I (70330) wifi:state: run -> init (fc0)
I (70330) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
W (70330) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: 96:83:c4:23:e4:60, reason code: 15 (WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT)
E (70340) network: network manager Unhandled Event NETWORK_WIFI_CONFIGURING_ACTIVE_STATE(WIFI_CONFIGURING_STATE) [EN_LOST_CONNECTION]
I (70360) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1

LaneaLucy, 2023-08-21:
i now got it to connect to a open AP but even then, i cant open the web gui

LaneaLucy, 2023-08-21:
what is the latest build known to work with t-watch2020v1?

sle118, 2023-08-22:
you can install the standard i2s-4mb build and configure the options as explained here: https://github.com/sle118/squeezelite-esp32#t-watch2020-by-lilygo 

note that you can activate the nvs editor under the "credits" tab if it's not already visible. By default it is available in recovery mode only and this credits tab option enables it in regular "application/squeezelite" mode.

sle118, 2023-08-22:
@LaneaLucy also note that in recent versions of the firmware, the Watch is a "known hardware board", which means it has a drop down (in system, or hardware, can't recall) where some models have pre-configured setup coming from this list:
https://gist.githubusercontent.com/sle118/dae585e157b733a639c12dc70f0910c5/raw/

LaneaLucy, 2023-08-22:
@sle118 problem is first, that WiFi only works with NOT encrypted aka open networks and even that not very good. Second, it looks like it missing the code to tell the energy management module to enable the power outputs for display and speaker

LaneaLucy, 2023-08-22:
I can't connect to the softap nor can it connect to wpa2-personal

LaneaLucy, 2023-08-22:
And compiling with t-watch main system target gives error about undeclared CONFIG_SET_GPIO

sle118, 2023-08-22:
for the record, here is what the option looks like: 
[image]

LaneaLucy, 2023-08-22:
@sle118 
Again, this is not the problem. I already set this. The problem is mainly wlan not working with everything other then open networks, which also has big problems on getting a IP and throughout. And the second problem is that it looks like the software don't tell the power management IC to switch on the power outputs for display and speaker. Without power to them, the esp can't communicate with them. But the bigger problem is still WiFi, because a open WiFi is a very big security risk...

sle118, 2023-08-22:
@LaneaLucy One thing you can do is to edit the sdkconfig and bridge the missing definitions with their corresponding values below. 
    "dac_config": "model=I2S,bck=26,ws=25,do=33,i2c=106,sda=21,scl=22",
    "set_GPIO": "21=amp,22=green:0,39=jack:0",
    "spi_config": "dc=27,data=19,clk=18",
    "display_config": "SPI,driver=ST7789,width=240,height=240,cs=5,back=12,speed=16000000,HFlip,VFlip",
    "spdif_config": ""
   ```

In your case, you're missing : `21=amp,22=green:0,39=jack:0`

LaneaLucy, 2023-08-22:
@sle118 do you actually read what I write?

sle118, 2023-08-22:
> @sle118 do you actually read what I write? Or is this a chat bot that answers?

@LaneaLucy Not sure if this is humor or a real question, but I'll go with the first case; if I was a chat bot I would actually fully read every comment and memorize them all.  In real life, I have little time right now to dedicate to this hobby project and in that little time, I try being as helpful as possible to newcommers. With the answer `21=amp,22=green:0,39=jack:0`, I was answering the question you had on the compiler output error. 

As for the WiFi not connecting you are the first one reporting the issue. I searched for WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT and found something that might be relevant for you: https://github.com/espressif/esp-idf/issues/6644

philippe44, 2023-08-22:
Not sure insulting or mocking the devs is going to be helpful. I did the port to TWatch but that was a while ago so I don't know if everything still works as it did. Now, I never added proper power management, so you'd have to add your own code. Finally, the WiFi stack works with non-opened network for sure

LaneaLucy, 2023-08-22:
> With the answer `21=amp,22=green:0,39=jack:0`, I was answering the question you had on the compiler output error.
> 
This would be something, that the var could be set, but with t-watch target it's not even declared and I don't know where it should be declared. Sorry, but this don't help without additional info.
But like I said, for me it looks like the ap202 or what the power management IC was called (it's some time ago the last time I programmed for the t-watch or t-beam), isn't instructed to enable the power outputs for the display and amp. By default they are disabled and only the output for the esp itself is enabled on power up. So if I remember right, it's connected trough i2c and first need to be told what voltages to output and then enable the outputs.

> As for the WiFi not connecting you are the first one reporting the issue. I searched for WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT and found something that might be relevant for you: [espressif/esp-idf#6644](https://github.com/espressif/esp-idf/issues/6644)
>
This I already read and crosschecked for duplicate Mac addresses inside my home network. But that even connecting to the softap don't works, looks more like a software or hardware problem. I don't know when the last time the software was tested on the t-watch, maybe something broke? I even tried older versions from years ago, but no luck. Compiling it myself didn't helped either with i2s default config.

And sorry for sounding rude. I just thought I write with a chat bot because my question where answered with (for me) random answers about t-watch config.

LaneaLucy, 2023-08-22:
Btw I can help with the power management code if needed

philippe44, 2023-08-22:
I've pushed a small fix for TWATCH2020

LaneaLucy, 2023-08-22:
> I've pushed a small fix for TWATCH2020

I try tomorrow

LaneaLucy, 2023-08-23:
@sle118 
I can't even connect to the squeezelite access point. That's where I get the handshake error...
And I only have 2.4 GHz at the moment here...

And did you also tried to set the known hardware board? Is the display and speaker working?

What version of the t-watch do you have?

sle118, 2023-08-23:
I have the TWATCH 2020 V1 and I think the built-in config works for the v2 only, as Philippe provided the details in the readme. The V1 actually has a different address on the APU i2c bus. so you can try: 
model=I2S,bck=26,ws=25,do=33,i2c=53,sda=21,scl=22

which worked for me. I'll try to update the gist

sle118, 2023-08-25:
I'll go ahead and close this now since there isn't much I can do to help without being able to reproduce on my end.

closed 2023-08-25 by sle118