sle118/squeezelite-esp32 #286: Docker build fails from 634/1525

alidaf, 2023-08-22:
The initial part completes...

`-- Configuring done`
`-- Generating done`
`-- Build files have been written to: /workspace/squeezelite-esp32/Git/squeezelite-esp32/build/bootloader`

But then fails from part 634/1525 with multiple errors...

`...`
`/workspace/squeezelite-esp32/Git/squeezelite-esp32/build/esp-idf/spotify/cspot/nanopb/generator/protoc-gen-nanopb: 13: exec: /workspace/squeezelite-esp32/Git/squeezelite-esp32/build/esp-idf/spotify/cspot/nanopb/generator/nanopb_generator.py: Permission denied
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 126.`

This happens even after turning off Spotify support in the menuconfig.

Trying to run nanopb_generator.py directly results in ...

`bash: ./build/esp-idf/spotify/cspot/nanopb/generator/nanopb_generator.py: /usr/bin/env: bad interpreter: Permission denied`

sle118, 2023-08-22:
Please describe all the steps you are taking, as it is impossible to deduct the branch or docker image you are using from what you provided.

alidaf, 2023-08-22:
I've been following the guide and using the images suggested:

Docker Desktop 
-> pull sle118/squeezelite-esp32-idfv435
Clone the repo
-> `git clone --recursive https://github.com/sle118/squeezelite-esp32.git`
Run the Docker image
-> `docker run -it -v `pwd`:/workspace/squeezelite-esp32 sle118/squeezelite-esp32-idfv435`
Change to the repo folder
-> `idf.py menuconfig`
-> configure the display
-> exit
-> `idf.py build`
-> errors!

alidaf, 2023-08-22:
@sle118  I've managed to install the generic firmware via the web installer. I didn't realise that the gpio settings could be made via the web interface so I don't need to build from source. Close this at your discretion.

sle118, 2023-08-22:
I'm happy that the basic generic build works for you. This way you can keep your devices up to date with the standard firmware update process. 

take care!

closed 2023-08-22 by sle118