sle118/squeezelite-esp32 #288: Need help: ethernet w5500

meijer3, 2023-08-25:
Hi
First of all thanks for the wonderful product.

My wifi is bad so I want to connect it to ethernet.
I do have a w5500 chip.
https://www.tinytronics.nl/shop/en/communication-and-signals/ethernet/modules/w5500-ethernet-module-wiz850io-compatible

I connected like this:

|Pin Name |  SPI2
|-- |  -- |
|CS |  5
|SCLK |  18
|MISO |  19
|MOSI |  23

Ground and 3.3v aswell

I do need a bit more help with the config. I am not that familiair with SPI and interuters and such

I configured spi
spi_config: ```data|mosi=23,clk=18,miso=19```
And I am not so sure about this. 
eth_config: `model=dm9051|w5500,cs=<gpio>,speed=<clk_in_Hz>,intr=<gpio>[,rst=<gpio>]`

Do I need to choose the model? or is this a 'both'- statement
Do I need the intr pin? Can that be 22?.
Should I even have the rst?

And what about `ethtmout`?

Thanks so much for all

philippe44, 2023-08-25:
in syntax [] means optional and | means either or. Other parameters are mandatory.

closed 2023-08-27 by philippe44

mimk97, 2023-11-22:
> Hi First of all thanks for the wonderful product.
>
> My wifi is bad so I want to connect it to ethernet. I do have a w5500 chip. [...]

Have you solved the Problem?

meijer3, 2023-11-26:
Not yet, it is on my todo list...