sle118/squeezelite-esp32 #289: Airplay crashes system

RoganDawes, 2023-08-26:
Not sure if this is the same issue as #260, but thought I would start a new issue as I was struggling to make sense of the debugging steps. This also happens immediately I try to play any audio via Airplay, I don't even get to changing tracks. My board is an "Old model config 1 A1S" with AC101, with speakers connected via the headphone out. It works fine (in very limited testing) via Bluetooth, using the latest Generic I2S version installed this morning via the web installer.

```
[00:03:05.278] handle_rtsp:659 [0x3fac7558]: responding:
RTSP/1.0 200 OK
Transport: RTP/AVP/UDP;unicast;mode=record;control_port=65091;timing_port=65092;server_port=65090
Session: DEADBEEF
Audio-Jack-Status: connected; type=analog
CSeq: 5

[00:03:05.288] output_thread_i2s:498 Output state is 0
[00:03:05.304] output_thread_i2s:564 Restarting I2S.
[00:03:05.308] equalizer_open:71 equalizer initialized 0
[00:03:05.321] handle_rtsp:440 [0x3fac7558]: received RECORD
[00:03:05.322] rtp_record:373 [0x3fad4774]: record 0 0
[00:03:05.323] raop_sink_cmd_handler:284 Stream
[00:03:05.335] handle_rtsp:659 [0x3fac7558]: responding:
RTSP/1.0 200 OK
Audio-Jack-Status: connected; type=analog
CSeq: 6

[00:03:05.363] handle_rtsp:440 [0x3fac7558]: received SET_PARAMETER
[00:03:05.365] handle_rtsp:609 [0x3fac7558]: SET PARAMETER volume -20.000000
[00:03:05.366] raop_sink_cmd_handler:317 Volume[0..1] 0.3333
[00:03:05.378] handle_rtsp:659 [0x3fac7558]: responding:
RTSP/1.0 200 OK
Audio-Jack-Status: connected; type=analog
CSeq: 7

[00:03:05.405] rtp_thread_func:708 [0x3fad4774]: 1st sync packet received
[00:03:05.512] raop_sink_cmd_handler:305 Play
[00:03:05.515] raop_sink_cmd_handler:310 Starting at 33145 (in 1923 ms)
[00:03:05.516] output_thread_i2s:498 Output state is 5
[00:03:05.567] handle_rtsp:440 [0x3fac7558]: received FLUSH
[00:03:05.568] rtp_flush:347 [0x3fad4774]: FLUSH ignored as same as RECORD (37613 - 3265458061)
[00:03:05.580] rtp_flush:356 [0x3fad4774]: flush 37613 3265458061
[00:03:05.582] handle_rtsp:659 [0x3fac7558]: responding:
RTSP/1.0 200 OK
Audio-Jack-Status: connected; type=analog
CSeq: 8

[00:03:06.273] discover_server:828 sending discovery 19
[00:03:07.443] output_thread_i2s:498 Output state is 2
[00:03:07.449] handle_rtsp:440 [0x3fac7558]: received SET_PARAMETER
[00:03:07.452] handle_rtsp:638 [0x3fac7558]: received JPEG image of 112196 bytes
[00:03:07.466] handle_rtsp:659 [0x3fac7558]: responding:
RTSP/1.0 200 OK
Audio-Jack-Status: connected; type=analog
CSeq: 9

[00:03:07.528] handle_rtsp:440 [0x3fac7558]: received SET_PARAMETER
[00:03:07.529] handle_rtsp:619 [0x3fac7558]: SET PARAMETER progress 220000/520000 progress: 3255724329/3265459973/3278693673

[00:03:07.542] handle_rtsp:659 [0x3fac7558]: responding:
RTSP/1.0 200 OK
Audio-Jack-Status: connected; type=analog
CSeq: 11

[00:03:07.701] handle_rtsp:440 [0x3fac7558]: received SET_PARAMETER
[00:03:07.702] handle_rtsp:628 [0x3fac7558]: received metadata
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400955b1  PS      : 0x00060b30  A0      : 0x8025c500  A1      : 0x3fac8f90
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3fac8f50
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3fac91d4  A13     : 0x00000000
A14     : 0x3f813170  A15     : 0x00000008  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400955b1  LEND    : 0x400955c1  LCOUNT  : 0xffffffff

Backtrace:0x400955ae:0x3fac8f90 0x4025c4fd:0x3fac8fa0 0x4025dc09:0x3fac92b0 0x400ff62d:0x3fac92e0 0x40109a56:0x3fac9330 0x40109c2d:0x3fac94d0

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
```

VaBanck, 2023-08-26:
The most stable version for AirPlay is version 1271.

philippe44, 2023-08-26:
it's fixed (we have an issue with build number though)

RoganDawes, 2023-08-27:
> it's fixed (we have an issue with build number though)

I'm not sure what this means. Should I use the latest version? Should I be using version 1271?

philippe44, 2023-08-27:
I think you have. Now a version 1501 that should do

RoganDawes, 2023-08-27:
> I think you have. Now a version 1501 that should do 

Thank you. I will try 1501 and report back.

philippe44, 2023-08-30:
I can't reproduce with my iPhone. Can you provide some logs (serial or telnet)?

RoganDawes, 2023-08-31:
Now I cannot reproduce either. If I manage, I will reopen with logs.

closed 2023-08-31 by RoganDawes