sle118/squeezelite-esp32 #290: Equalizer not working as expected

brizzer, 2023-08-26:
Hi,

Thanks for a remarkable piece of software!

I was doing some measurement and playing around with the equalizer and found that it's not working as expected. The frequency bands are too narrow especially when using negative values.

Here are some measurements using REW and Olimex ESP32-EVB's (crappy) speaker, ranging from all the taps at -13 to all the taps at +20. The green line is with all taps at 0:

[image]

You can clearly see deep dips at the tap frequency.

Here is a graph with the differences between the measurement and 0 where you also see the tops at positive tap values.
[image]

What is remarkable is that even at -2dB tap setting there is a -8dB dip at tap frequency.

I realise that this is probably a ESP-ADF issue but i wanted to confirm this here before submitting it there.

Regards / Martin

philippe44, 2023-08-26:
Yes, this is an ADF issue as this time I'm just using their stuff.

philippe44, 2023-08-27:
I'll close this one as this just their filter

closed 2023-08-27 by philippe44