sle118/squeezelite-esp32 #291: Add IR RC5 Protocol and repeat function

Renber78, 2023-08-26:
Hi,

  I use your code and I am satisfied with it, thank you for the work!

For hardware reasons I would like to use a remote control rather than buttons (simpler mechanical level). I tested with an NEC compatible remote control, it works well except the volume buttons because there is no repeat function, is it possible to add this functionality which is supported by the LMS?

I also have a large number of remote controls but most are with the RC5 protocol. Nowadays, finding an original slimdevices remote control is mission impossible. The number of brands using the NEC protocol is quite limited (Sanyo, Fisher, Nec, Mitsubishi, ...?). Wouldn't it be interesting to add the RC5 protocol in order to support so-called "media center" remote controls? These remote controls are very popular and can be found for a few € on alliexpress. (HP media remote)

I am not able to do code for RC5 but I can provide various .ir mapping as I have a lot of stock.

What do you think ?

philippe44, 2023-08-27:
There a version( not released) being built that should handle rc5. I don't have many other remotes and it does not seem to work with my TV, so maybe it's a problem but can you try and let me know. The log is verbose enough at the moment that it will tell you scan codes

Renber78, 2023-08-27:
Thank you for the work already done since my request.
Unfortunately I have no results with various RC5 remotes, nothing appears in the logs (with an NEC it always works unless IR:rc5, normal)
If you have an RC5 remote control (Philips for example) make sure it works by pressing in front of a camera (you should see the IR led light up)

Another point, after research, it is possible that the remote controls say "media center" use the RC6 protocol ... if you could add this one too :-)

For the repeat function, have you worked on it on this build?

THANKS

philippe44, 2023-08-27:
No, rc6 is not something that I'll support. For repeat, I have to check but I'm not sure this is easy to integrate with LMS and the way I handle or codes, so it's a bit unlikely. I'll see what I can do for rc5 but unless there is something obvious I've missed, I'm pretty out of options here

philippe44, 2023-08-27:
Repeat is handled. I don't think I have an rc5 remote, all packets I receive are 34 in length which is NEC or Pronto (AFAIU). Probably your remote are not rc5 either. I've added a log where you can see the whole received buffer so you can figure out the coding and type of IR. 

It's a byte dump organized by 16 bits packets
- take 2 bytes, the LSB is first so swap them
- the 1 bit is the value read by the IR receiver, inverted
- the next 15 bits are the duration in µs of the bit
- and so on...

For example, using NEC protocol : bb 22 ee 91 is
bb 22 => 22 bb => first bit is 0 => 1, duration is 22 bb  = 8891 µs
ee 91 => 91 ee => first bit is 1 => 0, duration is 11 ee = 4590 µs

This is a proper NEC header pattern. To analyze, chatGPT can be your friend

Renber78, 2023-08-28:
Thanks for the effort ;-)
I will test as soon as the build is online because I don't have the tools to compile.

wizmo2, 2023-08-28:
@philippe44.  I have a SqueezeAmp with the IR pre installed.  The latest release does not run in ota mode (reverts to recovery), and gives the following error.
```
I (2892) IR: Starting Infrared Receiver on gpio 0
E (2892) rmt: rmt_driver_install(1021): RMT driver already installed for channel
```

Had to remove the `0=ir` or add one of the new options to get it to boot.

philippe44, 2023-08-28:
That's interesting, I did not have that issue and I did not change to nec or rc5 to start with. What else do you have on your set_gpio line?

philippe44, 2023-08-28:
> Thanks for the effort ;-)
> 
> I will test as soon as the build is online because I don't have the tools to compile.
>
Thé build is there but you need to manually download the binary by going to the tags and select the one you want. It is not "released" which means it will not appear in the "update" of the firmware so you need to select the downloaded file

Renber78, 2023-08-28:
but sorry I can't find binary. Can you tell me exactly where it is?

wizmo2, 2023-08-28:
just the 

> That's interesting, I did not have that issue and I did not change to nec or rc5 to start with. What else do you have on your set_gpio line?

just the 0=ir

Renber78, 2023-08-28:
FYI: I just did a test with the SqueezeAmp-16 (501) firmware and the same config as you (0=ir) and I had no boot problem.

dfrein, 2023-09-05:
I can also confirm that an update to a recent version without changing <gpio>=ir leads to a boot loop (and reflashing also required to erase flash memory). However, if I change from "0=ir" to "0=ir:nec" just before flashing to I2S-4MFlash-1524, the new builds can be flashed without further problems.

Renber78, 2023-09-05:
I notice that it is not the repeat function that is the problem but the volume setting itself (I believe that the new loudness option has something to do with it)
A lot of data is sent to the server during a volume change.

philippe44, 2023-09-05:
> I notice that it is not the repeat function that is the problem but the volume setting itself (I believe that the new loudness option has something to do with it)
> 
> A lot of data is sent to the server during a volume change.
>
Indeed when loudness is activated, there is a lot happening when changing volume (and I've already minimized it a fair bit). I don't think you will experience stuttering is loudness is disabled.

philippe44, 2023-09-05:
> I can also confirm that an update to a recent version without changing <gpio>=ir leads to a boot loop (and reflashing also required to erase flash memory). However, if I change from "0=ir" to [...]
That's weird. I'll investigate again

philippe44, 2023-09-05:
Did you have anything else on your set_GPIO line ?

Renber78, 2023-09-05:
`I don't think you will experience stuttering is loudness is disabled.`

it is off

dfrein, 2023-09-05:
Seems to be only an issue for me with build squeezelite-esp32-master-v4.3-I2S-4MFlash-16-21522.zip, perhaps a (different) issue with this specific build? Now I tested again and when flashing today's build (21524), everything seems to be fine, both tested with 0=ir and 0=ir:nec (and nothing else in my set_GPIO line), whereas 21522 again leads to a boot loop (build 21522 only tested with 0=ir). As this build is not an official / published one, and the newer one works for me, I would now not perform any further test and consider this issue as solved.

philippe44, 2023-09-05:
> Seems to be only an issue for me with build squeezelite-esp32-master-v4.3-I2S-4MFlash-16-21522.zip, perhaps a (different) issue with this specific build? Now I tested again and when flashing today's [...]
What bothers me is that @wizmo2 had the same issue with another build once and after that it went away. So I'm more concerned that it is something else that happens at first time the version runs and that might be a disaster when I release it. If you try again 1522 on a line with just XX=ir, does it still crash?

philippe44, 2023-09-05:
> `I don't think you will experience stuttering is loudness is disabled.`
> 
> 
> 
> it is off
>
Or that's very weird then. Can you confirm that the logs does not say anything about equalizer being active at every volume change? Can you also try the latest build where I removed all causes of unnecessary equalizer updates?

Renber78, 2023-09-05:
[02:13:12.943] equalizer_process:205 EQUALIZER INDEX 0 => gain:0.00, loudness:0.00,
[02:13:12.953] equalizer_process:205 EQUALIZER INDEX 1 => gain:0.00, loudness:0.00,
[02:13:12.964] equalizer_process:205 EQUALIZER INDEX 2 => gain:0.00, loudness:0.00,
[02:13:12.976] equalizer_process:205 EQUALIZER INDEX 3 => gain:0.00, loudness:0.00,
[02:13:12.977] equalizer_process:205 EQUALIZER INDEX 4 => gain:0.00, loudness:0.00,
[02:13:12.989] equalizer_process:205 EQUALIZER INDEX 5 => gain:0.00, loudness:0.00,
[02:13:12.990] equalizer_process:205 EQUALIZER INDEX 6 => gain:0.00, loudness:0.00,
[02:13:13.001] equalizer_process:205 EQUALIZER INDEX 7 => gain:0.00, loudness:0.00,
[02:13:13.013] equalizer_process:205 EQUALIZER INDEX 8 => gain:0.00, loudness:0.00,
[02:13:13.014] equalizer_process:205 EQUALIZER INDEX 9 => gain:0.00, loudness:0.00,
[02:13:13.026] equalizer_process:210 equalizer deactivated
This is the version taken this very morning so it seems to me to be the last one (the number is not displayed)

Renber78, 2023-09-05:
I have the same log several times between two volume action logs and that's when the sound stutters. I think this behavior is not expected

`I (2550920) IR: Scan Code (repeat) --- addr: 0x0019 cmd: 0x0011
[02:29:10.786] equalizer_process:205 EQUALIZER INDEX 0 => gain:0.00, loudness:0.00,
[02:29:10.788] equalizer_process:205 EQUALIZER INDEX 1 => gain:0.00, loudness:0.00,
[02:29:10.799] equalizer_process:205 EQUALIZER INDEX 2 => gain:0.00, loudness:0.00,
[02:29:10.800] equalizer_process:205 EQUALIZER INDEX 3 => gain:0.00, loudness:0.00,
[02:29:10.812] equalizer_process:205 EQUALIZER INDEX 4 => gain:0.00, loudness:0.00,
[02:29:10.813] equalizer_process:205 EQUALIZER INDEX 5 => gain:0.00, loudness:0.00,
[02:29:10.824] equalizer_process:205 EQUALIZER INDEX 6 => gain:0.00, loudness:0.00,
[02:29:10.836] equalizer_process:205 EQUALIZER INDEX 7 => gain:0.00, loudness:0.00,
[02:29:10.837] equalizer_process:205 EQUALIZER INDEX 8 => gain:0.00, loudness:0.00,
[02:29:10.848] equalizer_process:205 EQUALIZER INDEX 9 => gain:0.00, loudness:0.00,
[02:29:10.849] equalizer_process:210 equalizer deactivated
[02:29:10.925] equalizer_process:205 EQUALIZER INDEX 0 => gain:0.00, loudness:0.00,
[02:29:10.927] equalizer_process:205 EQUALIZER INDEX 1 => gain:0.00, loudness:0.00,
[02:29:10.938] equalizer_process:205 EQUALIZER INDEX 2 => gain:0.00, loudness:0.00,
[02:29:10.940] equalizer_process:205 EQUALIZER INDEX 3 => gain:0.00, loudness:0.00,
[02:29:10.951] equalizer_process:205 EQUALIZER INDEX 4 => gain:0.00, loudness:0.00,
[02:29:10.952] equalizer_process:205 EQUALIZER INDEX 5 => gain:0.00, loudness:0.00,
[02:29:10.964] equalizer_process:205 EQUALIZER INDEX 6 => gain:0.00, loudness:0.00,
[02:29:10.976] equalizer_process:205 EQUALIZER INDEX 7 => gain:0.00, loudness:0.00,
[02:29:10.978] equalizer_process:205 EQUALIZER INDEX 8 => gain:0.00, loudness:0.00,
[02:29:10.989] equalizer_process:205 EQUALIZER INDEX 9 => gain:0.00, loudness:0.00,
[02:29:10.990] equalizer_process:210 equalizer deactivated
[02:29:11.686] sendIR:94 sending IR code 190011
I (2551890) IR: Scan Code  --- addr: 0x0019 cmd: 0x0011
`

philippe44, 2023-09-05:
Are you sure it is build 1523?

Renber78, 2023-09-05:
I did: "git clone --recursive https://github.com/sle118/squeezelite-esp32.git" this morning around 8am, is this the right method?
I don't see a version number anywhere, where can I find it?

philippe44, 2023-09-05:
> I did: "git clone --recursive https://github.com/sle118/squeezelite-esp32.git" this morning around 8am, is this the right method? I don't see a version number anywhere, where can I find it?
>
Go on the GH site here https://github.com/sle118/squeezelite-esp32/actions, select the build you want (say the latest one) and scroll down till you see artifacts. Take the binary that fits your platform

Renber78, 2023-09-05:
Thanks for this tip (I've been looking for this for a long time)

As a result, I must not have had the latest version, I no longer have any stuttering now and no more log loudness.
Sorry for the waste of time.

dfrein, 2023-09-05:
Yes, it's still crashing with 522: I just tried again, and it was the same: starting with the web installer I erased the device and flashed the generic/I2S version provided with the web installer. In the next step I connected to the esp32 wifi and uploaded and committed my old configuration, performed a complete reboot (with power off), and flashed from the internal web interface squeezelite.bin, extracted from a fresh download from github actions of squeezelite-esp32-master-v4.3-I2S-4MFlash-16-21522.zip. After pressing exit recovery the device was again stuck in a boot loop. The same procedure, just with 21524, was without any problems. Please find attached my complete config, with regard to set_GPIO it's just 0=IR
[nvs_config_cirrus_1630835795341json.txt](https://github.com/sle118/squeezelite-esp32/files/12527642/nvs_config_cirrus_1630835795341json.txt)

philippe44, 2023-09-05:
> Yes, it's still crashing with 522: I just tried again, and it was the same: starting with the web installer I erased the device and flashed the generic/I2S version provided with the web installer. [...]
Thanks - so that is probably something else causing the boot loop because I have not modified anything wrt to IR between these release (I'll double check) but a least that gives me a hint how to reproduce

Renber78, 2023-09-05:
could be a clue:

E (2911) gpio: io_num=34 can only be input
E (2911) gpio: gpio_set_level(237): GPIO output gpio_num error

My gpio 34 is ir!

Renber78, 2023-09-05:
I flashed 522 is bootloop.
here is the log where it is blocking for me:

[00:01:00.014] output_init_embedded:100 init I2S/SPDIF
W (3741) DAC core: DAC does not use i2c
[00:01:00.025] output_init_i2s:324 configuring MCLK on GPIO -1
[00:01:00.026] output_init_i2s:336 invalid MCK gpio -1
E (3771) I2S: i2s_driver_install(921): I2S buffer length at most 1024 and more than 8
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

dfrein, 2023-09-05:
Just to be clear: it was just my initial assumption that "ir:nec" is now required. I checked again, with build 1522 both 0=ir and 0=ir:nec will lead to a boot loop (on the basis of the config file I have attached above). So indeed it has nothing to do with specific setting (this "0=ir" ).

philippe44, 2023-09-05:
oh yes, I forgot it - the 1522 has an issue with i2S. I pushed a wrong change. I'll delete it

Renber78, 2023-09-12:
I'm on the latest version (1532) and I use the same IR sensor as you. No problem with both protocols...
<io>=ir or <io>=ir:nec = NEC only
<io>=ir:rc5 = RC5 only

Both protocols are at 38khz and work with the same sensor.

Please note that some old sensors need 5V to function correctly.

philippe44, 2023-09-12:
> Has a bug been introduced with the option for :nec|rc5? With a new build I could only get a TSOP4838 to work without any additional parameters. I initially tried with :nec as it is a 38kHz receiver but neither option worked until I removed the option completely.
>
What syntax did you use *exactly*

wizmo2, 2023-09-15:
(https://github.com/sle118/squeezelite-esp32/commit/c7e4d9711c6c5bc8c13d185734b66d41744c543e) seems to have introduced a problem with RMT initialization.  I'm having a look now, but I get `RMT CHANNEL ERR` during initialization with the latest releases.

philippe44, 2023-09-15:
I don't have anything to test unfortunately. The idea was that some rmt parameters must be shared across the system

wizmo2, 2023-09-15:
See the justification.  Will see if I can work out what's happening. May be into the w/e (paying work to do first 😏 )

philippe44, 2023-09-15:
I'll re-read the changes to see if I can stop something as well

philippe44, 2023-09-15:
I think I forgot to set the interrupt, no?

Renber78, 2023-09-16:
I don't know if it's related but it seems that there is a bug with the set_GPIO.

My set_GPIO config:
34=ir:rc5

My log:
E (2788) gpio: io_num=34 can only be input
E (2788) gpio: gpio_set_level(237): GPIO output gpio_num error
I (2798) led: Configuring LEDs green:-1 (on:0 rmt:-1 -1% ), red:34 (on:0 rmt:-1 -1% )

It looks like he took the GPIO34 also for the red LED... but my remote control works correctly. (I don't have LEDs)

philippe44, 2023-09-16:
Indeed, fixed - good catch, thanks!

Renber78, 2023-09-16:
I still have the same error message with 1552

philippe44, 2023-09-16:
Rebuild is happening now

Renber78, 2023-09-16:
I was too fast ;-)
It actually works now.
THANKS

wizmo2, 2023-09-16:
> I think I forgot to set the interrupt, no?

@philippe44, The interrupt was not the issue with led_vu initialization.  I entered a PR with a fix.

closed 2023-09-24 by Renber78

Androssboy, 2024-02-14:
> Повтор обрабатывается. Я не думаю, что у меня есть пульт rc5, все пакеты, которые я получаю, имеют длину 34, то есть NEC или Pronto (AFAIU). Вероятно, ваш пульт тоже не rc5. Я добавил лог, где можно увидеть весь полученный буфер, чтобы можно было разобраться с кодировкой и типом IR.
> 
> Это байтовый дамп, организованный 16-битными пакетами.
> 
> * возьмите 2 байта, LSByte будет первым, поэтому поменяйте их местами
> * 1-й бит (MSbit) — это значение, считываемое ИК-приемником, инвертированное
> * следующие 15 бит — длительность бита в мкс.
> * и так далее...
> 
> Например, при использовании протокола NEC: bb 22 ee 91 равно bb 22 => 22 bb => первый бит равен 0 => 1, длительность 22 bb = 8891 мкс ee 91 => 91 ee => первый бит равен 1 => 0 , длительность 11 ee = 4590 мкс
> 
> Это правильный шаблон заголовка NEC. Для анализа вам может помочь ChatGPT. Что-то вроде: «Возьмите каждый сегмент в шестнадцатеричном формате, соберите их по 2, чтобы сформировать 16-битное значение. Поменяйте местами байты. Возьмите старший бит, инвертируйте его, это значение. Возьмите 15 младших битов, это длительность в мкс». И будьте осторожны, ChatGPT тоже извергает много всякой ерунды.

@philippe44
Dear philippe44!
I'm amazed by your intelligence and skills! Thank you very much for your titanic work.
I'm trying to connect an IR remote control to an ESP32-A1S that has squeezelite-esp32 installed, but I can't succeed. I counted the remote control button codes using Arduino and did everything according to the instructions from the documentation on GIThub.
After carefully reading the forums, I found your message which I quoted at the beginning of this post.
Is this procedure for converting bits and bytes necessary for every remote control and every button? Or is it enough to read the button code using an Arduino sketch and paste it into Slim_Devices_Remote.ir on the LMS server? I'm completely confused...
Please forgive me, but I am not a programmer and I am trying to repeat the project with my own hands. By repeating after you, I myself become smarter!