sle118/squeezelite-esp32 #294: ESP32 crashes after few minutes of audio playback (i2c_driver_delete)

LaurentMe, 2023-09-02:
Hi

I've successfully flashed the code on the ESP32 using both the ESP-IDF explorer extension in Visual Studio Code (v4.3.5 as the documentation suggests) and using the web installer (https://sle118.github.io/squeezelite-esp32-installer/). 

The module I'm using is the ESP32-WROVER-E which should be compatible. After configuring the external dac, a PCM5102 module, playback works perfectly. Bluetooth, Spotify Connect using CSpot, Airplay, it all works. 

But after a few minutes the whole device crashes and reboots. The error it throws reads: "i2c_driver_delete(395): i2c driver install error". According to the docs of Espressif, it's a function used to deinitialize the driver for i2c (https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32/api-reference/peripherals/i2c.html#delete-driver).

Has anyone an idea what the problem could be? Maybe I'm missing a configuration step?

Thanks in advance for your response!

Logs:
```
[00:08:05.673] discover_server:828 sending discovery 1
[00:08:10.673] decode_close:233 close decode
[00:08:10.753] deregister_external:472 Stopping AirPlay sink
[00:08:10.853] deregister_external:479 Stopping CSpot sink
[00:08:10.855] stream_close:441 close stream
[00:08:10.871] output_close_embedded:112 close output
E (128600) i2c: i2c_driver_delete(395): i2c driver install error
W (128600) messaging: cfg-audio-tmpl
squeezelite exited with error code 0

W (128610) messaging: cfg-audio-tmpl
squeezelite exited with error code 0

E (128610) messaging: cfg-audio-tmpl
Rebooting in 60 sec

E (128630) messaging: cfg-audio-tmpl
Rebooting in 60 sec
```

philippe44, 2023-09-02:
It's not a crash but you don't seem to have an LMS server running , right? The. You have to explicitly disable LMS connection by adding "-s -disable" on the command line

LaurentMe, 2023-09-02:
Makes sense. I read over that part of the documentation... Thank you!

closed 2023-09-02 by LaurentMe