sle118/squeezelite-esp32 #297: Artwork disappears

InZan3, 2023-09-11:
when playing Music over Airplay the artwork shows up for 1 Second and disappears, LMS is disabled. is it a Bug?

sle118, 2023-09-12:
you are the first one reporting this. Please provide more background. Which version are you running, the sequence of events to reproduce the issue, and ideally some logs.

InZan3, 2023-09-12:
Thanks for your reply :-)
My Hardware:
-ESP32-Wrover-IE
-SPI ST7789 Display 320x240
-PCM5102 DAC
I have tried several Versions, 2.12xx, 2.13xx and latest. Display stays blank on power on so I have to reboot to get start screen. Log says Timeout. sequence is: Power on, reboot, connect over Airplay, skip track. on every new track artwork shows up and disappear

sle118, 2023-09-12:
please share your NVS settings. 

- Go to the web UI
- Click "Credit" 
- Enable the "Show NVS Editor" checkbox; this allows viewing/changing nvs configuration even when not in recovery mode. 
- Click on the "NVS Editor" tab
- Scroll to the bottom and click "Download Config"
- Share the file here

I suspect you are possibly experiencing an issue with GPIO conflict. 

Also, you can get more logs by hooking up the board to a computer and, using the web installer (https://sle118.github.io/squeezelite-esp32-installer/), connect to the board's serial port. From there, you can view the logs: 
[image]

From there, logs can be downloaded

InZan3, 2023-09-12:
[nvs_config_squeezelite-e399c8_1694540771276json.txt](https://github.com/sle118/squeezelite-esp32/files/12589435/nvs_config_squeezelite-e399c8_1694540771276json.txt)
[esp-web-tools-logs (1).txt](https://github.com/sle118/squeezelite-esp32/files/12589437/esp-web-tools-logs.1.txt)

philippe44, 2023-09-12:
Is there anything displayed at all?

InZan3, 2023-09-12:
Yes, in the left top corner AirPlay is displayed as source, underneath the aktual Track and in the right corner the Time.
If i skip the Track, Artwork shows up, disapperears and 2 clefs shows up. In a log i read something like "no artwork send, using default. Everything else work as it should

philippe44, 2023-09-12:
Try to not resize it (don't add :1 after artwork)

philippe44, 2023-09-13:
Found & Fixed it - not in an official release yet but you can take the artefact of the last build

sle118, 2023-09-13:
https://github.com/sle118/squeezelite-esp32/commit/0002256630bf24488de1b9ac18ae9ac603bd261e

philippe44, 2023-09-13:
> Possibly fixed with [0002256](https://github.com/sle118/squeezelite-esp32/commit/0002256630bf24488de1b9ac18ae9ac603bd261e)
>
I don't understand

sle118, 2023-09-13:
I'm just cross referencing the commit here in case we need to search some day

InZan3, 2023-09-13:
Resize doesnt matter, same problem.
Sorry for the Question, how can i install the fix :-)

philippe44, 2023-09-13:
Go to the actions in GitHub, click the latest build, scroll down to the bottom snd download the version that suits your platform. Unzip the file and chose that squeezelite.bin from the manual update in the player webUI

InZan3, 2023-09-13:
Worked for me, Thank you very much :-)

closed 2023-09-14 by philippe44