sle118/squeezelite-esp32 #301: Loudness value it not used after a reboot

christianaubry, 2023-09-13:
First of all congratulation for your great work!

Changing the loudness value to 8 for example and applying does not save it. Both changing in the Audio tab or in the NVS Editor. After a reboot it is back to 0

Using fw: I2S-4MFlash-1532, mode: Squeezelite-ESP32

sle118, 2023-09-13:
Please use the appropriate template for submitting this type of issue.

Renber78, 2023-09-14:
Indeed, I also see this behavior.

- Change with NVS Editor -> value not saved after reboot
- Change with LMS Server (squeezeESP32 plugin) -> value saved after reboot

Tested with new version 1539

philippe44, 2023-09-14:
It's not exactly that. It's that if you change it "manually" in NVS, then LMS will overwrite it. The value from NVS is not sent to LMS at boot.

sle118, 2023-09-14:
Consider the NVS/Audio settings from the GUI as a way to do a temp override when not close to a LMS

closed 2023-09-16 by philippe44