sle118/squeezelite-esp32 #308: Bluetooth does not work / device not visible over bluetooth

hackintoshlover, 2023-09-21:
I have flashed the i2s firmware from https://sle118.github.io/squeezelite-esp32-installer/ to an esp32a1s audio kit.
The board boots up.
I connected to the webpage and did wifi configuration.
From the hardware section i selected the esp32 ac101 config > save > apply
[image]

From the Audio Section i changed device type to bluetooth, disabled squeezlite >save>apply.
[image]
[image]

But upon scanning for bluetooth devices, the device is not showing up.
I just keep getting timeout in status page.
[image]

Also when the device reboots / i manually reset it, there is some artwork displayed in the top right section which disappears in seconds.
[image]

Did i mess up?

sle118, 2023-09-21:
Right now, the UI cannot be used to disable squeezelite, as squeezelite needs to be alive for the audio stack to work. This being a squeezelite centric project, it wasn't a valid user case initially, but @philippe44 made some recent changes to allow this. See: https://github.com/sle118/squeezelite-esp32#additional-configuration-notes-from-the-web-ui

You will likely need to do this through the NVS editor, which can be enabled from the credits tab.

hackintoshlover, 2023-09-22:
Tried all of the above, nothing works.
Even the LMS server does not find the device over network.
Could you help with the following:
1) Can the device be used as a simple bluetooth speaker? if yes, how to configure? (since trying all settings, visiting LMS forum and trying all settings, nothing works)
2) Why do i keep getting timeout error on the status page?
3) Why does the artwork (headphone, speaker, etc logo) on top right corner disappear after few seconds?
4) Any step by step guide which is not available over internet?

Thanks for the wonderful project, just frustrated with it not working and want to get it to work.

hackintoshlover, 2023-09-22:
> Tried all of the above, nothing works. Even the LMS server does not find the device over network. Could you help with the following:
>
> 1. Can the device be used as a simple bluetooth speaker? if yes, [...]

The LMS server can now find the device, issue is while specifying server do not specify port.
1) Able to connect to bluetooth but packet drops, hence no sound.
2) Timeout eroor is gone.
3) Still disappears.
4) Need to create one i guess.

I am getting no audio over headphones or speaker though.

philippe44, 2023-09-22:
When you ask if it can be used as a simple speaker, I'm not sure then I follow what you want to do reading the rest. 

So you want to use (e.g.) your smartphone to send audio to the device or do you want the device to output its audio to a BT speaker? 

It does both, the latter being more a side project. Now if all you want is a BT receiver, I think there are better project for esp32 than ours.

hackintoshlover, 2023-09-22:
I know it does both, the problem is it's not working in my setup.
1) I am able to connect to bluetooth (yes, like a simple apeaker) but when i play audio there is no sound and i get the following error:
[image]

2) When i connect it to Logitech media server running on raspberry pi, it finds the esp32.
But on playing audio i get no sound, also controls don't have any effect.
[image]

sle118, 2023-09-22:
@hackintoshlover since you have a known a1s board, have you tried the various options for this board from the UI?

hackintoshlover, 2023-09-22:
Hi,
Thanks for your reply.
I have tried all parameter options in the web ui and also tried it over the commandline interface but no luck.
My board is Esp32a1s v2.2 with AC101.
I have no sound.
Will share logs shortly.

sle118, 2023-09-22:
> Hi, Thanks for your reply. I have tried all parameter options in the web ui and also tried it over the commandline interface but no luck. My board is Esp32a1s v2.2 with AC101. I have no sound. Will share logs shortly.

also share your NVS configuration

hackintoshlover, 2023-09-23:
Hi,
Attached is the NVS editor config and Console log.
[esp-web-tools-logs.txt](https://github.com/sle118/squeezelite-esp32/files/12706306/esp-web-tools-logs.txt)
[nvs_config_squeezelite-752520_1695468053600json.txt](https://github.com/sle118/squeezelite-esp32/files/12706307/nvs_config_squeezelite-752520_1695468053600json.txt)

philippe44, 2023-09-23:
Log says it's not an ac101 codec, so no sound indeed

hackintoshlover, 2023-09-23:
Ok so what should i do?
I am sure it's an ac101 because the datasheet mentions the same and using esp idf it works with ac101 codec.

PS: Where does it say in the log that it's not an ac101.

philippe44, 2023-09-23:
Timestamp 3234. This board probably has another wiring for it's i2s or you have something blocking the i2s gpio. They are a mess, to be polite

sle118, 2023-09-23:
> It's working now. I had a wrong datasheet. Tried esp32a1s v2.2 variant 2. How do i control volume using gpio?

Please share your working NVS so I can compare and possibly add as a known board

hackintoshlover, 2023-09-24:
Attached is working NVS config.

[nvs_config_squeezelite-752520_1695536641363json.txt](https://github.com/sle118/squeezelite-esp32/files/12707941/nvs_config_squeezelite-752520_1695536641363json.txt)

1) Could you please help me with GPIO controls for volume and playback?

2) Can chromecast sudio support be added so that it works natively with android as airplay?

philippe44, 2023-09-24:
> Attached is working NVS config. [...]
The README is your best friend for the GPIO config. 
Chromecast cannot be added, it's a limitation of Chromecast itself. Device must run a certified undocumented Google system which is much more CPU and memory hungry.

hackintoshlover, 2023-09-24:
I have gone thorough the readme before asking you about the same.
The issue is i don't understand where to paste the commands.

1) So a possible config would be

set_GPIO: 21=amp,22=green:0,39=jack:0
a button mapping:
 [{"gpio":5,"normal":{"pressed":"ACTRLS_TOGGLE"}},{"gpio":18,"pull":true,"shifter_gpio":5,"normal":{"pressed":"ACTRLS_VOLUP"}, "shifted":{"pressed":"ACTRLS_NEXT"}}, {"gpio":23,"pull":true,"shifter_gpio":5,"normal":{"pressed":"ACTRLS_VOLDOWN"},"shifted":{"pressed":"ACTRLS_PREV"}}]

where to do the above?
I see

2) "It can also send audio to UPnP, Sonos, ChromeCast and AirPlay speakers/devices." from readme. What does it mean exactly?

3) In the audio section of webui, there is a bluetooth audio option.
If i correctly understand, i guess it helps in selcting output audio source. I2S and spdif are on-board. How to use the bluetooth option?

hackintoshlover, 2023-09-24:
Hi, Just to check i changed audio setting in webui from I2s to bluetooth and started receiving error as in the attached logs and the setup no longer works.
Even after changing it back from bluetooth to i2s it still does not work.
[esp-web-tools-logs.txt](https://github.com/sle118/squeezelite-esp32/files/12708082/esp-web-tools-logs.txt)

philippe44, 2023-09-24:
> Hi, Just to check i changed audio setting in webui from I2s to bluetooth and started receiving error as in the attached logs and the setup no longer works. Even after changing it back from bluetooth [...]
The squeezelite command line is erroneous, as the log says

hackintoshlover, 2023-09-24:
Yes, i saw that.
How to solve it except re-uploading the squeezelite firmware.
I did not do anything apart from changing the audio options.

philippe44, 2023-09-24:
the "it" refers to the LMS capabilities, no the squezeesp32 capabilities

hackintoshlover, 2023-09-24:
If i get it correctly, you mean LMS is throwing error?
If that's the case why is nothing working, not even bluetooth.
I just did a code re-upload everything is fine now.
Strange.

hackintoshlover, 2023-09-24:
> I have gone thorough the readme before asking you about the same. The issue is i don't understand where to paste the commands.
>
> 1. So a possible config would be
>
> set_GPIO: [...]

I tried adding the config from (1) to actrls_config in webui editor, but no success buttons don't work.

sle118, 2023-09-24:
> Hi, Just to check i changed audio setting in webui from I2s to bluetooth and started receiving error as in the attached logs and the setup no longer works. Even after changing it back from bluetooth [...]

Have a careful look at the log, as it provides all the clues to the error you are getting. More specifically, and just as the Linux/Windows versions of the squeezelite would behave, the logging option needs a valid input. In your case, it was set to 'true', which squeezelite does not recognize. The option should be set to something like: all=info under logging level.

Additionally, you can go to the NVS tab and look for the autoexec1 value. This is the command line that will be used when executing squeezelite. You can change it there directly if you want, the web UI is there as an attempt to simplify and document the options.

As for the Bluetooth option, it is there for users with bare WROVER boards with no DAC output. It's technically possible to stream to a Bluetooth speaker from the WROVER as the source, for example to playback some LMS content, but it is not a recommended setup as it doesn't have as much robustness.

hackintoshlover, 2023-09-24:
> Have a careful look at the log, as it provides all the clues to the error you are getting. More specifically, and just as the Linux/Windows versions of the squeezelite would behave, the logging [...]

Hi,
I tried the bluetooth audio option but remained clueless.
On switching to bluetooth the esp scans for devices over ble as seen in serial terminal, but how to initiate connection?

hackintoshlover, 2023-09-24:
> The README is your best friend for the GPIO config.
>
> Chromecast cannot be added, it's a limitation of Chromecast itself. Device must run a certified undocumented Google system which is much more CPU [...]

I definitely buy the chromecast point.
Espressif released the matter sdk, qhile developing with the same, it qorks only with wrover boards with upwards of 16mb memory.
They have provided google home, apple homekit certification.

Probably that might help.

sle118, 2023-09-24:
> Hi, I tried the bluetooth audio option but remained clueless. On switching to bluetooth the esp scans for devices over ble as seen in serial terminal, but how to initiate connection?

in the audio section, under ```Bluetooth Audio Output Options``` you set the name and PIN code of the device you want the WROVER to connect to, then you choose "Bluetooth" as the output. That's it. The WROVER will try to connect to the BT speaker and play audio over it.  But this is more of an experimental feature and _**not recommended**_ as we don't regression test this your mileage will vary.

philippe44, 2023-09-24:
> I definitely buy the chromecast point.
> 
> Espressif released the matter sdk, qhile developing with the same, it qorks only with wrover boards with upwards of 16mb memory.
> 
> They have provided google home, apple homekit certification.
> 
> 
> 
> Probably that might help.
>
I don't understand what you mean by "that might help". AFAIK, matter is for provisioning and controlling smart home devices but that does not make a device compliant with the Chromecast protocol. 

Especially, CC run a small web server-like runtime environment where you can load and run your Html +JS "applications". These must be Google signed and hosted and the CC devices request them from Google server to load them in from a reference ID. So you can't just create your own CC player.

hackintoshlover, 2023-09-24:
> in the audio section, under `Bluetooth Audio Output Options` you set the name and PIN code of the device you want the WROVER to connect to, then you choose "Bluetooth" as the output. That's it. The [...]

That's exactly what i did before posting the question.
I had my jbl go speaker in pairing mode, the esp32 scanned it but was not able to connect to it. So i just dropped the idea.

hackintoshlover, 2023-09-24:
> I don't understand what you mean by "that might help". AFAIK, matter is for provisioning and controlling smart home devices but that does not make a device compliant with the Chromecast protocol. [...]

What i mean is espressif has provided free (for development purpose) certification for devices. Since matter devices can only be connected if they are certified by the specific device vendor, the espressif repo might help in dealing with certification of devices. Rest you always know better.

Any plans of adding snapcast support though?

philippe44, 2023-09-24:
We are talking about a totally different type of certification. Do anyway, Chromecast is impossible on this device and our system. I don't have plans for snapcast

sle118, 2023-09-24:
There is at least one project that I know of for snapcast on the ESP32 and they are building it from the ground up for this very specific purpose. Our goal with this project was always to replace a squeezebox device running in a LMS environment and everything else (Spotify, airplay and Bluetooth) was a nice to have. Our efforts were better spent in supporting various displays and hardware options. 

At the end of the day, it would probably be easier to add airplay to a native snapcast player than adding snapcast to our project

hackintoshlover, 2023-09-25:
Thanks for the help.
Eagerly waiting for line-in support on esp32a1s

closed 2023-09-25 by hackintoshlover