sle118/squeezelite-esp32 #309: Cspot disconnecting

Renber78, 2023-09-23:
When I exit Spotify during playback, the screen returns to LMS after around 1 minute. The Spofify spot is accessible and can be reconnected (but not immediately)

When I exit Spotify and there is no current playback (paused), the screen remains frozen on the old playback and does not return to LMS. The Spotify spot is accessible and cannot be reconnected.

FYI, I exit Cspot mode by changing the player on the Android app.

Su csopt, I also often encounter this error when connecting but this does not prevent it from working:

`I TrackQueue.cpp:252: Got audio key
I TrackQueue.cpp:275: Received access key, fetching CDN URL...
E (250638) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (250638) task_wdt:  - IDLE (CPU 1)
E (250638) task_wdt: Tasks currently running:
E (250638) task_wdt: CPU 0: IDLE
E (250638) task_wdt: CPU 1: cspot_player
E (250638) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401E55BE:0x3FFBED40 0x40084875:0x3FFBED60 0x40272A2F:0x3FFBD5B0 0x400DD4B9:0x3FFBD5D0 0x40280EF7:0x3FFBD5F0

E (250638) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008F90D:0x3FFBF570 0x40084875:0x3FFBF590 0x400940F1:0x3FB5A000 0x400821C3:0x3FB5A010 0x4008F9A0:0x3FB5A030 0x401F7D85:0x3FB5A050 0x401EE155:0x3FB5A070 0x401ECA3F:0x3FB5A0B0 0x40165257:0x3FB5A0D0 0x40164E39:0x3FB5A0F0 0x40275959:0x3FB5A110 0x402360A9:0x3FB5A130 0x40164D12:0x3FB5A150 0x40164D4A:0x3FB5A1C0 0x401617C2:0x3FB5A210 0x4015B47A:0x3FB5A280 0x4015B492:0x3FB5A2A0 0x4018CAF8:0x3FB5A2C0 0x4018D20D:0x3FB5A2E0 0x40190316:0x3FB5A340 0x4015BC5F:0x3FB5A390 0x4010805E:0x3FB5A440
`

It also reboots from time to time:

`I TrackQueue.cpp:252: Got audio key
I TrackQueue.cpp:275: Received access key, fetching CDN URL...

***ERROR*** A stack overflow in task downloader has been detected.

Backtrace:0x40081a79:0x3fff6900 0x40098505:0x3fff6920 0x40098ac5:0x3fff6940 0x40099741:0x3fff69c0 0x400985fc:0x3fff69f0 0x400985ae:0xa5a5a5a5 |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
`

philippe44, 2023-09-24:
I just fixed the overflow issue

philippe44, 2023-09-24:
cspot pause and disconnect fixed as well

Renber78, 2023-09-24:
I tested and no longer noticed the problems mentioned. Well done !

Connecting after disconnecting is always difficult, I don't know if this is normal?

philippe44, 2023-09-24:
It depends what you mean by difficult

Renber78, 2023-09-24:
That you have to wait a certain amount of time between two connections ~20-30sec

philippe44, 2023-09-24:
I don't see that. I can connect  play, switch to another connect device then connect back to this one in a row. Moving back to LMS idle screen is a different thing if this is what you refer to.

Renber78, 2023-09-24:
I just did the same tests as this morning and strangely it works now... I can no longer reproduce it. Maybe it was a problem with the Android app.
I had no logs on the ESP, just that the connection was taking its time.

closed 2023-09-24 by Renber78