sle118/squeezelite-esp32 #315: Bug when changing wifi network

Renber78, 2023-09-27:
It seems to me that there is a bug when changing wifi.
I moved my hardware to another location (another wifi), I got the page 192.168.4.1 and I was able to configure the new wifi. It worked. When I came home with it, I wanted to put the old wifi back on. The page 192.168.4.1 is there, I was able to choose my wifi, the connection is made but when I reboot, it still tries to connect to the previous wifi.
Weird thing:
- When scanning the previous wifi appears in the list even though it is not present where I am staying.
- The backup access point is still visible when connected to my home network and both are accessible.

I'm completely stuck, I've tried everything. Even in the telnet console with "join" it doesn't work, each time it reboots it goes back to the old wifi.
Only a firmware flash via RS-232 saves the device.

In the logs, the only abnormal thing I see is:

`E (97215) network: network manager Unhandled Event NETWORK_WIFI_CONFIGURING_ACTIVE_STATE(WIFI_CONFIGURING_STATE) [EN_CONNECTED]
`
`E (99285) network: network manager Unhandled Event NETWORK_WIFI_CONFIGURING_ACTIVE_STATE(WIFI_CONFIGURING_STATE) [EN_GOT_IP]`

I have the impression that it cannot overwrite old wifi data.

sle118, 2023-09-29:
Please share the full log, as this feels like peeking through a very small window and doesn't give me all surrounding context

sle118, 2023-10-01:
Going to close of for now. Feel free to reopen with more info. Thank you

closed 2023-10-01 by sle118

hackintoshlover, 2023-12-04:
Facing the same problem.
Attached is the log.
Help needed.
Thanks.
[esp-web-tools-logs.txt](https://github.com/sle118/squeezelite-esp32/files/13543873/esp-web-tools-logs.txt)