sle118/squeezelite-esp32 #317: boot loop on entering led_vu_config

NenuSB, 2023-09-29:
Whenever I enter a setting for ws2812 in led_vu_config, the ESP reboots a few times until it erases the NVS parameters. I have tried multiple GPIOs. And with or without a DAC configured.

firmware is I2S-4MFlash-1539
lms plugin is v0.501
hw is ESP32 WROVER devkit
DAC doesn't seem to matter, it happens with or without I2S DAC attached and/or configured.
nothing else connected

A clean install of firmware, with only led_vu_config entered results in this behaviour.
[esp-web-tools-logs.txt](https://github.com/sle118/squeezelite-esp32/files/12770047/esp-web-tools-logs.txt)

sle118, 2023-09-29:
Please follow this model for submitting a bug/issue. It allows us to better spend the little time we have for this project:

https://github.com/sle118/squeezelite-esp32/issues/new?assignees=&labels=&projects=&template=bug_report.md&title=

wizmo2, 2023-09-29:
Theres fix in place, but its not in the officials release yet.  You could try to download the latest build.

philippe44, 2023-09-29:
I think it's time for a release

NenuSB, 2023-09-30:
It's working now in 1588. Thanks!

closed 2023-09-30 by NenuSB