sle118/squeezelite-esp32 #319: text not visible in 1588 web interface

NenuSB, 2023-09-30:
The label text are not visibile with update 1588.
Mainly on the Wifi and NVS pages, both in Firefox and Chrome. 

This was OK in 1539.

[image]

sle118, 2023-09-30:
@NenuSB when opening a new issue, please follow the provided template as much as possible as it provides information we need upfront. For example, the template explains how to get the logs, which is critical in troubleshooting

sle118, 2023-09-30:
I don't have a quick fix for this. for now I recommend rolling back the version until I resolve the issue.

sle118, 2023-10-01:
This release, https://github.com/sle118/squeezelite-esp32/commit/f87b3adec2f13dc343f75207f0938b10dd52cab0, should fix the text color issue.

closed 2023-10-01 by sle118