sle118/squeezelite-esp32 #320: led_vu_config enhancements

NenuSB, 2023-09-30:
Suggestions for improvement in the VU meters with addressable leds:

    add an LED gain setting in the ESP32 settings in LMS:
    * I never have a player even close to 100% volume, usually~50%, but would like to see more then half of the leds do something.
    * adding extra dummy leds won't work, because the center for L/R would shift
    it doesn't seem to work if no DAC is configured. That's fine, I guess, but would be good to mention in the readme?

I am new to Github; so please tell me if I am doing something wrong with these submissions and/or if I shouldn't even be making them.

sle118, 2023-09-30:
I suggest @wizmo2 have a look at this

wizmo2, 2023-09-30:
<QUOTE>add an LED gain setting in the ESP32 settings in LMS</QUOTE>
I'll look into it.

<QUOTE>it doesn't seem to work if no DAC is configured</QUOTE>
What do you use as audio out?

NenuSB, 2023-09-30:
> What do you use as audio out?

Nothing. Suppose you want to use it purely as a VU meter, synched to some other squeezeboxes. But I don't think that's much of a use case, and I can always add a dummy entry anyway. 
So I don't think that is worth fixing, but maybe a mention in the readme.

sle118, 2023-10-01:
the default output is i2S and since the data is one way out from the ESP32, there's no need for an actual DAC to be connected.  Now this use case is yet another creative way to use our project!

NenuSB, 2023-10-01:
> there's no need for an actual DAC to be connected. 

But it is still necessary to define one in the settings, I found.
I entered something like "dac_config: model=I2S,bck=12,ws=32,do=33" without physically connecting anything. 
If you just want to use it for the VU, you should have plenty of spare GPIOs anyway, I guess.

sle118, 2023-10-03:
@NenuSB yes. To me this is a "hacky" way to use the firmware, since this wasn't one of the use cases we had in mind. I'm glad if it works for you!