sle118/squeezelite-esp32 #322: Balance option not available anymore

Eddie46848, 2023-10-02:
To help us resolve your issue as quickly as possible, p

### Describe the bug
Balance option not available anymore

### Preliminary Information

1. **Firmware Version**: 
2. https://github.com/sle118/squeezelite-esp32/releases/tag/I2S-4MFlash.16.1539.master-v4.3

### Hardware Details

Please describe your hardware setup:

ESP muse mn cast

### NVS Settings

{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "Muse",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "boutons",
  "airplay_name": "Muse-Airplay-f8e1b8",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "musemuse",
  "ap_ssid": "Muse-f8e1b8",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n \"Kitchen\"",
  "bat_config": "",
  "bt_name": "Muse-f8e1b8",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\t\"volume\":\t32767,\t\"deviceName\":\t\"Muse-f8e1b8\",\t\"bitrate\":\t160}",
  "dac_config": "model=I2S,bck=5,ws=25,do=26,mute=33:0,sda=18,scl=23",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "I2C,width=128,height=64,address=60,driver=SH1106,HFlip,VFlip",
  "enable_airplay": "N",
  "enable_bt_sink": "Y",
  "enable_cspot": "N",
  "equalizer": "10,8,8,7,5,4,1,3,9,11",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "Muse-f8e1b8",
  "i2c_config": "scl=23,sda=18,speed=400000,port=0",
  "jack_mutes_amp": "y",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "https://github.com/sle118/squeezelite-esp32/releases",
  "rotary_config": "A=32,B=19,SW=0,volume,knobonly",
  "set_GPIO": "12=ir:0",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "bck=5,ws=25,do=4",
  "stats": "N",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "Y"
}

### Issue Description

1. **Observed Behavior**: option to change the balance not available anymore, can't see it's been removed in the chagelog
2. 
3. **Steps to Reproduce**: Still available on other squeezebox devices running lms 8.4

sle118, 2023-10-02:
@Eddie46848 which version of the LMS plugin are you running?

Eddie46848, 2023-10-02:
Hi. Version v0.501. Thanks

philippe44, 2023-10-03:
Probably something with the LMS side of things then. There is no recent wrt balance and it still works as usual on my system.

michaelherger, 2023-10-03:
I think that if there was a change in LMS, you @philippe44 would have had to be the author 😂. I usually don't touch this part of LMS.

@Eddie46848 - what LMS version are you using?

Eddie46848, 2023-10-03:
Hi
Version: 8.4.0 - 1695847275
Thanks

michaelherger, 2023-10-03:
The ESP muse isn't a stereo device, is it? Would it ever have balance settings?

Eddie46848, 2023-10-03:
Yes it is  stereo it's the muse mn cast, the balance option has been there since day 1. But noticed a difference in the sound recently and realised it wasn't there anymore. I use it for the kitchen where the speakers are 3 meters apart.

philippe44, 2023-10-03:
Can you give  a screenshot?

Eddie46848, 2023-10-03:
[image]
[image]
[image]

The one without balance is the mn cast , the one with is a squeezebox touch. Screenshot of mn cast also.

Eddie46848, 2023-10-03:
Hi 
Sorry disregard screenshot, I'd swapped the device before I came out to my squeeze receiver for testing. I'll swap back over when I get home.
Thanks

Eddie46848, 2023-10-04:
Hi
So when I got home I reloaded the squeezelite firmware, balance is now there as a option. 
Thanks everyone

closed 2023-10-04 by Eddie46848