sle118/squeezelite-esp32 #327: Is this I2S Dac TAS5805M supported?

DerPicknicker, 2023-10-07:
Hi guys,

I am planing to build a device with this great software. But I am not sure if the software supports that specific i2s dac (TAS5805M)... I am planning to buy one of these: https://www.tindie.com/products/sonocotta/louder-esp32/

Another question: The central unit is the LMS. The multiroom-sync is done on the LMS, right? So if I stream to LMS I can select the device inside the webUI, the esp32 is not involved in that right? How long does it take to start the stream in LMS and the esp32 starts playing?

Best regards

sle118, 2023-10-07:
> Hi guys,
> 
> I am planing to build a device with this great software. But I am not sure if the software supports that specific i2s dac (TAS5805M)... I am planning to buy one of these: https://www.tindie.com/products/sonocotta/louder-esp32/
@philippe44 added the ability to send commands to DAC for models that aren't supported or of the box. So it is definitely doable with a bit of configuration. Alternatively, I think @philippe44 still has some SqueezeAmp in stock, which he sells at cost.

> Another question: The central unit is the LMS. The multiroom-sync is done on the LMS, right? So if I stream to LMS I can select the device inside the webUI, the esp32 is not involved in that right? How long does it take to start the stream in LMS and the esp32 starts playing?
You are correct. When LMS is used to stream audio, you can either use LMS web UI or one of the excellent control applications for Android or Apple devices. Our firmware boots in under 7 seconds and is ready to playback from there. The delay for playback from LMS is short but depends on your source. Local playback is relatively quick with very little delay. Streaming from Spotify involves a couple more things on the server but not a lot more delay.

It is possible to synchronize several players and LMS sends corrections when needed. Personally, I sync 3 zones on a regular basis, 2 of which are in the same space and you could swear that they are hard wired together.

DerPicknicker, 2023-10-07:
Hi @sle118 ...

thank you for the response.. I want to know if I buy one if you would support me to get this thing running. 

Do you know how much the SqueezeAmp is? 

Do you think I can build a 2.1 System wirelessly? So that I can group a subwoofer, two smaller speakers as one group and the sub is playing the lower frequencies and the other speaker are playing the other frequencies? 

@philippe44 
 Is this board pre-soldered? How much are them and does it have an external antenna?

sle118, 2023-10-07:
> thank you for the response.. I want to know if I buy one if you would support me to get this thing running.
>
Since we don't have one of these devices, we can't really help, but we can certainly provide some guidance, although the Logitech forums are probably a better place to get some support. Others have configured different DAC themselves.

> Do you know how much the SqueezeAmp is?
Can't remember. You should have a read here: https://forums.slimdevices.com/forum/user-forums/3rd-party-hardware/108002-announce-the-squeezeamp-next-generation-squeezeboxen-amplifier-receiver

I think there are some infos towards the end.

> Do you think I can build a 2.1 System wirelessly? So that I can group a subwoofer, two smaller speakers as one group and the sub is playing the lower frequencies and the other speaker are playing the other frequencies?

I have not tried that and not sure how cross over would work, but I suspect it's possible. @philippe44?

> @philippe44 Is this board pre-soldered? How much are them and does it have an external antenna?
They are pre soldered, yes, but no external antenna. They are also perfect for turning a pair of old speakers into a battery powered portable wireless speakers (I built two of these for my own usage). You can stream Bluetooth, AirPlay and Spotify (in addition to LMS of course)

DerPicknicker, 2023-10-07:
I will buy one of the tindy ones.. It's quite cheap and for my usecase it's ideal. Should I create an GitHub issue or post it in the linked forum?

philippe44, 2023-10-07:
The SqueezeAMP is fully assembled and comes with a case https://github.com/philippe44/SqueezeAMP

DerPicknicker, 2023-10-07:
> The SqueezeAMP is fully assembled and comes with a case https://github.com/philippe44/SqueezeAMP

Could you give me a figure how much is anSquerzeAMP? I found a old number in the thread (around 60$) is that still valid?

Do you think a wireless 2.1 System is possible? I am still thinking about the Tindie Offer for me the shipment is quite low and the price 20€ incl. esp32 seems fair. Do you have an link where I can check the manual config? I searched here in the GitHub but I found only something for tested hardware... Thank you in advance!

philippe44, 2023-10-07:
I just did a batch a few weeks ago. It cost me 58USD per unit, that include a casing. Shipment is usually USD12. It's a non-profit thing, means I'm just passing the cost the PCB assembly house is charging me + taxes. You could do a 2.1 but that needs more one device. The TI chip on SqueezeAMP is not 2.1

DerPicknicker, 2023-10-07:
Hi @philippe44 ... 

Thank you for the response.. I will think about. For now I will test it first with a cheaper one. I meant that the 2.1 system is controlled by LMS so that LMS sends the Bass-Channel to a different esp32 than the 2 speaker channels.

DerPicknicker, 2023-10-08:
@sle118 .. 

I found a max98357a in my electronics box. Can I use this for testing it out?

sle118, 2023-10-08:
> @sle118 ..
> 
> I found a max98357a in my electronics box. Can I use this for testing it out?

Most definitely yes. These i2s boards don't have an active i2c communication interface and therefore don't require commands to be sent for them to work. Just hook it up to a WROVER and you're good to go. Since i2s is a one way protocol, you could also probably just wire up these in parallel (e.g. one for speakers and one for sub, although these don't provide much power), but experimenting is the best way to find out

DerPicknicker, 2023-10-09:
@sle118 ... 

Thank you for your great Response.. I will try that as soon as possible. For now I tested only two x86 Devices (Mac and windows) I am blown away: 
Bad Wifi signal but it's still totally in sync. I just had to increase the delay of one player about 6ms... but it's awesome. What a Cool tech.. Now I will never buy Sonos stuff. Its crazy how good that LMS handles that.. Now I am thinking about buying Squeezeboxes lol...

sle118, 2023-10-10:
Best advice is to go and scourge the forums as they are a massive pile of unstructured knowledge. 

Fun fact is that slimproto, which is at the heart of most features including sync, is close to 20 years old or so, and was so well thought of that it still works wonder today.
https://en.m.wikipedia.org/wiki/Slim_Devices

The main reason I started this port of squeezelite to the esp32 with the invaluable contribution of Philippe (seriously, the project wouldn't be closer to half of what it is without him) was that squeezebox devices were hard to find and expensive. So our goal was to come up with a platform that could totally replace original hardware while being accessible by enthusiasts for a low cost.

DerPicknicker, 2023-10-10:
Here the Logitech Squeezes are cheap. A touch one cost around 50$ - 80$ and Radio ones starting at 35$... 

Only the duet is quite expensive 120$.. This device looks still quite Cool. 

@sle118 Just for my understanding the S3 are not part of the official release. Does it have big impacts on the performance? I thought the CPU is basically the same.I have some S3 laying around. But my development setup is already migrated to IDF 5, so would it be possible IF the performance is really improved about the normal esp32 that you can send me a pre-build binary?

philippe44, 2023-10-10:
I've made a build for the s3 and all the code currently in works on s3, I have one device working. @wizmo is tweaking it. You need esp-idf 4.4 at minimum. There a *BIG* improvement in CPU because of the memory and if you chose Octal SPI and Octal SPIRAM, it's 1/3 of the esp32 CPU., but you lose BT.

DerPicknicker, 2023-10-10:
@philippe44 .. 

I will try to build that. If not I will stay on esp32. My model has 16mb Flash and 8mb ram. Great model and cheap (8$..)

Do you know how long it will take until the S3 is also available as an "official" release?

philippe44, 2023-10-10:
It's a matter of finding the time for @sle118 and myself. But it should easily build - I'm building it everyday with the same code

haldi4803, 2023-10-10:
I've got an ESP-Louder here.
Tried to flash generic image via Webinstaller. But that bootloops.
I assume cuz it comes with a WemosD32 included which as a WROOM module but the ESP32 Louder has 4MB PSRAM on the AMP Board so you need to included that while building i assume.

Here is log after boot:
`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (27) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (27) boot: compile time 06:43:10
I (27) boot: chip revision: v3.1
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (104) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=59430h (365616) map
I (229) esp_image: segment 1: paddr=00069458 vaddr=3ffbdb60 size=04850h ( 18512) load
I (235) esp_image: segment 2: paddr=0006dcb0 vaddr=40080000 size=02368h (  9064) load
I (238) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=d27c0h (862144) map
I (481) esp_image: segment 4: paddr=001427e8 vaddr=40082368 size=12730h ( 75568) load
I (506) esp_image: segment 5: paddr=00154f20 vaddr=400c0000 size=00068h (   104) load
E (506) esp_image: Image length 1331120 doesn't fit in partition length 1310720
E (511) boot: Factory app partition is not bootable
E (517) esp_image: image at 0x150000 has invalid magic byte (nothing flashed here?)
E (525) boot: OTA app partition slot 0 is not bootable
E (531) boot: No bootable app partitions in the partition table`

gonna try to build my own with the proper pins...
<html><body>
<h3 tabindex="-1" id="user-content-louder-esp" dir="auto"><a class="heading-link" href="https://github.com/sonocotta/esp32-audio-dock#louder-esp">Louder ESP<svg class="octicon octicon-link" viewBox="0 0 16 16" version="1.1" width="16" height="16" aria-hidden="true"></svg></a></h3>

  | I2S CLK | I2S DATA | I2S WS | PSRAM CE | PSRAM CLK | TAS5805 SDA | TAS5805 SCL | TAS5805 PWDN | TAS5805 FAULT
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
ESP32 | 26 | 22 | 25 | 16 | 17 | 21 | 27 | 33 | 34

</body>
</html>

DerPicknicker, 2023-10-10:
@haldi4803 ... So did you get it running with audio..?

haldi4803, 2023-10-10:
This is way over my head.... but as far as i get it we simply need to pull pin33 low and then high again to get the DAC to boot.
But i have absolutely no idea how to do that on SqueezeElite-ESP32
Seems like i accidently set pin33 as mute.... but it's not like there is a damn mute button on Bluetooth or Airplay... so cant test.

DerPicknicker, 2023-10-10:
Ping @sle118 & @philippe44

haldi4803, 2023-10-10:
INDEED!
Found it. 
set_GPIO` 33=vcc` turned it on!

```
I (2372876) messaging: i2cdetect

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- 2d -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

------------------------------------------------------------------------------------
Detected the following devices (names provided by https://i2cdevices.org/addresses).
45 [2dh]- CAP1188 AD5248 AD5251 AD5252 CAT5171
--------
```

45 was correct. (found that in the [other Code](https://github.com/sonocotta/esp32-tas5805m-dac/blob/main/tas5805m.hpp)) `#define TAS5806M_ADDRESS 0x2D /*!< 7-bit address with a 15k pull up resistor */`

But it seems like the "init" is only the booting up...
We still need the "begin" part as well.

```
void tas5805m::begin()
{
    /* sound is ready */
    log_d("I2S is ready, setup of the audio amp begin");
    vTaskDelay(1000 / portTICK_RATE_MS);

    /* set PDN to 1 */
    gpio_set_level(TAS5806M_GPIO_PDN, 1);
    vTaskDelay(100 / portTICK_RATE_MS);

    log_i("Setting to HI Z");
    ESP_ERROR_CHECK(_write_byte(0x03, 0x02));
    vTaskDelay(100 / portTICK_RATE_MS);

    log_i("Setting to PLAY");
    ESP_ERROR_CHECK(_write_byte(0x03, 0x03));

    vTaskDelay(100 / portTICK_RATE_MS);

    int ret;
    uint8_t h70 = 0, h71 = 0, h72 = 0;
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, TAS5806M_ADDRESS << 1 | WRITE_BIT, ACK_CHECK_EN);
    i2c_master_write_byte(cmd, 0x70, ACK_CHECK_EN);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(I2C_TAS5806M_MASTER_NUM, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    if (ret != ESP_OK)
    {
        log_e("I2C ERROR");
    }
    vTaskDelay(1 / portTICK_RATE_MS);
    cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, TAS5806M_ADDRESS << 1 | READ_BIT, ACK_CHECK_EN);
    i2c_master_read_byte(cmd, &h70, ACK_VAL);
    i2c_master_read_byte(cmd, &h71, ACK_VAL);
    i2c_master_read_byte(cmd, &h72, NACK_VAL);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(I2C_TAS5806M_MASTER_NUM, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);

    log_d("0x70 Register: %d", h70);
    log_d("0x71 Register: %d", h71);
    log_d("0x72 Register: %d", h72);
}
```

philippe44, 2023-10-10:
You need to reproduce this using dac_controlset entry. The syntax is in the README

haldi4803, 2023-10-10:
Yeah...... plx halp.

```
{ <command>: [ {"reg":<register>,"val":<value>,"mode":<nothing>|"or"|"and"}, ... {{"reg":<register>,"val":<value>,"mode":<nothing>|"or"|"and"} ],
  <command>: [ {"reg":<register>,"val":<value>,"mode":<nothing>|"or"|"and"}, ... {{"reg":<register>,"val":<value>,"mode":<nothing>|"or"|"and"} ],
  ... }
```

Taking by example the SourceCode for others.
```
		config DAC_CONTROLSET
			string
			default "{ \"init\": [ {\"reg\":41, \"val\":128}, {\"reg\":18, \"val\":255} ], \"poweron\": [ {\"reg\":18, \"val\":64, \"mode\":\"or\"} ], \"poweroff\": [ {\"reg\":18, \"val\":191, \"mode\":\"and\"} ] }" if TWATCH2020
			default "{\"init\":[ {\"reg\":0,\"val\":128}, {\"reg\":0,\"val\":0}, {\"reg\":25,\"val\":4}, {\"reg\":1,\"val\":80}, {\"reg\":2,\"val\":0},	{\"reg\":8,\"val\":0}, {\"reg\":4,\"val\":192},	{\"reg\":0,\"val\":18}, {\"reg\":1,\"val\":0}, {\"reg\":23,\"val\":24}, {\"reg\":24,\"val\":2}, {\"reg\":38,\"val\":9}, {\"reg\":39,\"val\":144}, {\"reg\":42,\"val\":144}, {\"reg\":43,\"val\":128}, {\"reg\":45,\"val\":128}, {\"reg\":27,\"val\":0}, {\"reg\":26,\"val\":0}, {\"reg\":2,\"val\":240}, {\"reg\":2,\"val\":0},	{\"reg\":29,\"val\":28}, {\"reg\":4,\"val\":48}, {\"reg\":25,\"val\":0}, {\"reg\":46,\"val\":33}, {\"reg\":47,\"val\":33} ]}" if MUSE
			default ""	
```
so we got the init command which sends the value 128 into the registry nr 41 ? and then 255 into reg 18

Going by his code to set hi Z
`    ESP_ERROR_CHECK(_write_byte(0x03, 0x02));` which uses Registry 0x03 aka `#define TAS5806M_DEVICE_CTRL_2_REGISTER 0x03`  which should be Register 3 on offset 3h according to documentation on chapter 7.6.1.3
Setting it to 10 = hi-Z and 11 = Play or do i set that to 2 = hi-Z and 3 = Play as it does not support hex?
the 5ms delay that is required there cannot be reproduced but is just a "let's hope thats slow enough" ?

That would end in:  `init:[{"reg":3, "val":2}, {"reg":3, "val":3}]`

BTW i was wrong about set_GPIO.... after a reboot that does not work anymore -.- i2cdetect doesn't find the TAS5805M anymore. No matter what i do.

philippe44, 2023-10-10:
I can extend the json parser to include delay and gpio setting

philippe44, 2023-10-10:
Ok, it's done : you have 2 extra command words
- "delay":<value>
- "gpio":<gpio>,"level":<0|1>
See log to verify it works as expected, I've not tested it

philippe44, 2023-10-10:
You need to get the build artefacts directly, version will not be released yet

anabolyc, 2023-10-11:
Hey Folks

I'm the author of [Louder-ESP](https://www.tindie.com/products/sonocotta/louder-esp32/) and happy to help. 
I actually did run squeezelite-esp32 on it, and it worked great, but it took some effort, and I didn't document it unfortunately.

First of all, I'm happy to provide few samples to developers for delivery cost only (~$5), if any of you have commitment to actually run it. I would like to test it myself as well )

Second, there is [software sample](https://github.com/sonocotta/esp32-audio-dock/tree/main/firmware/esp32-i2s-bare) that includes all the necessary to bootstrap TAS5805 DAC. It is essentially startup pin bootstrap + sequence of I2C commands, nothing fancy.

Last point. Board provides external power connector (9..25V). It is not necessary for ESP32 when it is connected to USB, however it should be connected if you expect DAC to startup properly. Clearly you cannot pull ~50W of power of USB 
[image]

DerPicknicker, 2023-10-11:
@anabolyc ...

I want to order some of them. But I am not a developer so I will do it on Tindie. If it's out of stock are you planning to re-stock them? Also in Future with an ESP32-S3...? 

Another great option would be USB-C so you can get up to 20V via USB... Just in case you're planning to update the board.

anabolyc, 2023-10-11:
> @anabolyc ...
> 
> I want to order some of them. But I am not a developer so I will do it on Tindie. If it's out of stock are you planning to re-stock them? Also in Future with an ESP32-S3...?

I have 3 in stock still. With S3 so far not so many code samples with I2S, so I didn't foresee a lot of interest.

> 
> Another great option would be USB-C so you can get up to 20V via USB... Just in case you're planning to update the board.

Not that easy. You need negotiation IC to actually request it from charger. Which should support it as well, so.. maybe some day:)

haldi4803, 2023-10-11:
> Another great option would be USB-C so you can get up to 20V via USB... Just in case you're planning to update the board.

Won't happen.
The USB port is made for Data and not Power.
If you want to Power via USB-c that's no issue. There are [Adapter Cables](https://de.aliexpress.com/item/1005005768434193.html?spm=a2g0n.productlist.0.0.2adc5e4bPkrTcj&browser_id=e9b3b5797bbf4dcbb171577f5f60f54a&aff_trace_key=5113338f2a844836a883932a8b819ae6-1695038931676-05965-_DmDMgsR&aff_platform=msite&m_page_id=voxhybccazfsnqd18b1e10bb9a1f54bfa461a15945&pdp_npi=4%40dis%21CHF%215.43%212.44%21%21%215.89%21%21%402101c5a716970164948397025ea1b8%2112000034281871759%21sea%21CH%21707567368%21&algo_pvid=3f05f509-3ea9-43c2-937d-54b21206f20c)  that trigger Power Delivery Voltage to Jack. With 100W
But you can't use it for Serial Data and Power at the same time.

BTT. Glad to hear you got it working once.
Then that means the 5ms delay in booting should not be an issue.
I'll try and fiddle around a bit more this evening.
I was referring to your code to find out what exactly needs to happen to boot, but sadly I'm quite inexperienced in that field.

That would actually be quite a nice sales argument "Comes out of the box with working pre installed software" if the devs allow selling hardware with pre installed software.

anabolyc, 2023-10-11:
> 
> That would actually be quite a nice sales argument "Comes out of the box with working pre installed software" if the devs allow selling hardware with pre installed software.

I don't think I should sell other people's software ;)

DerPicknicker, 2023-10-11:
I will try to build the latest code from @philippe44 ... if it works I can test as well. I hope that IDF5 is working fine. It took me hours to get this running LOL. 

@anabolyc I need more than 3. If esplouder works with the support of @haldi4803 I will definitely order some of them. I have some nice speakers laying around which I would like to use in a much modern way.

anabolyc, 2023-10-11:
> 
> @anabolyc I need more than 3. If esplouder works with the support of @haldi4803 I will definitely order some of them. I have some nice speakers laying around which I would like to use in a much modern way.

As soon as it works, let me know, I'll request more from production.

philippe44, 2023-10-11:
@DerPicknicker - you don't need to rebuild. It's built but it will not appear in the official updates. You need to download from the "actions" artifacts

haldi4803, 2023-10-11:
@philippe44 sorry to bother you. The 5ms Delay does NOT seem to be needed.

current config: 
set_GPIO `33=vcc`
dac_controlset `poweron:[{"reg":3, "val":2}, {"reg":3, "val":3}]`

And it works when using via Bluetooth <3

DerPicknicker, 2023-10-11:
@haldi4803 ... 

**WOW! Great for this great documentation!**

I have some Additional questions: 
- Does the Amp automatically switch to deepsleep? 
- 0,4W with with WiFi connected? Could you try to open the webui and measure the voltage again..?

Thank you!

haldi4803, 2023-10-11:
>     * Does the Amp automatically switch to deepsleep?
>     * 0,4W with with WiFi connected? Could you try to open the webui and measure the voltage again..?

Nope... Still need to configure that correctly. Pretty sure it works somehow :)
0.4W in Idle with WiFI connected! When you have the WebUI open it jumps up to 0.8W-1.4W

DerPicknicker, 2023-10-11:
> >    
> 0.4W in Idle with WiFI connected! When you have the WebUI open it jumps up to 0.8W-1.4W
> 
> 

If you found this out, this would be epic! 0,4W idle is nothing compared to Sonos and others.

haldi4803, 2023-10-11:
@anabolyc  there is this knocking sound of the speakers, happening every now and then.
https://www.youtube.com/watch?v=dQGN24mMhbs

any clue where that might be coming from? Audio isn't really my speciality.
Buffer issues?

DerPicknicker, 2023-10-11:
@haldi4803 ... I would check the powersupply and the connection to the speakers first.

haldi4803, 2023-10-11:
> @haldi4803 ... I would check the powersupply and the connection to the speakers first.

Switched to a different Power Source still happening.
I think it's happening left/right speaker separately so shouldnt be speaker cables?

haldi4803, 2023-10-11:
@anabolyc seems to be an issue with the Voltage!
Reduce it to 17V absolutly NO issues at all.
Increase it to 19V. Here and there some issues. go up to 23V really bad Sound quality.
so must be Power Supply related! Sorry :(

philippe44, 2023-10-11:
> @haldi4803 ...
> 
> **WOW! THANK YOU for this great documentation!**
> 
> I have some Additional questions:
> 
> * Does the Amp automatically switch to deepsleep?
> * 0,4W with with WiFi connected? Could you try to open the webui and measure the voltage again..?
> 
> Thank you!
>
Re power and speaker/headset, I've updated the README but basically it's when the dac or amplifier is powered on/off after the idle period (-C) and when you switch between headset and speakers using jack detection. It is not related to deep_sleep mode.

DerPicknicker, 2023-10-12:
@philippe44 .. 

So if i2c controlset is used the amp shutdowns via the command after some time, because no audio was released from the LMS right or did I misunderstood you?

anabolyc, 2023-10-12:
Any signs of overheating?

sle118, 2023-10-12:
@haldi4803 can you or someone else with a working device save the NVS and share it here? I will add it too the list of known devices under hardware/preset options in the web UI so newcomers find it easier to configure

haldi4803, 2023-10-12:
> @haldi4803 can you or someone else with a working device save the NVS and share it here? I will add it too the list of known devices under hardware/preset options in the web UI so newcomers find it easier to configure

IF I have a fully working configuration.
Yes.
But right now it's not yet working as I want it to :)
Doesn't automatically boot the AMP. Seems like I need to try with the delay.

haldi4803, 2023-10-12:
i hate json so much....
But it's working! Automatically Boots up AMP, goes to AMP Deepsleep instantly, and when you play a tune it goes to play mode. if you turn the ESP-Louder of in LMS it goes back to DeepSleep and uses about 0.27-0.33W

(standard version not the one with delay and GPIO pins in controlset)

dac_controlset: `{"init":[{"reg":3,"val":2},{"reg":3,"val":3}],"poweron":[{"reg":3,"val":3}],"poweroff":[{"reg":3,"val":0}]}`

i increased the Stream and Output Buffer from 500:1000 to 1000:2000 because i still have enough free RAM that way.
Or not?
[image]

here is the nvs config @sle118 
[nvs_config_ESPLouder_1697127125985json.txt](https://github.com/sle118/squeezelite-esp32/files/12884722/nvs_config_ESPLouder_1697127125985json.txt)

sle118, 2023-10-13:
I would suggest not tweaking buffers unless you're trying to fix an issue. Memory is precious and some operations could momentarily need more RAM for a small peak of time. @philippe44 might have a different opinion, as he's the king of RAM optimizations.

I'll try to incorporate the NVS to the presets soon if you are ok with that config

haldi4803, 2023-10-13:
I think it works really well like that.

DerPicknicker, 2023-10-13:
> @sle118 ...
> Is there a way to support the development? Like Paetreon, BuyMeACoffee something in this direction?

@sle118 ..?

sle118, 2023-10-13:
> > @sle118 ...
> > Is there a way to support the development? Like Paetreon, BuyMeACoffee something in this direction?
> 
> @sle118 ..?

By passing the word around and converting other people to the project I guess. Up until now, we were always driven by the challenge and fun of doing this and pushing the limits on this little device. So for me, showing signs of appreciations and knowing this project helps others is already a good thing.

sle118, 2023-10-13:
> > That would actually be quite a nice sales argument "Comes out of the box with working pre installed software" if the devs allow selling hardware with pre installed software.
> 
> I don't think I should sell other people's software ;)

Some companies have done it and our licence allows it

DerPicknicker, 2023-10-14:
@sle118 ... 

Did you add the ESPLouder Config in the general Build..?

philippe44, 2023-10-14:
I usually wait till configuration are fully validated my a few users before making it built-in. Re SPI, yes you can. I have a board with SPI used for W5500, a display and a GPIO expander.

philippe44, 2023-10-14:
> EDIT: Do you think you can combine Ethernet with ESPLouder..? The Pins are not the Problem (there are plenty of them available...), but do I need to use the SPI-Multiplexer pins or its just optional if you want to get more speed...? Some of the multiplexer pins are used by the amplifier.. So I want to use the W5500 Ethernet option.
>
Also, if you are referring to the usage of the special pins for SPI in the GPIO matrix mux that gives you up to 40MHz speed, yes using these is optional, you just won't get full speed as mentioned here https://docs.espressif.com/projects/esp-idf/en/v4.3.6/esp32/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux

DerPicknicker, 2023-10-15:
@philippe44 ..

I will buy one of the LouderESP. I will test the config. If it works fine would be great if you would add the config in the general build. I will try to buy one soon... I am doing some research which loudpeakers are great for the rooms with an small footprint.. My big ones are to big for each room 😅

haldi4803, 2023-10-15:
> I am doing some research which loudpeakers are great for the rooms with an small footprint.

And what are your findings?
Didn't get many decent answers when I [asked](https://www.reddit.com/r/BudgetAudiophile/comments/176jiu2/cheap_compact_speakers/)  ...

DerPicknicker, 2023-10-15:
> > I am doing some research which loudpeakers are great for the rooms with an small footprint.
> 
> 
> 
> And what are your findings?
> 
> Didn't get many decent answers when I [asked](https://www.reddit.com/r/BudgetAudiophile/comments/176jiu2/cheap_compact_speakers/)  ...

Nothing special: 
- Quadral Aurum 3 (quite small but great bass) 
- Dali OnWall 
...
But I am still searching for an good single speaker for each room like an Amazon echo but with much better sound.

philippe44, 2023-10-15:
> @philippe44 ..
>
> I will buy one of the LouderESP. I will test the config. If it works fine would be great if you would add the config in the general build. I will try to buy one soon... I am doing [...]
There are 2 levels of integration, one is really just add a JSON dac_controlset and the other is direct C code driver-type. Once you are 100% sur of the config and as said, another user confirmed it, I'll see what's the best option.

[7 more replies]