sle118/squeezelite-esp32 #328: Crashes while running AirPlay

emlynmac, 2023-10-08:
To help us resolve your issue as quickly as possible, please follow these guidelines when submitting an issue. Providing all the necessary information will save both your time and ours.

### Describe the bug
Seeing crashes when running AirPlay on 16 bit build 1598

### Preliminary Information

1. FW: Squeezeamp-1598
2. **Plugin Version**: N/A

### Hardware Details

Please describe your hardware setup:
ESP32 Wrover B 16M with squeeze amp v3, no charger populated.

### NVS Settings

Follow these steps to share your NVS settings:

1. Open the web UI of your device.
2. Click on the "Credit" tab.
3. Enable the "Show NVS Editor" checkbox. This allows you to view or change the NVS configuration even when not in recovery mode.
4. Navigate to the "NVS Editor" tab.
5. Scroll to the bottom and click "Download Config".
6. Share the downloaded content here.

<details>
<pre><code>
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "Garage",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "Garage",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "<redacted>",
  "ap_ssid": "Garage",
  "autoexec": "1",
  "autoexec1": "squeezelite  -o i2s  -b  500:2000 -C  30 ",
  "bat_config": "",
  "bt_name": "Garage",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "20",
  "bypass_wm": "0",
  "cspot_config": "{\t\"deviceName\":\t\"Garage\",\t\"bitrate\":\t160}",
  "dac_config": "",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "Y",
  "enable_bt_sink": "Y",
  "enable_cspot": "N",
  "equalizer": "0,0,0,0,0,0,0,0,0,0",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "Garage",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "https://github.com/sle118/squeezelite-esp32/releases",
  "rotary_config": "",
  "set_GPIO": "0=ir",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "N",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "N"
}
</code></pre>
</details>

### Logs

To share logs:

1. Connect your player to a computer using a USB cable. Use a built-in Serial-USB adapter if your player has one, or an external USB adapter otherwise.
2. Go to the [web installer](https://sle118.github.io/squeezelite-esp32-installer/).
3. Click "Connect to Device".
4. Select the appropriate serial port.
5. Click "Logs And Console".
6. Download the logs and share them here. Please remove any sensitive information like Wi-Fi passwords or MAC addresses.
- **If the problem occurs soon after booting**: Share the full log until the issue occurs.
- **If the problem occurs later during playback**: Trim the logs to include information just before and after the problem occurs.

<code>

.[0;32mI (3382) console: Running command autoexec1 = squeezelite  -o i2s  -b  500:2000 -C  30 .[0m
.[0;33mW (3392) config: Value not found for key autoexec2.[0m
squeezelite-esp32> .[0;32mI (3392) squeezelite_cmd: Calling squeezelite.[0m
[00:00:01.922] sb_controls_init:249 initializing audio (buttons/rotary/ir) controls (raw:0)
[00:00:01.933] sb_displayer_init:327 no display or led visualizer for LMS
.[0;32mI (3422) DAC core: DAC uses I2C port:0, sda:27, scl:26.[0m
.[0;33mW (3432) DAC core: I2C read failed.[0m
.[0;32mI (3432) TAS575x/8x: Detected TAS @0x4c.[0m
[00:00:01.950] output_init_i2s:359 invalid MCK gpio -1
.[0;32mI (3442) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=12.[0m
.[0;32mI (3442) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0.[0m
[00:00:01.975] equalizer_set_samplerate:121 equalizer sample rate 44100
.[0;33mW (3472) BT_AV: BT Sink pin code is: [1234] .[0m
.[0;32mI (3472) BTDM_INIT: BT controller compile version [2c56073].[0m
.[0;32mI (3862) BT_AV: Bluetooth Init complete.[0m
I (5252) wifi:new:<3,0>, old:<1,0>, ap:<255,255>, sta:<3,0>, prof:1
I (5802) wifi:state: init -> auth (b0)
I (6802) wifi:state: auth -> init (200)
I (6802) wifi:new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
.[0;33mW (6812) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: 82:8a:20:8d:ec:65, reason code: 2 (WIFI_REASON_AUTH_EXPIRE).[0m
.[0;32mI (6822) network_handlers: Wifi was disconnected from previous access point. Waiting to connect..[0m
.[0;32mI (18332) network_handlers: Timer: Wifi Polling timeout .[0m
.[0;32mI (18332) network_wifi: Wifi Connecting to AP-IoT....[0m
I (20022) wifi:new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (20022) wifi:state: init -> auth (b0)
I (21022) wifi:state: auth -> init (200)
I (21022) wifi:new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
.[0;33mW (21022) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: 82:8a:20:8d:ec:65, reason code: 2 (WIFI_REASON_AUTH_EXPIRE).[0m
.[0;32mI (21042) network_handlers: Wifi was disconnected from previous access point. Waiting to connect..[0m
  .[0;31mE (593901) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:.[0m
.[0;31mE (593901) task_wdt:  - IDLE (CPU 0).[0m
.[0;31mE (593901) task_wdt: Tasks currently running:.[0m
.[0;31mE (593901) task_wdt: CPU 0: wifi.[0m
.[0;31mE (593901) task_wdt: CPU 1: IDLE.[0m
.[0;31mE (593901) task_wdt: Print CPU 0 (current core) backtrace.[0m

....

Backtrace:0x401E5BBE:0x3FFBED40 0x40084891:0x3FFBED60 0x4027F21B:0x3FFCAA00 0x4027DEA7:0x3FFCAA20 0x40209BD2:0x3FFCAA40 0x4020A2D1:0x3FFCAA60 0x40209008:0x3FFCAA80 0x4027B5E1:0x3FFCAAA0 
</code>

### Issue Description

1. **Observed Behavior**: Crash during AirPlay playback
2. **Expected Behavior**: No crash 
3. **Steps to Reproduce**: Connect over airplay from Mac / iPhone and wait for a few minutes

Sometimes also seeing:

<code>
.[0;31mE (128822) messaging: cfg-audio-tmpl
Rebooting in 60 sec

emlynmac, 2023-10-08:
Just updated to the latest build and experienced different error / reset:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6480
load:0x40078000,len:15868
load:0x40080400,len:4384
entry 0x40080684
I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (27) boot: compile time 22:28:30
I (27) boot: chip revision: 1
I (30) qio_mode: Enabling default flash chip QIO
I (35) boot.esp32: SPI Speed      : 80MHz
I (40) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (80) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 recovery         factory app      00 00 00010000 00140000
I (95) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (110) boot: End of partition table
I (115) boot_comm: chip revision: 1, min. application chip revision: 0
I (122) esp_image: segment 0: paddr=00150020 vaddr=3f400020 size=af030h (716848) map
I (343) esp_image: segment 1: paddr=001ff058 vaddr=3ffbdb60 size=00fc0h (  4032) load
I (344) esp_image: segment 2: paddr=00200020 vaddr=400d0020 size=1b30d4h (1781972) map
I (876) esp_image: segment 3: paddr=003b30fc vaddr=3ffbeb20 size=04a7ch ( 19068) load
I (883) esp_image: segment 4: paddr=003b7b80 vaddr=40080000 size=1e2dch (123612) load
I (926) esp_image: segment 5: paddr=003d5e64 vaddr=400c0000 size=00068h (   104) load
I (941) boot: Loaded app from partition at offset 0x150000
I (941) boot: Disabling RNG early entropy source...
I (953) psram: This chip is ESP32-D0WD
I (953) spiram: Found 64MBit SPI RAM device
I (953) spiram: SPI RAM mode: flash 80m sram 80m
I (956) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (963) cpu_start: Pro cpu up.
I (967) cpu_start: Starting app cpu, entry point is 0x40081408
I (0) cpu_start: App cpu up.
I (1470) spiram: SPI SRAM memory test OK
I (1483) cpu_start: Pro cpu start user code
I (1483) cpu_start: cpu freq: 240000000
I (1483) cpu_start: Application information:
I (1486) cpu_start: Project name:     SqueezeAMP
I (1491) cpu_start: App version:      SqueezeAmp-1598
I (1497) cpu_start: Compile time:     Oct  4 2023 23:00:33
I (1503) cpu_start: ELF file SHA256:  0000000000000000...
I (1509) cpu_start: ESP-IDF:          v4.3.5-dirty
I (1515) cpu_start: Min chip rev:     v1.0
I (1519) cpu_start: Max chip rev:     v3.99 
I (1524) cpu_start: Chip rev:         v1.0
I (1529) heap_init: Initializing. RAM available for dynamic allocation:
I (1537) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1543) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1549) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1555) heap_init: At 3FFC6F28 len 000190D8 (100 KiB): DRAM
I (1561) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1568) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1574) heap_init: At 4009E2DC len 00001D24 (7 KiB): IRAM
I (1581) spiram: Adding pool of 4018K of external SPI memory to heap allocator
I (1589) spi_flash: detected chip: generic
I (1593) spi_flash: flash io: qio
W (1597) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
[... 980 lines cut ...]
[00:13:36.440] buffer_put_packet:504 [0x3fac91e4]: fill [level:1 rec:178] [W:4596 R:4595]
[00:13:44.410] buffer_push_packet:570 [0x3fac91e4]: drain [level:0 head:2232 ms] [W:5593 R:5593] [req:200 sil:2 dis:0]
[00:13:44.439] buffer_put_packet:504 [0x3fac91e4]: fill [level:1 rec:186] [W:5598 R:5597]
[00:13:52.392] buffer_push_packet:570 [0x3fac91e4]: drain [level:0 head:2242 ms] [W:6594 R:6594] [req:206 sil:2 dis:0]
[00:13:52.436] buffer_put_packet:504 [0x3fac91e4]: fill [level:1 rec:192] [W:6600 R:6599]
[00:14:00.381] buffer_push_packet:570 [0x3fac91e4]: drain [level:0 head:2243 ms] [W:7595 R:7595] [req:215 sil:2 dis:0]
[00:14:00.428] buffer_put_packet:504 [0x3fac91e4]: fill [level:1 rec:200] [W:7601 R:7600]
[00:14:03.021] rtp_thread_func:768 [0x3fac91e4]: terminating
[00:14:03.025] cleanup_rtsp:684 [0x3fac6470]: RTP thread aborted
[00:14:03.027] cleanup_rtsp:700 [0x3fac6470]: Remote search thread aborted
W (213822) messaging: cfg-audio-tmpl
squeezelite exited with error code 0

W (213822) messaging: cfg-audio-tmpl
squeezelite exited with error code 0

E (213832) messaging: cfg-audio-tmpl
Rebooting in 60 sec

E (213832) messaging: cfg-audio-tmpl
Rebooting in 60 sec

```

Then gets into a reboot loop

sle118, 2023-10-09:
Dave Matthews. Good choice :)

Now it looks like you have some WiFi issues on your network. The messages below are indicating some association issues

emlynmac, 2023-10-09:
@sle118 thanks, I'm also going to be looking at getting my local environment up and running for building / debugging.
Any tips for doing this would be much appreciated, I've got docker installed so far but I don't think I can pass my serial port through to the docker build environment in MacOS - or can I?
Maybe time to set it up properly on macOS.
Are the build instructions up-to-date?

sle118, 2023-10-09:
First thing to do also is to copy the content of build-script/yourhardwareofchoice to sdkconfig.

philippe44, 2023-10-09:
Do you have a LMS server running on your network?

emlynmac, 2023-10-09:
I do not, but i can set one up if that would help debug

emlynmac, 2023-10-09:
> First thing to do also is to copy the content of build-script/yourhardwareofchoice to sdkconfig.

OK, done, copied the SqueezeAmp config across, removed CSPOT and updated from 4mb to 16mb flash.
`idf.py build` now building fine.

Now I need to go set up a linux VM so I can actually flash...

philippe44, 2023-10-09:
So if you do not have a LMS server, it will reboot every 5-10 mins. It's in bold here https://github.com/sle118/squeezelite-esp32#additional-configuration-notes-from-the-web-ui

emlynmac, 2023-10-09:
> So if you do not have a LMS server, it will reboot every 5-10 mins. It's in bold here https://github.com/sle118/squeezelite-esp32#additional-configuration-notes-from-the-web-ui. The RTP thread aborts because the system goes reboot.

Was this a recent change? It had been working fine for a few months until I updated from 1306 to latest.

So adding `-s -disable`  to the command line will fix this up?

philippe44, 2023-10-09:
Yes that was a change a few versions ago as the WiFi sometimes get stall and would never recover connection

emlynmac, 2023-10-09:
OK, so that explains the reboot loop.

Does it also explain the AirPlay crash, or was that fixed in updating from the older version to the more recent?
I can try it out for a bit and see I guess.

philippe44, 2023-10-09:
The crash IMHO happened because of the reboot

emlynmac, 2023-10-09:
> The crash IMHO happened because of the reboot

I can buy that :)

And if the way to have it auto-reconnect is to have LMS on my network, then I will happily have that.

closed 2023-10-09 by emlynmac

philippe44, 2023-10-09:
Let's see how it goes - don't hesitate to re-open this if it happens again.