sle118/squeezelite-esp32 #331: ! v2.1616 -> BUG (Web interface)

Renber78, 2023-10-12:
Quick message to report that today's version 'v2.1616-16' is bugged in the web interface.
- Status information is missing
- Hardware options are missing (SPI/Display/...)

Return to 'v2.1607-16' -> ok

sle118, 2023-10-12:
Some hardware options have drifted from their original implementation, as features were added. This means GPIO in the status tab would be missing entries and that some hardware options would no longer generate the correct NVS format. I am thinking about a possible way to resolve this, but until then hardware options will have to be set via the NVS editor following the readme intructions.

Renber78, 2023-10-12:
ok, if it's voluntary that suits me.
I was already configuring via NVS.
My message was in case you hadn't seen it (bug)

sle118, 2023-10-12:
I appreciate the input, sometimes the workflow build isn't perfectly aligned with the local build, so having feedback is always appreciated!

closed 2023-10-12 by sle118