sle118/squeezelite-esp32 #332: Battery voltage indication broken after updating to firmware version 1616

CornelisJ, 2023-10-12:
### Describe the bug
After updating the firmware of my ESP Muse Luxe to version 2.1616-16, the battery voltage indication is broken.
The status LED on the Muse is always red, and the voltage reported in the LMS Player Information is 0V.
After reverting to firmware version 1607, the status LED becomes green again, and the voltage reported by LMS is 4.15V. 

### Preliminary Information
Firmware Version: 2.1616, 16-bit
Plugin Version: v0.501

### Hardware Details
ESP Muse Luxe

sle118, 2023-10-13:
When posting a big, please follow the provided template. We're missing logs and NVS at a gate minimum

RASPIAUDIO, 2023-10-13:
Hello dear Squeezelite team! 
let me try to document this one : 
-Logs attached[nvs_config_squeezelite-deef4_1697203858060json.txt](https://github.com/sle118/squeezelite-esp32/files/12894036/nvs_config_squeezelite-deef4_1697203858060json.txt)
-NVS [esp-web-tools-logs.txt](https://github.com/sle118/squeezelite-esp32/files/12894007/esp-web-tools-logs.txt)

Led WS2812 stays red whatever the battery level, probably something related to the ACD on pin GPIO33?

Battery indicator is working fine with earlier version firmware version 1607.

Thanks again for this great project
Kind regards,
Olivier

sle118, 2023-10-13:
I have identified an issue that was mistakenly introduced in my last commit. I'll release a fix later today

sle118, 2023-10-13:
Please install the most recent release to address this

closed 2023-10-13 by sle118

CornelisJ, 2023-10-13:
I can confirm that battery voltage indication is OK again with firmware release 2.1619.
Thank you very much for the quick repair!