sle118/squeezelite-esp32 #339: BiWiring support for 2ch Amplifiers

DerPicknicker, 2023-10-16:
Hi guys, 

As you know I am a newbie in LMS + SQUEEZLITE... I was thinking about the possibility of using an 2ch amp for driving an BiWiring speaker. It would be a nice feature if you can have some simple (I know that audio on an esp32 is a tough thing ...) Filter which maps the 2ch in High/Low frequencies. 

This would allow to use for example an esplouder (2x23W) for better speakers which needs more power. In this case you can have a way powerful speaker. Because the woofer as real 23W. 

Hope you get my point. Anyways thank you for reading and sharing your thoughts!

philippe44, 2023-10-17:
I'm not sure SqueezeESP32 as much to add here. You can already do that with some PA I think. Even the TAS I'm using (or some sub-version) might have biwiring. I'm not sure

DerPicknicker, 2023-10-17:
Hi @philippe44 ... 

Or let's say it in different words: Is there a way inside SqueezeLite / LMS to send the same data to both Chanel's. So 
I can make use of an more powerful amp. 

Did you get my point? For example the EspLouder has 2x23W but for an subwoofer one channel is to less power..

philippe44, 2023-10-17:
yes, there is a mono mode, sorry this is what I was hinting to

DerPicknicker, 2023-10-17:
> yes, there is a mono mode, sorry this is what I was hinting to

Okay issue can be closed. A little offtopic: could you send me a binary for esp32s3..? I can't get the build system running. I tried via IDF. Not successfully. I will try via docker but let's see. If you already have an build on your computer would be great if you can attach it here ;-)

closed 2023-10-17 by DerPicknicker

philippe44, 2023-10-17:
problem is that there are different builds depending on the board end even their dev board has different GPIO for the RGB led (written on the back)

DerPicknicker, 2023-10-19:
@philippe44 .. With the docker image I can build the s3 branch?

philippe44, 2023-10-19:
I don't know, I don't use it but you should be able to build with an idf 4.4 and our repository. I installed that on a computer recently and did not do anything special AFAIR.