sle118/squeezelite-esp32 #343: Guru Meditation Error - can't reconnect

ElFishi, 2023-10-23:
### Describe the bug
Every few hours or so my ESP crashes with a Guru Meditation Error. After the ensuing reboot the ESP can't connect to the server anymore. Only another reboot will clear the issue

### Preliminary Information

1. **Firmware Version**: most recently fw: I2S-4MFlash-1621, completely fresh flash of the entire fileset. But happens also with earlier versions
2. **Plugin Version**: v0.501

### Hardware Details

- **ESP32 Module**: ESP32 WROVER-IB
- **Board Type**: Home brew
- **DAC Chip**: ES9038Q2M, I2S & I2C interfaces
- **Additional Hardware**: I2C OLED

### NVS Settings

<details><pre><code>{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "Squeezelite",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "ESP32-AirPlay-Scarlet",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "squeezelite-Scarlet",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W  -n \"Scarlet\"",
  "bat_config": "",
  "bt_name": "ESP32-BT-Scarlet",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"volume\":14093,\"deviceName\":\"Scarlet-cspot\",\"bitrate\":320,\"zeroConf\":1}",
  "dac_config": "model=i2s,bck=25,ws=33,do=32,scl=13,sda=14,i2c=72",
  "dac_controlset": "{\"init\":[ {\"reg\":1,\"val\":12} ],\"poweron\":[ {\"reg\":1,\"val\":12} ]}",
  "dhcp_tmout": "8",
  "disable_ps": "n",
  "display_config": "I2C,width=128,height=64,address=60,driver=SSD1306",
  "enable_airplay": "N",
  "enable_bt_sink": "N",
  "enable_cspot": "1",
  "equalizer": "0,0,0,0,0,0,0,0,0,0",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "Scarlet",
  "i2c_config": "scl=4,sda=2,speed=400000",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "https://github.com/sle118/squeezelite-esp32/releases",
  "rotary_config": "",
  "set_GPIO": "19=green:1,18=red:1,26=power,15=ir",
[... 620 lines cut ...]
Set-Cookie: Squeezebox-expanded-FAVORITES=1; path=/

Set-Cookie: Squeezebox-expanded-PLUGINS=0; path=/

Set-Cookie: Squeezebox-expanded-PLUGIN_MY_APPS_MODULE_NAME=1; path=/

Set-Cookie: Squeezebox-expanded-RADIO=0; path=/

[01:43:49.130] read_vorbis_header:211 id acquired
[01:43:49.139] read_vorbis_header:247 comment skipped succesfully
[01:43:49.181] read_vorbis_header:229 codec up and running (rate: 44100, channels:2)
[01:43:49.183] vorbis_decode:294 setting track_start
[01:43:49.192] vorbis_decode:384 no frame decoded
[01:43:55.005] _output_frames:150 track start sample rate: 44100 replay_gain: 65536
[01:44:12.965] grfb_handler:755 brightness 2

---snippetysnap---
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: ESP crashes after longer periods of normal operation. Does not reconnect upon first reboot.
2. **Expected Behavior**: play w/o crash
3. **Steps to Reproduce**: Run, wait ....

sle118, 2023-10-24:
Thank you for following the bug submission template. You have the gold Star ⭐ here!

It seems like the crash happens when decoding Vorbis?

ElFishi, 2023-10-24:
Thank you, but you guys are doing such a magnificent job here. The least I can do is follow your suggestions. 

It happens when streaming via spotty which is 99% of what the ESP does. Is that Vorbis?

sle118, 2023-10-24:
Yes it is. Are you able to try streaming a different source and see if this happens? For example if you have a local collection compressed with FLAC or something like that.

Vorbis is the area that @philippe44 knows so perhaps he would have a clue.

ElFishi, 2023-10-24:
I can stream local FLAC. Will try.

ElFishi, 2023-10-25:
Seems like two problems atm (current fw is I2S-4MFlash-1627),

One is the Guru Meditation error. So far I have not been able to trigger it by playing local FLAC. 

The other is that network connection currently is hit and miss. 
Log of unsuccessful reboot
<details><pre><code>
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6948
load:0x40078000,len:13772
load:0x40080400,len:4532
entry 0x400806a0
(61) boot: Chip Revision: 1
(63) boot_comm: chip revision: 1, min. bootloader chip revision: 0
(39) boot: ESP-IDF v4.0.2-206-g5630b17e1 2nd stage bootloader
(39) boot: compile time 05:46:08
(39) boot: Enabling RNG early entropy source...
(45) qio_mode: Enabling default flash chip QIO
(50) boot: SPI Speed      : 80MHz
(54) boot: SPI Mode       : QIO
(58) boot: SPI Flash Size : 4MB
(62) boot: Partition Table:
(66) boot: ## Label            Usage          Type ST Offset   Length
(73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
(80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
(88) boot:  2 phy_init         RF data          01 01 0000f000 00001000
(95) boot:  3 recovery         factory app      00 00 00010000 00140000
(103) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
(110) boot:  5 settings         WiFi data        01 02 003f0000 00010000
(118) boot: End of partition table
(122) boot_comm: chip revision: 1, min. application chip revision: 0
(129) esp_image: segment 0: paddr=0x00150020 vaddr=0x3f400020 size=0xb4660 (738912) map
(376) esp_image: segment 1: paddr=0x00204688 vaddr=0x3ffbdb60 size=0x05a3c ( 23100) load
(384) esp_image: segment 2: paddr=0x0020a0cc vaddr=0x40080000 size=0x05f4c ( 24396) load
(394) esp_image: segment 3: paddr=0x00210020 vaddr=0x400d0020 size=0x1b3408 (1782792) map
(967) esp_image: segment 4: paddr=0x003c3430 vaddr=0x40085f4c size=0x18390 ( 99216) load
(1004) esp_image: segment 5: paddr=0x003db7c8 vaddr=0x400c0000 size=0x00068 (   104) load
(1023) boot: Loaded app from partition at offset 0x150000
(1023) boot: Disabling RNG early entropy source...
(1023) psram: This chip is ESP32-D0WD
(1028) spiram: Found 64MBit SPI RAM device
(1033) spiram: SPI RAM mode: flash 80m sram 80m
(1038) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
(1045) cpu_start: Pro cpu up.
(1049) cpu_start: Starting app cpu, entry point is 0x40081408
(0) cpu_start: App cpu up.
(1552) spiram: SPI SRAM memory test OK
(1565) cpu_start: Pro cpu start user code
(1565) cpu_start: cpu freq: 240000000
(1565) cpu_start: Application information:
(1568) cpu_start: Project name:     Squeezelite-ESP32
(1574) cpu_start: App version:      I2S-4MFlash-1627
(1580) cpu_start: Compile time:     Oct 25 2023 01:53:05
(1586) cpu_start: ELF file SHA256:  0000000000000000...
(1592) cpu_start: ESP-IDF:          v4.3.5-dirty
(1597) cpu_start: Min chip rev:     v1.0
(1602) cpu_start: Max chip rev:     v3.99 
(1607) cpu_start: Chip rev:         v1.0
(1612) heap_init: Initializing. RAM available for dynamic allocation:
(1619) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
(1625) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
(1632) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
(1638) heap_init: At 3FFC6F28 len 000190D8 (100 KiB): DRAM
(1644) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[... 398 lines cut ...]
[00:00:07.469] output_thread_i2s:585 Restarting I2S.
[00:00:07.522] equalizer_process:222 equalizer deactivated
[00:00:07.530] write_cb:141 setting track_start
[00:00:07.549] _output_frames:65 start buffer frames: 4096
[00:00:07.551] _output_frames:150 track start sample rate: 44100 replay_gain: 0
[00:00:07.552] output_thread_i2s:518 Output state is 2
[00:00:37.223] grfb_handler:755 brightness 2
(182871) httpd_handlers: serving / to peer 192.168.0.142 port 30948
(183431) httpd_handlers: serving /commands.json to peer 192.168.0.142 port 31204
(183591) httpd_handlers: serving /messages.json to peer 192.168.0.142 port 32228
(185031) network_wifi: Initiating wifi network scan
(187901) httpd_handlers: serving /messages.json to peer 192.168.0.142 port 31460(188031) httpd_txrx: httpd_sock_err: error in send : 104(188041) httpd_txrx: httpd_sock_err: error in recv : 104
[00:03:29.958] stream_thread:337 end of stream (12549788 bytes)
[00:03:43.337] decode_thread:100 decode complete
[00:03:43.481] codec_open:283 codec open: 'f'
[00:03:43.484] stream_sock:508 connecting to 192.168.0.105:9000
[00:03:43.489] stream_sock:582 header: GET /stream.mp3?player=30:ae:a4:c5:yy:yy HTTP/1.0

[00:03:43.624] stream_thread:264 headers: len: 494
HTTP/1.1 200 OK
Server: Logitech Media Server (8.3.1 - 1676361197)
(...)
</code></pre></details>

Currently I have several unsuccessful reboots for one successful.

ElFishi, 2023-10-25:
Went back to streaming spotty. Recorded another crash and connection failure.
<details><pre><code>
(...)
[02:55:33.727] stream_thread:337 end of stream (11362844 bytes)
[02:55:42.670] vorbis_decode:380 end of decode
[02:55:42.672] decode_thread:100 decode complete
[02:55:42.910] codec_open:283 codec open: 'o'
[02:55:42.911] vorbis_open:399 OPENING CODEC
[02:55:42.915] stream_sock:508 connecting to 192.168.0.105:9000
[02:55:42.920] stream_sock:582 header: GET /stream.mp3?player=30:ae:a4:c5:yy:yy HTTP/1.0

[02:55:43.045] stream_thread:264 headers: len: 491
HTTP/1.1 200 OK

Server: Logitech Media Server (8.3.1 - 1676361197)

Connection: close

Content-Type: audio/ogg

Set-Cookie: Squeezebox-albumView=1; path=/

Set-Cookie: Squeezebox-expandPlayerControl=true; path=/

Set-Cookie: Squeezebox-expanded-MY_MUSIC=0; path=/

Set-Cookie: Squeezebox-expanded-FAVORITES=1; path=/

Set-Cookie: Squeezebox-expanded-PLUGINS=0; path=/

Set-Cookie: Squeezebox-expanded-PLUGIN_MY_APPS_MODULE_NAME=1; path=/

Set-Cookie: Squeezebox-expanded-RADIO=0; path=/

[02:55:43.435] read_vorbis_header:211 id acquired
[02:55:43.441] read_vorbis_header:247 comment skipped succesfully
[02:55:43.485] read_vorbis_header:229 codec up and running (rate: 44100, channels:2)
[02:55:43.487] vorbis_decode:294 setting track_start
[02:55:43.490] vorbis_decode:384 no frame decoded
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400941ef  PS      : 0x00060830  A0      : 0x801968ce  A1      : 0x3ffe9630  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00001000  A5      : 0x00000000  
A6      : 0x0000000d  A7      : 0x00000100  A8      : 0x3fb945dc  A9      : 0x3fb39528  
A10     : 0x3fb39698  A11     : 0x3fb94650  A12     : 0x3fb89b14  A13     : 0x0000001d  
A14     : 0x000000d0  A15     : 0x00008000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400940d5  LEND    : 0x400940f7  LCOUNT  : 0x00000000  

Backtrace:0x400941ec:0x3ffe9630 0x401968cb:0x3ffe9640 0x4018c98e:0x3ffe9720 0x401019c1:0x3ffe9750 0x400f7354:0x3ffe9790 0x400d58c8:0x3ffe97c0

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6948
load:0x40078000,len:13772
load:0x40080400,len:4532
entry 0x400806a0
I (62) boot: Chip Revision: 1
I (64) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v4.0.2-206-g5630b17e1 2nd stage bootloader
I (39) boot: compile time 05:46:08
I (40) boot: Enabling RNG early entropy source...
[... 489 lines cut ...]
Set-Cookie: Squeezebox-expanded-MY_MUSIC=0; path=/

Set-Cookie: Squeezebox-expanded-FAVORITES=1; path=/

Set-Cookie: Squeezebox-expanded-PLUGINS=0; path=/

Set-Cookie: Squeezebox-expanded-PLUGIN_MY_APPS_MODULE_NAME=1; path=/

Set-Cookie: Squeezebox-expanded-RADIO=0; path=/

[02:59:17.220] output_thread_i2s:518 Output state is 1
[02:59:17.222] output_thread_i2s:585 Restarting I2S.
[02:59:17.225] equalizer_process:222 equalizer deactivated
[02:59:17.317] read_vorbis_header:211 id acquired
[02:59:17.329] read_vorbis_header:247 comment skipped succesfully
[02:59:17.370] read_vorbis_header:229 codec up and running (rate: 44100, channels:2)
[02:59:17.375] vorbis_decode:294 setting track_start
[02:59:17.402] vorbis_decode:384 no frame decoded
[02:59:17.855] _output_frames:65 start buffer frames: 26176
[02:59:17.858] _output_frames:150 track start sample rate: 44100 replay_gain: 65536
[02:59:17.860] output_thread_i2s:518 Output state is 2
(...)
</code></pre></details>

Will go back to streaming flac.

sle118, 2023-10-26:
What bothers me is the auth failures.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#auth-phase

ElFishi, 2023-10-26:
[AVM FRITZ!Box 7530](https://en.avm.de/products/fritzbox/fritzbox-7530/) 
latest fw  FRITZ!OS: 7.57

philippe44, 2023-10-26:
Re Vorbis there is something in the code that should never have made its way there, but I let it slip. I don't think it does explain your issue, but it is worth trying.

ElFishi, 2023-10-26:
v2.1630?

philippe44, 2023-10-26:
yes

ElFishi, 2023-10-26:
running...

ElFishi, 2023-10-26:
Got another crash streaming spotty. This time triggered by skipping to the next track in the playlist in Squeezelite-X.
<details><pre><code>
[01:44:21.699] stream_thread:337 end of stream (5711881 bytes)
[01:44:37.603] vorbis_decode:380 end of decode
[01:44:37.604] decode_thread:100 decode complete
[01:44:37.765] codec_open:283 codec open: 'o'
[01:44:37.770] stream_sock:508 connecting to 192.168.0.105:9000
[01:44:37.774] stream_sock:582 header: GET /stream.mp3?player=30:ae:a4:c5:43:bc HTTP/1.0

[01:44:37.958] stream_thread:264 headers: len: 491
HTTP/1.1 200 OK

Server: Logitech Media Server (8.3.1 - 1676361197)

Connection: close

Content-Type: audio/ogg

Set-Cookie: Squeezebox-albumView=1; path=/

Set-Cookie: Squeezebox-expandPlayerControl=true; path=/

Set-Cookie: Squeezebox-expanded-MY_MUSIC=0; path=/

Set-Cookie: Squeezebox-expanded-FAVORITES=1; path=/

Set-Cookie: Squeezebox-expanded-PLUGINS=0; path=/

Set-Cookie: Squeezebox-expanded-PLUGIN_MY_APPS_MODULE_NAME=1; path=/

Set-Cookie: Squeezebox-expanded-RADIO=0; path=/

[01:44:38.257] read_vorbis_header:211 id acquired
[01:44:38.269] read_vorbis_header:247 comment skipped succesfully
[01:44:38.313] read_vorbis_header:229 codec up and running (rate: 44100, channels:2)
[01:44:38.320] vorbis_decode:294 setting track_start
[01:44:38.333] vorbis_decode:384 no frame decoded
[01:44:49.042] _output_frames:150 track start sample rate: 44100 replay_gain: 65536
[01:44:50.106] grfs_handler:807 scroller current size 1280 (w:262)
[01:44:50.121] grfs_handler:807 scroller current size 1560 (w:262)
[01:46:16.365] decode_flush:251 decode flush
[01:46:16.370] output_flush:446 flush output buffer
[01:46:16.407] output_thread_i2s:518 Output state is 0
[01:46:16.413] codec_open:283 codec open: 'o'
[01:46:16.415] stream_sock:508 connecting to 192.168.0.105:9000
[01:46:16.424] stream_sock:582 header: GET /stream.mp3?player=30:ae:a4:c5:43:bc HTTP/1.0

[01:46:16.580] stream_thread:264 headers: len: 491
HTTP/1.1 200 OK

Server: Logitech Media Server (8.3.1 - 1676361197)

Connection: close

Content-Type: audio/ogg

Set-Cookie: Squeezebox-albumView=1; path=/

Set-Cookie: Squeezebox-expandPlayerControl=true; path=/

Set-Cookie: Squeezebox-expanded-MY_MUSIC=0; path=/

Set-Cookie: Squeezebox-expanded-FAVORITES=1; path=/

Set-Cookie: Squeezebox-expanded-PLUGINS=0; path=/

Set-Cookie: Squeezebox-expanded-PLUGIN_MY_APPS_MODULE_NAME=1; path=/

Set-Cookie: Squeezebox-expanded-RADIO=0; path=/

[01:46:16.783] output_thread_i2s:518 Output state is 1
[01:46:16.854] read_vorbis_header:211 id acquired
[01:46:16.864] read_vorbis_header:247 comment skipped succesfully
[01:46:16.910] read_vorbis_header:229 codec up and running (rate: 44100, channels:2)
[01:46:16.916] vorbis_decode:294 setting track_start
[01:46:16.922] vorbis_decode:384 no frame decoded
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400941ef  PS      : 0x00060030  A0      : 0x8019690a  A1      : 0x3ffe95b0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00001000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000100  A8      : 0x3fbb6b0c  A9      : 0x3fbc2110  
A10     : 0x3fbc2280  A11     : 0x3fbb6b80  A12     : 0x3fbd7b1c  A13     : 0x0000001d  
A14     : 0x000000d0  A15     : 0x00008000  SAR     : 0x00000002  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x40094054  LEND    : 0x40094070  LCOUNT  : 0x00000000  

Backtrace:0x400941ec:0x3ffe95b0 0x40196907:0x3ffe95c0 0x4018c9ca:0x3ffe96a0 0x401019b9:0x3ffe96d0 0x400f734c:0x3ffe9710 0x400d58c0:0x3ffe9740

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57
</code></pre></details>

Followed by a `discover_server:836 error sending discovery`, just as before. 
In the end there is a reboot that did connect alright.
``` 
(...)
[01:47:45.230] discover_server:832 sending discovery 1
[01:47:50.231] decode_close:233 close decode
[01:47:50.330] deregister_external:500 Stopping CSpot sink
[01:47:50.331] stream_close:441 close stream
[01:47:50.340] output_close_embedded:112 close output
W (128662) messaging: cfg-audio-tmpl
squeezelite exited with error code 0

W (128662) messaging: cfg-audio-tmpl
squeezelite exited with error code 0

E (128672) messaging: cfg-audio-tmpl
Rebooting in 60 sec

E (128672) messaging: cfg-audio-tmpl
Rebooting in 60 sec

I (188682) wifi:flush txq
I (188682) wifi:stop sw txq
I (188682) wifi:lmac stop hw txq
ets Jun  8 2016 00:22:57

```

ElFishi, 2023-10-27:
Next observation:
Streaming Spotify via LMS/Spotty leads to a crash every 10k sec or so: 
<details><pre><code>
[01:46:16.783] output_thread_i2s:518 Output state is 1
[01:46:16.854] read_vorbis_header:211 id acquired
[01:46:16.864] read_vorbis_header:247 comment skipped succesfully
[01:46:16.910] read_vorbis_header:229 codec up and running (rate: 44100, channels:2)
[01:46:16.916] vorbis_decode:294 setting track_start
[01:46:16.922] vorbis_decode:384 no frame decoded
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400941ef  PS      : 0x00060030  A0      : 0x8019690a  A1      : 0x3ffe95b0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00001000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000100  A8      : 0x3fbb6b0c  A9      : 0x3fbc2110  
A10     : 0x3fbc2280  A11     : 0x3fbb6b80  A12     : 0x3fbd7b1c  A13     : 0x0000001d  
A14     : 0x000000d0  A15     : 0x00008000  SAR     : 0x00000002  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x40094054  LEND    : 0x40094070  LCOUNT  : 0x00000000  

Backtrace:0x400941ec:0x3ffe95b0 0x40196907:0x3ffe95c0 0x4018c9ca:0x3ffe96a0 0x401019b9:0x3ffe96d0 0x400f734c:0x3ffe9710 0x400d58c0:0x3ffe9740

ELF file SHA256: 0000000000000000

Rebooting...

(...)

[04:25:08.476] read_vorbis_header:211 id acquired
[04:25:08.486] read_vorbis_header:247 comment skipped succesfully
[04:25:08.533] read_vorbis_header:229 codec up and running (rate: 44100, channels:2)
[04:25:08.535] vorbis_decode:294 setting track_start
[04:25:08.538] vorbis_decode:384 no frame decoded
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400941ef  PS      : 0x00060230  A0      : 0x8019690a  A1      : 0x3ffe96a0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00001000  A5      : 0x00000000  
A6      : 0x0000000d  A7      : 0x00000100  A8      : 0x3fb71b78  A9      : 0x3fb7164c  
A10     : 0x3fb717bc  A11     : 0x3fb71bec  A12     : 0x3fbe7bc8  A13     : 0x0000001d  
A14     : 0x000000d0  A15     : 0x00008000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400941f6  LEND    : 0x40094201  LCOUNT  : 0x00000000  

Backtrace:0x400941ec:0x3ffe96a0 0x40196907:0x3ffe96b0 0x4018c9ca:0x3ffe9790 0x401019b9:0x3ffe97c0 0x400f734c:0x3ffe9800 0x400d58c0:0x3ffe9830

ELF file SHA256: 0000000000000000

Rebooting...

(...)

[07:01:14.501] read_vorbis_header:211 id acquired
[07:01:14.510] read_vorbis_header:247 comment skipped succesfully
[07:01:14.557] read_vorbis_header:229 codec up and running (rate: 44100, channels:2)
[07:01:14.565] vorbis_decode:294 setting track_start
[07:01:14.574] vorbis_decode:384 no frame decoded
[07:01:25.193] _output_frames:150 track start sample rate: 44100 replay_gain: 65536
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e5300  PS      : 0x00060630  A0      : 0x800e41ba  A1      : 0x3f837480  
A2      : 0x00000000  A3      : 0x3f814678  A4      : 0x00000042  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x000000ff  A8      : 0x0000001b  A9      : 0x3f814678  
A10     : 0x00000001  A11     : 0x3f81466c  A12     : 0x00000000  A13     : 0x3f814610  
A14     : 0x00000001  A15     : 0x0000000a  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000020  LBEG    : 0x400956b5  LEND    : 0x400956c5  LCOUNT  : 0xfffffffe  

Backtrace:0x400e52fd:0x3f837480 0x400e41b7:0x3f8374b0 0x400e443b:0x3f8374e0 0x400d677d:0x3f837500 0x40094586:0x3f837520 0x400930e5:0x3f837540 0x40093174:0x3f837560 0x4026023d:0x3f837590 0x4025cf6c:0x3f8376b0 0x4026029b:0x3f8379c0 0x402833e6:0x3f8379f0 0x4009c5d3:0x3f837a20 0x400e42b5:0x3f837a70

ELF file SHA256: 0000000000000000

[... 60 lines cut ...]
A6      : 0x00000000  A7      : 0x00000100  A8      : 0x3fb50610  A9      : 0x3fbf6f88  
A10     : 0x3fbf70f8  A11     : 0x3fb50684  A12     : 0x3fbb5304  A13     : 0x0000001d  
A14     : 0x000000d0  A15     : 0x00008000  SAR     : 0x00000002  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400941f6  LEND    : 0x40094201  LCOUNT  : 0x00000000  

Backtrace:0x400941ec:0x3ffe96a0 0x40196907:0x3ffe96b0 0x4018c9ca:0x3ffe9790 0x401019b9:0x3ffe97c0 0x400f734c:0x3ffe9800 0x400d58c0:0x3ffe9830

ELF file SHA256: 0000000000000000

Rebooting...

(...)
</code></pre></details>

The last lines before `Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.`
are 
<pre><code>
[01:46:16.922] vorbis_decode:384 no frame decoded
[04:25:08.538] vorbis_decode:384 no frame decoded
[07:01:25.193] _output_frames:150 track start sample rate: 44100 replay_gain: 65536
[09:59:02.501] vorbis_decode:384 no frame decoded
[12:45:02.528] vorbis_decode:384 no frame decoded
[15:35:39.096] vorbis_decode:384 no frame decoded
</code></pre>
times between each crash are 9532, 9377, 10657, 9960, and 10237 seconds, respectively.

ElFishi, 2023-10-27:
ran another 5 hrs this morning, playing local flac from my hard drive ith  fw: I2S-4MFlash-1630 and without problems.

philippe44, 2023-10-27:
So it's very likely a vorbis problem. But then it can be in my codec wrapper or in the decoder itself. I'll re-re-re-read the wrapper but for the codec, I can't do anything

philippe44, 2023-10-28:
So it ended being a memory that I should have seen before but become obvious when you listen to vorbis for many hours :smile:. It should be fixed in the latest build (not released yet, I'm waiting for your confirmation.

Thanks very much for finding that issue!

ElFishi, 2023-10-28:
Hang in there, I can only get back to it on Monday. 
How would I get that latest build to confirm?

philippe44, 2023-10-29:
Scroll to artifacts https://github.com/sle118/squeezelite-esp32/actions/runs/6675518757

sle118, 2023-10-29:
well, I did push a release to resolve the recovery boot loop so the most release should contain the fix

ElFishi, 2023-10-30:
Nailed it. 

I streamed spotify via spotty for 5hrs yesterday and for 8.5 hrs today without crash. 

Also I can't reproduce the connection error today. 
Was my router having a bad day? idk. 

So I am closing this issue now. 
Well done lads.

closed 2023-10-30 by ElFishi