sle118/squeezelite-esp32 #344: Serial input not working

ElFishi, 2023-10-25:
### Describe the bug
Commands entered via the serial interface are not received correctly.

### Preliminary Information

1. **Firmware Version**: fw: I2S-4MFlash-1627
2. **Plugin Version**: v0.501

### Hardware Details

Please describe your hardware setup:

- **ESP32 Module**: ESP32 WROVER-IB
- **Board Type**: Home brew
- **DAC Chip**: ES9018k2m, I2S, I2C
- **Additional Hardware**: OLED screen

### NVS Settings
<details>
<pre><code>
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "Squeezelite",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "ESP32-AirPlay-Scarlet",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "squeezelite-Scarlet",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W  -n \"Scarlet\"",
  "bat_config": "",
  "bt_name": "ESP32-BT-Scarlet",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"volume\":14093,\"deviceName\":\"Scarlet-cspot\",\"bitrate\":320,\"zeroConf\":1}",
  "dac_config": "model=i2s,bck=25,ws=33,do=32,scl=13,sda=14,i2c=72",
  "dac_controlset": "{\"init\":[ {\"reg\":1,\"val\":12} ],\"poweron\":[ {\"reg\":1,\"val\":12} ]}",
  "dhcp_tmout": "8",
  "disable_ps": "n",
  "display_config": "I2C,width=128,height=64,address=60,driver=SSD1306",
  "enable_airplay": "N",
  "enable_bt_sink": "N",
  "enable_cspot": "1",
  "equalizer": "0,0,0,0,0,0,0,0,0,0",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "Scarlet",
  "i2c_config": "scl=4,sda=2,speed=400000",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "https://github.com/sle118/squeezelite-esp32/releases",
  "rotary_config": "",
  "set_GPIO": "19=green:1,18=red:1,26=power,15=ir",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "n",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "D",
  "wifi_smode": "A"
}
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: Commands entered on a terminal emulator (tried hterm, arduino IDE and Putty) via the serial interface are not received correctly by the ESP. sometimes upon pressing Enter the first letter of the entered command is echoed. When hitting Enter several times one letter appears per key press. Finally, the command is executed. On other occasions no character is echoed only the CR+LF.
2. **Expected Behavior**: commands entered via the serial interface are echoed correctly, parsed and executed.
3. **Steps to Reproduce**: Hook up serial cable, enter text

ElFishi, 2023-10-25:
I might add that via telnet input is echoed (albeit twice) and executed:
<pre><code>
squeezelite-esp32> heap
heap

I (144582) cmd_system: Heap internal:90367 (min:90143) (largest block:81920)
external:595747 (min:590687) (largest block:589824)
dma :83287 (min:83079) (largest block:81920)
I (144612) messaging: heap
Heap internal:90483 (min:90111) (largest block:81920)
external:595747 (min:590687) (largest block:589824)
dma :83403 (min:83047) (largest block:81920)
squeezelite-esp32>

squeezelite-esp32>

</code></pre>

philippe44, 2023-10-26:
Does it do the same on telnet when *NO* serial is connected?

ElFishi, 2023-10-26:
The log in Telnet looks the same as posted above without a serial cable attached and both telnet_enable Y and telnet_enable D.

philippe44, 2023-10-26:
echoed as well?

ElFishi, 2023-10-26:
no serial connection, NVS telnet_enable D
<pre><code>
squeezelite-esp32> heap
heap

I (41601) cmd_system: Heap internal:91123 (min:90775) (largest block:81920)
external:811935 (min:791575) (largest block:802816)
dma :84043 (min:83711) (largest block:81920)
I (41621) messaging: heap
Heap internal:91123 (min:90739) (largest block:81920)
external:811935 (min:791575) (largest block:802816)
dma :84043 (min:83675) (largest block:81920)
squeezelite-esp32>

squeezelite-esp32>
</code></pre>

no serial connection, NVS telnet_enable Y
<pre><code>
squeezelite-esp32> heap
heap

I (70312) cmd_system: Heap internal:89879 (min:89671) (largest block:81920)
external:1195903 (min:1138107) (largest block:1179648)
dma :82799 (min:82607) (largest block:81920)
I (70312) messaging: heap
Heap internal:89879 (min:89671) (largest block:81920)
external:1195903 (min:1138107) (largest block:1179648)
dma :82799 (min:82607) (largest block:81920)
squeezelite-esp32>

squeezelite-esp32>

</code></pre>

philippe44, 2026-10-09:
Lots of fixes done on console and telnet

closed 2026-10-09 by philippe44