sle118/squeezelite-esp32 #345: Programming problems

Robyxxx, 2023-10-25:
Hi everyone,
I'm trying programming my ESP32 WROOVER (TTGO T7 Mini32 V1.5 ESP32-WROVER). Both using the online tool (https://sle118.github.io/squeezelite-esp32-installer/), and the expressif tool, I get the same error on the log.
How I can solve? Thanks!

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (52) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (53) boot: compile time 01:50:41
I (53) boot: chip revision: v3.1
I (57) qio_mode: Enabling default flash chip QIO
I (62) boot.esp32: SPI Speed      : 80MHz
I (67) boot.esp32: SPI Mode       : QIO
I (71) boot.esp32: SPI Flash Size : 4MB
I (76) boot: Enabling RNG early entropy source...
I (81) boot: Partition Table:
I (85) boot: ## Label            Usage          Type ST Offset   Length
I (92) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (100) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (107) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (115) boot:  3 recovery         factory app      00 00 00010000 00140000
I (122) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (130) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (137) boot: End of partition table
I (142) boot: Defaulting to factory image
I (146) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=59458h (365656) map
I (256) esp_image: segment 1: paddr=00069480 vaddr=3ffbdb60 size=04850h ( 18512) load
I (262) esp_image: segment 2: paddr=0006dcd8 vaddr=40080000 size=02340h (  9024) load
I (265) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=d27c0h (862144) map
I (507) esp_image: segment 4: paddr=001427e8 vaddr=40082340 size=12758h ( 75608) load
I (532) esp_image: segment 5: paddr=00154f48 vaddr=400c0000 size=00068h (   104) load
E (532) esp_image: Image length 1331168 doesn't fit in partition length 1310720
E (537) boot: Factory app partition is not bootable
E (543) esp_image: image at 0x150000 has invalid magic byte (nothing flashed here?)
E (551) boot: OTA app partition slot 0 is not bootable
E (557) boot: No bootable app partitions in the partition table
ets Jul 29 2019 12:21:46

sle118, 2023-10-26:
I will try the installer tomorrow and figure out

kwaek, 2023-10-26:
I get exactly the same error saying 'E (532) esp_image: Image length 1331168 doesn't fit in partition length 1310720' and the rest, when I'm trying to flash ESP16 squeeze lite image to LILYGO® TTGO T7 V1.5 Mini32 ESP32-WROVER-B. I tried using the standard WROVER development module... same issue happened. Please help.

philippe44, 2023-10-26:
The issue is somewhere in the webUI code that is 20kB larger (.rodata of js or related, not C code) than it is on my local build which means that it does not fit anymore in its partition. I don't know why, this is @sle118's realm but I suspect something needs to be re-run in term of building the UI code.

kwaek, 2023-10-26:
the following are the logs screen messages for ESP32-WROVER-B dev module:

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (52) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (52) boot: compile time 01:50:41
I (53) boot: chip revision: v1.0
I (57) qio_mode: Enabling default flash chip QIO
I (62) boot.esp32: SPI Speed      : 80MHz
I (67) boot.esp32: SPI Mode       : QIO
I (71) boot.esp32: SPI Flash Size : 4MB
I (76) boot: Enabling RNG early entropy source...
I (81) boot: Partition Table:
I (85) boot: ## Label            Usage          Type ST Offset   Length
I (92) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (99) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (107) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (115) boot:  3 recovery         factory app      00 00 00010000 00140000
I (122) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (130) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (137) boot: End of partition table
I (141) boot: Defaulting to factory image
I (146) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=59458h (365656) map
I (255) esp_image: segment 1: paddr=00069480 vaddr=3ffbdb60 size=04850h ( 18512) load
I (261) esp_image: segment 2: paddr=0006dcd8 vaddr=40080000 size=02340h (  9024) load
I (265) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=d27c0h (862144) map
I (507) esp_image: segment 4: paddr=001427e8 vaddr=40082340 size=12758h ( 75608) load
I (532) esp_image: segment 5: paddr=00154f48 vaddr=400c0000 size=00068h (   104) load
E (532) esp_image: Image length 1331168 doesn't fit in partition length 1310720
E (537) boot: Factory app partition is not bootable
E (543) esp_image: image at 0x150000 has invalid magic byte (nothing flashed here?)
E (551) boot: OTA app partition slot 0 is not bootable
E (557) boot: No bootable app partitions in the partition table

philippe44, 2023-10-26:
Take an older recovery image (or just an older version) meanwhile to get started

kwaek, 2023-10-26:
sorry.. I'm not sure how to do that... also the last time I flash an WROVER was quite a while back..... I basically used the web tools to flash.. don't think I can choose the version. Below is the current one I'm using without any issue.....

[squeezelite-esp32](https://github.com/sle118/squeezelite-esp32)
© 2020, philippe44, sle118, daduke
[This software is released under the MIT License.](https://opensource.org/licenses/MIT)

This app would not be possible without the following libraries:

squeezelite, © 2012-2019, Adrian Smith and Ralph Irving. Licensed under the GPL License.
esp32-wifi-manager, © 2017-2019, Tony Pottier. Licensed under the MIT License.
SpinKit, © 2015, Tobias Ahlin. Licensed under the MIT License.
jQuery, The jQuery Foundation. Licensed under the MIT License.
cJSON, © 2009-2017, Dave Gamble and cJSON contributors. Licensed under the MIT License.
esp32-rotary-encoder, © 2011-2019, David Antliff and Ben Buxton. Licensed under the GPL License.
tarablessd1306, © 2017-2018, Tara Keeling. Licensed under the MIT license.

philippe44, 2023-10-26:
@sle118 will likely fox the build tomorrow then

Robyxxx, 2023-10-26:
thank you all for the help!

kwaek, 2023-10-26:
thanks... appreciate it

lebolk, 2023-10-26:
Hi there ! I've got the exact same issue.
WROVER-E ref.XXN4R8 ESPRESSIF
Thanks for your help @sle118

dwhacks, 2023-10-27:
having the same issue with a generic wrover board. Had to go all the way back to 21621 to get it to work.

kwaek, 2023-10-28:
@dwhacks ... would you mind telling me how can I flash it with the old binary without using the web tools ? I think I managed to download the binary but I tried using ESPhome web tools to install the binary but it doesn't boot up properly.

sle118, 2023-10-28:
I have identified the issue and implemented a fix in my local development environment, but before I can release it, I have to do some additional testing. 

I will try my best to release this weekend.

Thank you for your understanding.

dwhacks, 2023-10-28:
> @dwhacks ... would you mind telling me how can I flash it with the old binary without using the web tools ? I think I managed to download the binary but I tried using ESPhome web tools to install the binary but it doesn't boot up properly.

The instructions are in the first post of the thread: 

_"If you don't use a SqueezeAMP where I pre-load the basic firmware then you should first flash your device with a version found here https://github.com/sle118/squeezelite-esp32/tags, download one of the .zip version and use the espressif download tool here [https://www.espressif.com/en/support...ad/other-tools](https://www.espressif.com/en/support/download/other-tools). You want to download:

    ota_data_initial.bin at 0xd000
    bootloader.bin at 0x1000
    partition-table.bin at 0x8000
    recovery.bin at 0x10000"_

kwaek, 2023-10-29:
Thanks @dwhacks !!   Looking forward also for @sle118 new fix ...

sle118, 2023-10-29:
Fixed in https://github.com/sle118/squeezelite-esp32/commit/58fac99e95d3c968325b549e4cbecccb7debeedf

closed 2023-10-29 by sle118

kwaek, 2023-10-29:
is working fine now... thanks @sle118 !!!

Robyxxx, 2023-10-29:
The installer (https://sle118.github.io/squeezelite-esp32-installer/) is now working with the right version of the firmware?

d0m1n1qu3, 2023-10-29:
works for me too .. thanks for fixing that :-)

lebolk, 2023-10-29:
Thanks a lot ! It fixed the installer issue !
Do you guys achieve to play with webui on labtop ? On my side it only works on smartphone.
On PC I Can only see the menu.

kwaek, 2023-10-29:
@lebolk .. yes.. I'm able to access the web-ui from the ip address without any issue...