sle118/squeezelite-esp32 #347: First: wouldn't reconnect to Wifi after flashing. Now won't boot at all.

cobbr2, 2023-10-31:
To help us resolve your issue as quickly as possible, please follow these guidelines when submitting an issue. Providing all the necessary information will save both your time and ours.

### Describe the bug
SqueezeAmp doesn't retain Wifi information across reboot; upon reflashing seems not to boot at all.

### Preliminary Information

1. Original: **SqueezeAmp-1336**. Now, whatever is current on the website as of 2023-10-28
2. **Plugin not installed**: Mention the version of the plugin installed on your LMS (Logitech Media Server); still running LMS 7.8.0 since my new VBR thing ain't working yet.

### Hardware Details

Please describe your hardware setup:

-  SqueezeAmp (purchased winter 2022-2023). Not sure whether it's Boost or not, I think it *is*.
- **Board Type**: If applicable, e.g., ESP32 audio kit, etc.
- **DAC Chip**: Specify the DAC chip you are using.
- **Additional Hardware**: Power switch, 2 18640 Li batteries, 15V power supply

### NVS Settings

Unretrievable

### Logs

Repeats
```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (52) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (52) boot: compile time 06:02:17
I (53) boot: chip revision: v1.0
I (57) qio_mode: Enabling default flash chip QIO
I (62) boot.esp32: SPI Speed      : 80MHz
I (67) boot.esp32: SPI Mode       : QIO
I (71) boot.esp32: SPI Flash Size : 4MB
I (76) boot: Enabling RNG early entropy source...
I (81) boot: Partition Table:
I (85) boot: ## Label            Usage          Type ST Offset   Length
I (92) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (99) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (107) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (115) boot:  3 recovery         factory app      00 00 00010000 00140000
I (122) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (130) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (137) boot: End of partition table
I (141) boot: Defaulting to factory image
I (146) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=59548h (365896) map
I (256) esp_image: segment 1: paddr=00069570 vaddr=3ffbdb60 size=04850h ( 18512) load
I (261) esp_image: segment 2: paddr=0006ddc8 vaddr=40080000 size=02250h (  8784) load
I (265) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=d2700h (861952) map
I (507) esp_image: segment 4: paddr=00142728 vaddr=40082250 size=12848h ( 75848) load
I (532) esp_image: segment 5: paddr=00154f78 vaddr=400c0000 size=00068h (   104) load
E (532) esp_image: Image length 1331216 doesn't fit in partition length 1310720
E (537) boot: Factory app partition is not bootable
E (543) esp_image: image at 0x150000 has invalid magic byte (nothing flashed here?)
E (551) boot: OTA app partition slot 0 is not bootable
E (557) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57
```

### Issue Description

1. **Observed Behavior**: Now, I think the problem is that after flashing, the unit still isn't capable of booting.  Previously:
> I successfully flashed my SqueezeAmp using https://sle118.github.io/squeezelite-esp32-installer/​ (and an old Arduino 2 USB - > Serial adapter). It scans and finds my wifi. But it does *not* retain the WIFI across re-boots.

> I've changed its System -> Device Name, so now what happens is on reboot, it's in Wifi discovery mode, but with the network > name that matches that device name. I can set the Wifi up again, and it will hook up. Sometimes it hooks up to LMS when I do > that, sometimes it doesn't. It claims to be in "SqueezeAmp" mode once I've got the wifi set up.

3. **Expected Behavior**:  Once the Wifi network is selected, it should connect to that network on every boot unless it can't reach that network.
4. **Steps to Reproduce**: Hard to say: I flashed it the first time, then I think I restarted the unit, and it did the normal discovery.  Then I set up the wifi, went through the change in network, changed the device name, played a song, and said "Victory!".  The next day I booted and it wasn't on the network.  It did advertise its new device name as a network, though, so I was able to go through discovery and hookup again.  This repeated.  I did 
change some other settings that also seem to stick OK across reboots; in particular, I've got the [https://www.buydisplay.com/i2c-blue-0-91-inch-oled-display-module-128x32-arduino-raspberry-pi](https://sle118.github.io/squeezelite-esp32-installer/) display hooked up and it seems to be working (though I'm still running LMS 7.x on my main server, so no plugin yet). When it hooks all the way up, it correctly identifies the LMS server IP. When it doesn't, it displays "SqueezeESP32".​

On the advice of sle118 on the SqueezeAmp forum post, I re-flashed with the latest at https://sle118.github.io/squeezelite-esp32-installer/ on Saturday 28 October.  I'm using an old [Arduino-2 adapter](https://docs.arduino.cc/retired/boards/arduino-usb-2-serial-micro) to flash, which means I use the hardware reset button to get it to load.

After re-flashing, I don't get a clean boot, as above.  I suspect I need to do something after flashing and before power-cycling, but I don't know what.

philippe44, 2023-10-31:
There was an issue with the version prior to 1634 which unfortunately did not write correctly the recovery partition. If you try again *now*, it should write properly a recovery partition as well as and the application partition.

sle118, 2023-10-31:
See https://github.com/sle118/squeezelite-esp32/issues/345

sle118, 2023-11-03:
Ok assuming no news is good news. Closing for now

closed 2023-11-03 by sle118

cobbr2, 2023-11-05:
Thanks folks, finally got around to trying the new load and it worked great.  Thanks for all the good work!